bambulab/BambuStudio #1801: Show how much filament was already printed.

JonasTechnik, 2023-05-12:
It would be good to know how much filament was already printed. This could be useful if you want to know how much filament was wasted if you stop a print that failed.

sodachen22, 2023-05-15:
Printed layer number and printed percentage are already displayed, can you get a rough estimate on this information?
And if there is unacceptable printing quality issue in the printed part, the only way is cancel this printing job even if a lot of filament has been used. The printed layer number can tell the user the exact layer if he wants to start the next printing job from this layer and glue it to the previous part.

JonasTechnik, 2023-05-17:
I mean a live view while printing

closed 2023-05-20 by JonasTechnik