bambulab/BambuStudio #1807: Prusament PETG profile add

athesi, 2023-05-14:
Hello.

Prusament PETG is one from the best petg filaments and really common in Europe but printing with X1C this filament is a real labor. Please find a profile. Regards

madproducts, 2023-05-25:
HI. I was facing the same issue with the X1 Carbon.  Finally I copied the settings from PrusaSlilcer for Prusament PETG (0.02mm layer height) . I used a Textured PEI Sheet and ran flow calibration/bed leveling with the print. The results were a lot better.  0.4mm nozzle (mine is hardened steel but stainless steel should work just as good).  Attaching the printer and filament profile files. hope this helps. 
Note : Change file extension from ".txt" to ".json" if you want to import into Bambu Studio.
[0.20mm Prusament PETG @BBL X1C.txt](https://github.com/bambulab/BambuStudio/files/11560639/0.20mm.Prusament.PETG.%40BBL.X1C.txt)
[Prusament PETG  @BBL X1C.txt](https://github.com/bambulab/BambuStudio/files/11560640/Prusament.PETG.%40BBL.X1C.txt)

QingZhangBambu, 2023-05-25:
good point. we will test it. and adjust our parameters

Xiuqi-Huang-CN, 2023-08-01:
> HI. I was facing the same issue with the X1 Carbon. Finally I copied the settings from PrusaSlilcer for Prusament PETG (0.02mm layer height) . I used a Textured PEI Sheet and ran flow [...]

Hello, so sorry to reply you so late. We have find out the reason. Prusa Slice has set a much lower printing speed for PETG, and if you adjust Bambu Slicer's PETG profile to have the same low speed, it will work very well, too. In fact, in Bambu slice, you should only adjust PETG's max volumetric speed from 13 to 12 or 10 to  lower the speed about 20% to get a good enough finish of the print, and don't have to set very low printing speed like Prusa Slice that will cost much longer time.

pontos-dev, 2023-10-27:
I can recommend OurAngryBadger (OAB) Filament and printer profiles to Bambu studio, he have many profiles, and the profiles is very very good, and he has Prusament profiles too

Tranilator, 2023-11-24:
Just wanted to share some thoughts on why Bambu Studio supporting Prusament filament officially is super important.

Prusament filament is a big deal, especially across Europe. It's known for its quality and a lot of us are big fans. But here's the thing – there are tons of Prusa users out there who are really interested in jumping into the Bambu community and using your products. The catch? The lack of official Prusament support in Bambu Studio is kind of an unnecessary roadblock. Particularly for those sitting on existing Prusament filament supplies.

We've been getting by, tweaking settings manually or copying profiles from PrusaSlicer or elsewhere. It works, but it's not ideal. Having everything ready to go in Bambu Studio would make uptake and life so much easier and more welcoming for these potential new members of our community.

I noticed you guys were testing and sharing tips on the filament, which is fantastic. But the last update from Bambu was back in August and we're still waiting for that official profile support. Bringing this support to life could really open doors for many Prusa users looking to join us and explore what Bambu has to offer.

longwa, 2023-12-12:
I'm a Prusa MK3 user that just sold my MK3 and switched to X1C. PLA printing with Bambu filament was amazing, I was happy I had sold my MK3. Switching to PETG, I wondered if I made a mistake. The out of the box printing with Prusa Galaxy Black PETG, for example, was terrible. I cobbled together a bunch of different advice online and fixed the profile to work a bit better, but it's still a long way from the reliability of the MK3 for printing PETG.

I know, many people will say I just used the Generic PETG and have never had an issue. That's great, but why not support more filament with testing and profiles out of the box. I know you want people to buy your filament, but in the interest of the community it would be great to support more with professionally tuned parameters that make your machine look good.

Xiuqi-Huang-CN, 2023-12-13:
@longwa @Tranilator @pontos-dev Hello, everyone. Custom filament functionality has been available in the 1.8 Version Studio, and you can add any filament profiles by yourselves easily., and you can ask the filaments supplier for suitable filaments profiles and Bambu team for some recommended parameters.

KickingKing, 2024-01-10:
LoL you and your request is funny :D
Prusa is/was the biggest BBL  competition in some segments. Why do you think we have A1 mini and A1? To kill what is left of Prusa.
And you think they will do Prusament profile (I am here searching for one now :D)? No way mate, no waaay
Then people would like Creality one, and so on and so on. Have you seen Prusa slicer with BBL or Ultimaker or whatsoever profiles inside?

Misiu, 2024-01-19:
Does anyone have settings for A1?
I just switched from Prusa Mini+ to Bambu Lab A1 and have a bunch of Prusa PETG filaments I'd like to use.
What is the procedure for adding profiles? There are many brands popular in Europe that are not added.