bambulab/BambuStudio #1809: Incorrect imperial weight displayed after slicing

toddanglin, 2023-05-15:
When Bambu Studio is configured to show imperial units (in, oz), it forgets to convert the "filament" weight value to ounces after slicing in the "Preview" view.

**Steps to reproduce the behavior:**
1. Open Bambu Studio and make sure units are set to Imperial in "Settings..."
2. Add a model (any model) to the project and slice
3. In the "Color Scheme" overlay, choose "Filament" and note the imperial weight value in ounces
4. Now, go to "Settings..." and change Bambu Studio to metric units
5. Note the new (correct!) metric value displayed for weight in grams
6. As you can see, Bambu Studio shows the same numeric value for grams and ounces and does not properly convert between units. Ounces should be a smaller number than grams.

**Screenshots**
_Metric units (correct)_
[image]

_Values after changing settings to Imperial units_
[image]

In this example, ounces should be shown as 0.94 instead of 26.55. Meanwhile, the conversion between meters and inches is correct.

**Desktop (please complete the following information):**
 - OS: macOS 13.3.1
 - Bambu Studio Version: 1.6.2.4

LiZ-Li-BBL, 2023-05-19:
We'll check this issue.

LiZ-Li-BBL, 2023-07-21:
This issue has been fixed. Please update your version of BambuStudio to this : https://github.com/bambulab/BambuStudio/releases/tag/v01.07.00.65

closed 2023-07-21 by LiZ-Li-BBL