bambulab/BambuStudio #1810: Auto Support Failed with Chamfered Area

BREACHER556, 2023-05-15:
**Describe the bug**
Hi All,

Somehow the slicer cannot generate auto support in certain areas (Pic 1), cannot figure out why.

Not sure is that the top surface is too sharp. Or does the bottom surface have an angle? (Pic 2)

I tried to change the design to give it less chamfer / top chamfer only  / bottom chamfer only, and finally able to generate support when the flat area is 1mm or less or no top chamfer(see pic 3, 4 & 5)

Any idea why? (Sure I can change the design or add support manually, but I want to know if is there a parameter I need to follow to get auto support)

Also, seems like auto support will be generated when I set the Threshold Angle to 45°, but then now support is everywhere.

Thank you!

**3mf File for This Bug**
[Chamfer Support Test.zip](https://github.com/bambulab/BambuStudio/files/11474656/Chamfer.Support.Test.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. open the 3mf file
2. Slice it
3. See error
4. To get auto support works for all, angle need to be at least 45°

**Expected behavior**
As above

**Screenshots**
See attached Zip file

**Desktop (please complete the following information):**
 - Window 11
 - Version 1.6.2.4

tangketan, 2023-05-29:
Thanks for reporting the issue. It's caused by the "small overhang removal" function. We'll try to fix it in the next version.

tangketan, 2023-05-31:
I've added an option to let you disable the function "small overhang removal".
But the gap between the object the supports are still too large as the blade is too sharp. I'm not sure how useful it is.
[image]

closed 2023-08-10 by tangketan