bambulab/BambuStudio #1812: Multi color print purge efficiency

Kurt-Venable, 2023-05-16:
When doing prints with 3 or more colors you could save a lot of filament by arranging changes with the least amount of purging. For instance it may save filament going from black to orange to white instead of black to white to orange. This would be more effective with the more colors in use and the order would need to change on each layer as the start color will always be the finish color of the previous layer. Just looking at a recent 4 color part I had it could make around a 300 mm3 difference per layer. Add in hundreds of layers and it really adds up.

zhimin-zeng-bambulab, 2023-05-19:
@Kurt-Venable 
Our current material change order is calculated based on the minimum flushing volume. 
Could you  provide a picture you find the error phenomenon?   And could you  share your 3mf file?

Billiam, 2023-06-19:
@zhimin-zeng-bambulab Sure!

Here's a 3mf file with a common use case: text on the first two layers using a different filament.

[color-test.zip](https://github.com/bambulab/BambuStudio/files/11793975/color-test.zip)

Color order is important here for two reasons: 

1. Some colors print better depending on order, especially with fine details and
2. Minimizing filament swapping

BambuStudio tries to print the text first here, no matter what I do. This should only require two color changes, but BambuSlicer uses three instead:

Layer 1: background filament, change to text filament
Layer 2: text filament, change to background filament

Purge volume doesn't affect the order in this case, both of these print 
![Screenshot_20230619_183714](https://github.com/bambulab
/BambuStudio/assets/242008/7c518e83-97bf-4c1d-bf4b-af90b911d099)
[image]
the same:

zhimin-zeng-bambulab, 2023-07-10:
https://github.com/bambulab/BambuStudio/issues/1816  the problem is the same with this one, we will enhance in the future.

zhimin-zeng-bambulab, 2023-08-18:
In the past, the printing order of the first layer was fixed, depending on the shape of the model. 
In the latest version(v1.7.4),  you can modify the printing order of the materials in the first layer.
The printing order of the non-first layer is automatically calculated, ensuring Minimum total flush volume per layer

You can have a try in the latest version.

closed 2024-01-18 by zhimin-zeng-bambulab