bambulab/BambuStudio #1814: Latest AppImage Crashes Trying To Display P1P Video

nmcbride, 2023-05-16:
**Describe the bug**
When I try to display the video from the P1P using the latest AppImage (Bambu_Studio_linux_fedora_v01.06.02.04-20230427094209.AppImage) one of two things happens:

1.  It asks me for the IP and Access Code -> I enter that info -> It says it connects and then crashes.
2. It just crashes.

**Expected behavior**
Should stream the video as it does in earlier versions.

I have two previous versions downloaded as well:
- Bambu_Studio_linux_fedora_v01.05.00.61_20230314200047.AppImage
- Bambu_Studio_linux_fedora_v01.06.00.65_20230417141721.AppImage

These both stream the video perfectly fine. Albeit low quality and low frames on the P1P.

nmcbride, 2023-05-17:
@lanewei120 

According to the issue you linked the AppImage I am using is the one that has already had the libjpeg version downgraded in it.

I'm on Manjaro.

This is what is on my system:
libjpeg.so
libjpeg.so.62
libjpeg.so.62.2.0
libjpeg.so.8
libjpeg.so.8.2.2
libjpegxr.so
libjpegxr.so.0
libjpegxr.so.0.0.0

lanewei120, 2023-05-17:
then we need to check the libjpeg used by gstreamer
for example:
readelf -d /lib64/gstreamer-1.0/libgstjepg.so
---------------------------
0x0000000000000001 (NEEDED)             共享库：[libgstreamer-1.0.so.0]
 0x0000000000000001 (NEEDED)             共享库：[libgobject-2.0.so.0]
 0x0000000000000001 (NEEDED)             共享库：[libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             共享库：[libgstbase-1.0.so.0]
 0x0000000000000001 (NEEDED)             共享库：[libgstvideo-1.0.so.0]
 0x0000000000000001 (NEEDED)             共享库：[libjpeg.so.62]
 0x0000000000000001 (NEEDED)             共享库：[libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [[libgstjpeg.so](http://libgstjpeg.so/)]
 0x000000000000000c (INIT)               0x3000
 0x000000000000000d (FINI)               0x9de8
 0x0000000000000019 (INIT_ARRAY)         0xd890
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0xd898
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)

on my Fedora 36, it uses libjpeg.so.62

nmcbride, 2023-05-20:
Looks like mine is .8

 0x0000000000000001 (NEEDED)             Shared library: [libgstreamer-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgstbase-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgstvideo-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgobject-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libjpeg.so.8]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]

lanewei120, 2023-05-24:
the root cause is the libjpeg version built with the appimage is jpeg6
and on your system gstreamer uses jpeg8

We just found on Fedora, some uses jpeg6 and some uses jpeg8
and maybe we have following fix methods:
1. built gstreamer related plugins as static libraries into BambuStudio which use the same jpeg version
or
2. release two images for fedora

WayneManion, 2023-07-31:
Isn't the whole point of AppImage that it has all the dependencies built-in? The idea is that you only have to release one AppImage that works in all distributions that support AppImage.  

I am running Arch fwiw. 

bambustudio-git on  master [!] took 5h40m42s 
❯ readelf -d /lib64/gstreamer-1.0/libgstjpeg.so 

Dynamic section at offset 0xba10 contains 29 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libgstreamer-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgstbase-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgstvideo-1.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libglib-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libgobject-2.0.so.0]
 0x0000000000000001 (NEEDED)             Shared library: [libjpeg.so.8]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libgstjpeg.so]
 0x000000000000000c (INIT)               0x3000
 0x000000000000000d (FINI)               0x8cdc
 0x0000000000000019 (INIT_ARRAY)         0xc930
 0x000000000000001b (INIT_ARRAYSZ)       8 (bytes)
 0x000000000000001a (FINI_ARRAY)         0xc938
 0x000000000000001c (FINI_ARRAYSZ)       8 (bytes)
 0x000000006ffffef5 (GNU_HASH)           0x310
 0x0000000000000005 (STRTAB)             0xe60
 0x0000000000000006 (SYMTAB)             0x338
 0x000000000000000a (STRSZ)              3012 (bytes)
 0x000000000000000b (SYMENT)             24 (bytes)
 0x0000000000000007 (RELA)               0x1b88
 0x0000000000000008 (RELASZ)             3408 (bytes)
 0x0000000000000009 (RELAENT)            24 (bytes)
 0x000000000000001e (FLAGS)              BIND_NOW
 0x000000006ffffffb (FLAGS_1)            Flags: NOW NODELETE
 0x000000006ffffffe (VERNEED)            0x1b18
 0x000000006fffffff (VERNEEDNUM)         2
 0x000000006ffffff0 (VERSYM)             0x1a24
 0x000000006ffffff9 (RELACOUNT)          26
 0x0000000000000000 (NULL)               0x0

WayneManion, 2023-08-01:
I am running Arch. I am running the AUR package (built from source) from here: https://aur.archlinux.org/packages/bambustudio-git

The application doesn't crash when I click the play button to view the camera stream, but I can't see any video either. It just says:

Loading Failed [2]!

The gdb logs from running the built-from-source version of the app is here: https://pastebin.com/bhUK8JAh

lanewei120, 2023-08-01:
> Isn't the whole point of AppImage that it has all the dependencies built-in? The idea is that you only have to release one AppImage that works in all distributions that support AppImage.

I am not familiar with this,
I think it is caused by gstreamer not included in the appimage, gstreamer uses libjpeg which has a different version

nmcbride, 2023-08-01:
It is frustrating that the appimage isn't self contained.

I recently just switched over to NixOS and tried to run the appimage and couldn't. Then I thought I'd use distrobox and run it but it needed a ton of dependencies to be installed.

Ended up installing from source on NixOS and also have the black screen to the P1P with with the Loading Failed error.

Could we get a flatpak build that is self contained that we could just install from flathub like prusaslicer?

aduewel, 2023-11-09:
> I am not familiar with this, I think it is caused by gstreamer not included in the appimage, gstreamer uses libjpeg which has a different version

Would it be really complicated to include it but as benefit only have one appimage which is supported in every distro?

aduewel, 2023-12-20:
Update: BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage works on Debian 12!

benoitm974, 2024-06-16:
> Update: BambuStudio_linux_fedora_v01.08.02.56-20231213.AppImage works on Debian 12!

Cnfirmed for me Debian 12 video crash using ubuntu, using fedora version require to install libosmesa6 but then it works and video doesn't crash. Using the ubuntu even after installing libosmesa6 doesn't help though.