bambulab/BambuStudio #1815: Bambu Studio prevents desktop monitor from going to sleep

tangcla, 2023-05-16:
**Describe the bug**
When Bambu Studio is open, my computer doesn't turn the monitor off after 5 minutes (which it normally does).

**Expected behavior**
I expect that the computer would continue with its monitor off as defined by the system, but Bambu Studio seems to keep the computer awake.

**Desktop (please complete the following information):**
Windows 11 Pro, 22H2 (build 22621.1702)

KD7LRJ, 2023-05-18:
My system does this also if I leave the video from the camera playing. If you switch away from the video window, it will sleep correctly. It's nice to sometimes leave the video playing in a window to keep an eye on progress so I changed the setting in my system's Power Options to allow sleeping while video is playing:
[image]

Just be aware that this also causes the monitor to shut off while streaming a movie which isn't ideal... :)

Photogad, 2023-05-19:
+1 for this
Recently my computer was running too long because Bambu Studio did not let my computer sleep and someone used remote desktop to hack into my computer and steal my bank account.
I have since disabled remote desktop features in windows, but had the computer gone to sleep the hacker would not have been able to do this

longrackslabs, 2024-09-23:
We will look back on this terrible change to this non-problem as the tipping point where Bambu Studio lost to Orca.

bambu123, 2024-10-11:
……