bambulab/BambuStudio #1816: Unable to define which filaments print first, for each layer

mattie47, 2023-05-17:
**Describe the bug**

BambuStudio does not seem to follow a consistent process for deciding which filament color prints first on each layer.

I've read on bambu forums to solve this by:
* Ordering Filament in the slicer to the priority of filament color

[image]

**This however does not consistently work, and I can find myself slicing the same STL with differing results**

I have also tried other settings with no luck:
* Slicing/combining multiple objects
* Changing filament order in the slicer, and also "default Color"
* Changing "Order of inner wall, outer wall, infill" etc
* Tried the two latest versions of Bambu Slicer

Nothing seems to impact how the slicer decides which filament color is first. 

**As a user, I NEED the ability to do this, to ensure the first layer on the print bed has one filament color, before another.** 

**Use case**
In my case, white needs to go down before black, to ensure text on the object I'm printing comes out consistently. If the black goes first, then the white text becomes unreadable.

**3mf File for This Bug**
I've attached in a zip file:
* A sample STL which can be imported into Bambu.
* 3MF file

**To Reproduce**

I have tried many things to "workaround" but consistently get black generated first.

Please see attached 3MF, as well as the below video

Video details:
- Open Bambu 1.6.0.65 (note that I'd downgraded from 1.6.2.4 which is where I was also encountering this problem)
- Drag in the STL
- Assign default color of STL as black
- Fill in text with white
- Slice file
- **observe issue, despite filament order having white, then red, then black, then support W**

I then show the same STL with same steps open in Orca Slicer, and show white is printed before black. 

https://github.com/bambulab/BambuStudio/assets/4696925/97fd89b3-c663-439c-b7aa-64fa7fe0a35b

**Expected behavior**
G-Code generated is prioritized in order of Colors defined in Filament Ordering within the slicer. I.e. White, then Red, Then Black. 

Alternatively, a feature which makes this ability more obvious for user control. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Good:**
The following is from Orca Slicer:

[image]

**Bad:**
White is generated Last:

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: Attempted Bambu 1.6.2.4, and then Bambu 1.6.0.65

Thanks

mattie47, 2023-05-18:
**TL;DR - I think I've solved the issue with my specific file, however still being able to more explicitly set the first print colour would be useful**

With "dots" or things painted a certain colour which are "too small", the slicer seems to decide to print the stuff around it first. 
See the below video demoing this, as well as a "workaround" for the STL to use 0.6 circle as a replacement:

https://github.com/bambulab/BambuStudio/assets/4696925/dbcf9e49-260e-4290-8bdb-8b0f23229479

Perhaps this is due to nozzle size and print size and first layer line width default of 0.5 for a 0.4 nozzle?

**Analysis - Below is just some notes, but not of complete relevance**

I think I've figured out what's different between Bambu and the print I have with Orca slicer.

In Bambu I have:

```
;===== prepare print temperature and material ==========
M104 S[nozzle_temperature_initial_layer] ;set extruder temp
G91
G0 Z10 F1200
G90
G28 X
M975 S1 ; turn on 
G1 X60 F12000
G1 Y245
G1 Y265 F3000
M620 M
M620 S[initial_no_support_extruder]A   ; switch material if AMS exist
    M109 S[nozzle_temperature_initial_layer]
    G1 X120 F12000

    G1 X20 Y50 F12000
    G1 Y-3
    T[initial_no_support_extruder]
    G1 X54 F12000
    G1 Y265
    M400
M621 S[initial_no_support_extruder]A
M620.1 E F{filament_max_volumetric_speed[initial_no_support_extruder]/2.4053*60} T{nozzle_temperature_range_high[initial_no_support_extruder]}
```

In Orca I have:
```
;===== prepare print temperature and material ==========
M104 S[nozzle_temperature_initial_layer] ;set extruder temp
G91
G0 Z10 F1200
G90
G28 X
M975 S1 ; turn on 
G1 X60 F12000
G1 Y245
G1 Y265 F3000
M620 M
M620 S[initial_tool]A   ; switch material if AMS exist
    M109 S[nozzle_temperature_initial_layer]
    G1 X120 F12000

    G1 X20 Y50 F12000
    G1 Y-3
    T[initial_tool]
    G1 X54 F12000
    G1 Y265
    M400
M621 S[initial_tool]A
```

The key difference I see is around M620, and M621

From looking at the raw Gcode, I see the following each AMS colour change:
```
M620 S0A - I assume AMS 1 slot 1
M620 S1A - I assume AMS 1 slot 2
M620 S2A - I assume AMS 1 slot 3
M620 S3A - I assume AMS 1 slot 4
```

I've tried changing the gcode to the following:
```
;===== prepare print temperature and material ==========
M104 S[nozzle_temperature_initial_layer] ;set extruder temp
G91
G0 Z10 F1200
G90
G28 X
M975 S1 ; turn on 
G1 X60 F12000
G1 Y245
G1 Y265 F3000
M620 M
M620 S[initial_tool]A   ; switch material if AMS exist
    M109 S[nozzle_temperature_initial_layer]
    G1 X120 F12000

    G1 X20 Y50 F12000
    G1 Y-3
    T[initial_tool]
    G1 X54 F12000
    G1 Y265
    M400
M621 S[initial_tool]A
M620.1 E F{filament_max_volumetric_speed[initial_tool]/2.4053*60} T{nozzle_temperature_range_high[initial_tool]}
```

I tried editing this, but still didn't seem to make a change. I couldn't find anything on those two variables to know which is "better" practice to have.

Photogad, 2023-05-19:
I will +1 this. It would be nice to pick which filament prints first for multicolor prints, this would help solve some printing problems when the printer decide to print certain small details first and they can become detached from the bed.

coryisakson, 2023-05-25:
+1, I am printing a lot of things with white and other colors on the first layer. I have found that when other colors go first they often leave artifacts in the white layer. When I can get the white to print first I have consistently good outcome.

mattie47, 2023-05-25:
One thing for consideration around potential improvements:
- It may be desirable to only specify print color order for the first layer. 

What I'm printing is a black box/case with white logo/text on first layer, and then white text/labels on different layers outside of the box

If white is prioritized in the print order, it will use white infill on subsequent layers. It seems this is impacting the overall finish of the outside black case. Perhaps this could be tuned with the "right settings" but my prints were definitely better on the outside of the box when black was also being used as in-fill. 

Right now I'm looking at slicing two files, and then merging the gcode to get around the first layer color challenges.

coryisakson, 2023-05-25:
I recommend creating 2 STL files and then importing as a single object. You
can then set the colors of each object. Still can't decide what color
prints first, but it gives you excellent control over individual colors of
the model.

Mrblazed, 2023-05-28:
Please allow us to choose the first layer color order.

zhimin-zeng-bambulab, 2023-05-29:
Currently, our first layer is printed according to the minimum area of each color, in descending order of size, which can ensure the highest printing success rate.

If it is for a higher printing success rate, the current first layer printing order is the best. If it is to remove artifacts, it can be achieved by setting a larger flushing volume. We are also evaluating whether it is necessary to develop the adjustment of the first layer order.

Mrblazed, 2023-05-29:
> Currently, our first layer is printed according to the minimum area of each color, in descending order of size, which can ensure the highest printing success rate.
> 
> If it is for a higher printing success rate, the current first layer printing order is the best. If it is to remove artifacts, it can be achieved by setting a larger flushing volume. We are also evaluating whether it is necessary to allow to modify the first layer order.

Thank you.  In certain circumstances, such as lettering, being able to control the first layer color order can be the sole difference between a high quality print and a low quality print.   I appreciate your consideration.

cchangdev, 2023-06-19:
+1. I absolutely need this as a feature.

cchangdev, 2023-06-19:
The way I've found to get around this is kind of silly, but it seems to work.  Add a tiny primitive, like a cube, to the object outside of the object you're printing, then export that as a single STL. It won't work if you just add it to the plate in addition to the model you're printing. Re-import the STL and paint the bottom of the cube (or whatever you added) the color you want to print second. It will then switch the order of the colors.

Vorta, 2023-06-21:
I have a print that needs to use white and black filament. Black should go only on the first layer. That means if the black goes first, there is only one filament change needed during the print. However, the slicer always selects white first, which means it needs to load white first, change to black thereafter and, finally, do the last switch back to white for the rest of the print. It's wasteful and makes no sense. I'm surprised I cannot force priority of the filament, especially when it is only for a single layer.

cchangdev, 2023-06-21:
> Of course you can. Go to objects vs global in the slicer. Select the Color for your object to be the same Color you want for the first layer, then do a color change on the second layer to the Color you want….

This is for color print sequence within the same layer

FluxDragon, 2023-06-28:
Just piling on in case it helps prioritize addressing this feedback.
+1, I am printing a lot of things with white and other colors on the first layer. I have found that **when other colors go first they often leave artifacts in the white layer.**

Badboy49er, 2023-07-02:
This should be very simple honestly, it needs to be an option to be able to select which filaments print first or it needs to be simple as it prints in the order it’s in the AMS. There’s no reason not to have it this way.

tomparvin, 2023-07-02:
Adding to comments above - the AMS simply isn't usable in certain scenarios as a result of not being able to define the order in which the colours print within layers. I'm actually very surprised this isn't possible, or at least, there's a checkbox to say 'print in AMS reel order' so I can manually order the reels within the AMS in the order I want them to print.

Please fix this issue as soon as possible!

dantetemp, 2023-07-08:
+1 badly need this. @mattie47 thanks for workaround with small dot!

turtlecreations, 2023-07-14:
manual setting of the print order would be immensely appreciated for all the prints that need first-layer color changes. This could be the same setting as with brim, automatically or manually.

JamShady, 2023-07-19:
Another +1 for this feature please

rickgnyc, 2023-07-20:
+1
Please add this feature. I often use 2-colors on the first layer, with text inlaid into that first layer. The prints come out much cleaner  if the text goes down first.

bgateb, 2023-07-23:
+1 for this feature. it's immensely frustrating to not be able to specify the print order.

bradMatias, 2023-07-24:
@zhimin-zeng-bambulab 
> Currently, our first layer is printed according to the minimum area of each color, in descending order of size, which can ensure the highest printing success rate.
>
> If it is for a higher printing [...]

I'm not sure if this is not true on 1.7.0.65

Here are 2 example files of the same model painted:
[Brad_blackFirst.zip](https://github.com/bambulab/BambuStudio/files/12141382/Brad_blackFirst.zip)
[image]

[Brad_whiteFirst.zip](https://github.com/bambulab/BambuStudio/files/12141387/Brad_whiteFirst.zip)
[image]

The only difference here is started with an all white model by right clicking -> select filament for Brad_whiteFirst.zip and I started with an all black model with Brad_blackFirst.zip.  Both are using the same amount of each filament on the first layer.

SaltWei, 2023-07-24:
Hi guys. We have added this function into 1.8 developing plan. Not in 1.7 version

SaltWei, 2023-07-24:
> this function

User defines the filament order of first layer

rickgnyc, 2023-07-24:
> Hi guys. We have added this function into 1.8 developing plan. Not in 1.7 version

Thank you for this news!

bradMatias, 2023-07-26:
I came up with a workaround based on @mattie47 's suggestion.  This will allow you to force the slicer to print any color before a single color.

STL: [Color_select.zip](https://github.com/bambulab/BambuStudio/files/12167657/Color_select.zip)
In my video example I want black to print first, so I set the attached model [Color_select.stl](https://github.com/bambulab/BambuStudio/files/12167657/Color_select.zip) to black.  I want white to print after black, so I color 1 circle on the bottom of the [model](https://github.com/bambulab/BambuStudio/files/12167657/Color_select.zip) white, you can add more colors if you want more to print before the base color as well.  Hope this helps someone!

https://github.com/bambulab/BambuStudio/assets/3682725/09c1a80e-40bd-45e9-a32c-650ffad7235b

mike-serinus, 2023-08-01:
Definitely plus 1 this - thanks for adding to the dev list

Ed19777, 2023-08-10:
Very important feature! Waiting for 1.8...! Thank you in advance!

parrothd, 2023-08-15:
Please add.. :)

egroven, 2023-08-15:
Please add this feature! By layer would be awesome too in most cases you only need it for the first few layers!

rickgnyc, 2023-08-16:
I am so proud to be an owner of an X1C. So far my experience with support, my printer, and all things Bambu has been so positive. Keep up the great work!!

SaltWei, 2023-08-17:
Included in latest release https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52.

parrothd, 2023-08-17:
Hurray!

ffafe5, 2023-10-28:
@SaltWei, this issue should not be closed. The new feature in v01.07.04.52 only offers order customization for the FIRST LAYER, and the request was for control of this in all layers. That is just as useful, for text and other patterns on the top of a model.

SaltWei, 2023-10-28:
> the request was for control of this in all layers. That is just as useful, for text and other patterns on the top of a model.

Actually, we have printed some models, and find that the filament order for first layer is really useful for some bed adhesive issue.
But we have not found the necessity for other layer. Could you raise some real examples with 3mf so we can re-print the model to have a look?

ffafe5, 2023-10-29:
Literally just painted text, and for the same reason as anyone who wants text on the first layer: to enforce clarity of these parts with priority over the surrounding edges. My project is personal to me. Sounds like a really simple feature that you were already halfway to supporting. Maybe I'll have to go add it to Orca myself and then y'all will "thank" them for the feature copy into BS

parrothd, 2023-10-29:
For me having control of all layers would be great, thanks for fixing the first layer. 

The issue I'm having is my object is white base with red text. I want to have several layers of red so the red letters pop. With only one layer the letters look dull so I do 6 layers deep. 

Sometimes after the first layer the red gets dragged into the white area and shows thru many layers. It think if it always printed white first(like on the first layer) then it may help reduce red blobs showing thru.

parrothd, 2023-10-29:
For me the reason isn't for adhesion, its so when you print multiple colors you limit the chance of a dark color being dragged into a lighter color area and ruins the first layer, the layer people see. 😀

SaltWei, 2023-10-29:
> The issue I'm having is my object is white with red text. I want to have several layers of red so the red letters pop. With only one layer of red the letters look dull so I do 6 layers deep of red.
> 
> Sometimes after the first layer the red gets dragged into the white area and shows thru many layers. It think if it always printed white first(like on the first layer) then it may help reduce red blobs showing thru. I've given up on trying to print in petg since it fails about 50% of the time with red blob somewhere in the white showing thru.

Thank you for the detail explanation！

SaltWei, 2023-10-29:
> Maybe I'll have to go add it to Orca myself and then y'all will "thank" them for the feature copy into BS

Just do it if you want.

parrothd, 2023-10-29:
It would be awesome if it could be set per layer.
Layer 1: white then black
Layer 2: black then white
Layer 3: white then black.

SaltWei, 2023-10-29:
> Thank you for the detail explanation！

> It would be awesome if it could be set per layer.
> Layer 1: white then black
> Layer 2: black then white
> Layer 3: white then black.

Yes. This can be developed

parrothd, 2023-10-29:
Oh man,

How about

Layer 1 white then black, extra retract and extra z hop,wait a few seconds then go wipe. 

Really anything that keeps colors from ozing then being dragged across. 👍😀

Dominic-Lentini, 2023-11-18:
I had zero issues with this for an entire year because I was able to get it to go by the AMS filament order and now all of a sudden it isn't working at all and is completely ignoring the filament order.

Pi-wi74, 2023-11-27:
many thank^s for this addons ; work perfectly now for me ;)

tedtieken, 2023-11-30:
+1 for the ability to set color sequence on every level.  This would be useful for creating consistent line widths on embedded text.

OzDruiD, 2023-12-06:
Hi all,  been struggling with a similar issue,  Colour sequence order, when _Print by OBJECT_ and using more than one colour, withg the same use case    - lay the TEXT first,  print 2nd\other colour around it.

Example thread in offical discord also but try and redo here
https://discord.com/channels/970163353542606888/1181758346680864808

Not the developer, but example model here - "text" in the end of the "CAPS" printed face down on the text side, with the text wanting to be printed first to ensure best quality (like other comments above).

Add 2 + objects and lay some text - print by colour mode - and ensure text iss the selected "1st" priority filament sequence.
[image]

This is what happens with the FIRST object in Print by object - ignores the white priority colour
[image]
reaching the 2nd object in the slicer preview - it clearly shows the 2nd object, printing the "text" white colour first before the rest of the 2nd cap - in blue.
[image]

This is what SHOULD happen - but this is a picture taken BY LAYER print - white is first etc.
[image]

and as expected, both objects white text first (as layer mode)
[image]

qp0n4ik, 2023-12-10:
looking forward to seeing this function in the slicer!

Pi-wi74, 2023-12-10:
> looking forward to seeing this function in the slicer!

it's already done ! you can select filament order ...

qp0n4ik, 2023-12-10:
> > looking forward to seeing this function in the slicer!
> 
> it's already done ! you can select filament order ...

Oh I haven't read this thread fully thank you for pointing it out

ffafe5, 2023-12-11:
WRONG; the ask here is to select filament order in ALL LAYERS, and it only supports FIRST LAYER

cannelloniAngelo, 2024-01-08:
[image]

cannelloniAngelo, 2024-01-08:
[image]

SaltWei, 2024-01-12:
> Hi all, and @SaltWei - maybe you have some Bambu Input - I been struggling with a similar issue, Colour sequence order, when _Print by OBJECT_ and using more than one colour, withg the same use case - lay the TEXT first, print 2nd\other colour around it.
> 
> Example thread in offical discord also but try and redo here https://discord.com/channels/970163353542606888/1181758346680864808
> 
> Not the developer, but example model here - "text" in the end of the "CAPS" printed face down on the text side, with the text wanting to be printed first to ensure best quality (like other comments above).
> 
> Add 2 + objects and lay some text - print by colour mode - and ensure text iss the selected "1st" priority filament sequence. ![image](https://private-user-images.githubusercontent.com/20792799/288257598-85a1bf3b-b876-4f0d-bb6d-629cad5046b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMjcwNzIsIm5iZiI6MTcwNTAyNjc3MiwicGF0aCI6Ii8yMDc5Mjc5OS8yODgyNTc1OTgtODVhMWJmM2ItYjg3Ni00ZjBkLWJiNmQtNjI5Y2FkNTA0NmI3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDAyMzI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMzNjE0NDhkYWFkYTlhODY3ODg3YTQwZjJiYmQwOTkzNjgxZDAzNzMzMDU3Y2M5ZDE5NmJhMmU2ZjZhMTRlMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8rLN_vUlgA6CAOFQRNit9bFu7jwTQX2pm4J0NQwvgPw)
> 
> This is what happens with the FIRST object in Print by object - ignores the white priority colour, yet review the purge lines - they ARE white.. so priority worked.. ![image](https://private-user-images.githubusercontent.com/20792799/288259341-99c50a0a-18f2-4658-8cc4-601cf3c213d9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMjcwNzIsIm5iZiI6MTcwNTAyNjc3MiwicGF0aCI6Ii8yMDc5Mjc5OS8yODgyNTkzNDEtOTljNTBhMGEtMThmMi00NjU4LThjYzQtNjAxY2YzYzIxM2Q5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDAyMzI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE3YzA5ZGI2YWMzN2M3MjJhZjM2N2Y2ZDc0ZWIyNTkyY2Q2NjY1ZWRmODhmMDljZGIxNDhiNTMyODBlNTI5ZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TQibyA1UyqStSygzkcaSanSn_gNpllL1nhb3qDKcA0o)
> 
> reaching the 2nd object in the slicer preview - it clearly shows the 2nd object, printing the "text" white colour first before the rest of the 2nd cap - in blue. ![image](https://private-user-images.githubusercontent.com/20792799/288258409-ee684f30-122e-4881-ab5c-fa9d76ee7c5e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMjcwNzIsIm5iZiI6MTcwNTAyNjc3MiwicGF0aCI6Ii8yMDc5Mjc5OS8yODgyNTg0MDktZWU2ODRmMzAtMTIyZS00ODgxLWFiNWMtZmE5ZDc2ZWU3YzVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDAyMzI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjM2RhMzllYTMzZmE2NzYzNDYyZjNhMTZhNmE2NTgxNTgzMGRlOGRmMmI3NDIzMmU0OWRiZWYzNzc2OWE4N2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lkw1MAQhtwFjDeS1QAOfaf0tPWWOWaMjRYObhwy9bWc)
> 
> This is what SHOULD happen - but this is a picture taken BY LAYER print - white is first etc. ![image](https://private-user-images.githubusercontent.com/20792799/288259073-b3ed296e-4c8f-40e9-8c2c-94cd3baa7b80.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMjcwNzIsIm5iZiI6MTcwNTAyNjc3MiwicGF0aCI6Ii8yMDc5Mjc5OS8yODgyNTkwNzMtYjNlZDI5NmUtNGM4Zi00MGU5LThjMmMtOTRjZDNiYWE3YjgwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDAyMzI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThjNWM1MmZhOWUyMDdmMjllZWI1ZGNhNWU2M2M1NWQxYWM4OGE3NjBhODFjYjNkNDUzZTJmNWQxYjA4YmQ2N2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.b2QkVzKobUgngPy_aDaEwhQ0cR0C5GkCkh1QNdc7kyE)
> 
> and as expected, both objects white text first (as layer mode) ![image](https://private-user-images.githubusercontent.com/20792799/288257917-1849059b-1a98-4111-a00c-c612851cd5fb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUwMjcwNzIsIm5iZiI6MTcwNTAyNjc3MiwicGF0aCI6Ii8yMDc5Mjc5OS8yODgyNTc5MTctMTg0OTA1OWItMWE5OC00MTExLWEwMGMtYzYxMjg1MWNkNWZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEyVDAyMzI1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM5ODhjODUyYTMwYTJjZmZlZWYwMDcxZTI0MDQyOGFhOTk0MjVkZTFiNmRhZWE3MzM1ZGFkMTQ3YjkxNTAzZWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TCKqA3MtFtvm0-LVH_6foNE5AmPO_wtXBlvFY7_r0tE)

@zhimin-zeng-bambulab Please have a look

glebonator, 2024-01-19:
Thank you for adding the first-layer ordering, but please, please consider doing the same for all other layers!

I will give a usecase that I'm struggling with: I have a mostly transparent case for some electronic gizmo, so layers 1&2 are transparent PCTG forming the outer skin. Then layers 3-9 are the rest of the case shell which includes an inlay with some logo/text. I would like to be able to choose to print the logo/text with the colored PETGs first, but can't, it somehow always chooses the transparent PCTG to go first, which then causes the logo to be fuzzy & text not sharp. The bigger the item, the less the effect, but sometimes small features are needed and choosing the filament color order would help fix it.

What's weird is that omehow BS randomly decides which color order is printed for each layer. I've read that it goes by larger area first but I only see it for layer 3, the first layer where the features appear. On other layers it's definitely random or going by some logic unknown to me) for each layer. 

Layer3, clear first, then the rest:
[image]

Layer4, white, blue first, then transparent, then pink
[image]

Playing around with the object ordering in a group doesn't work, the overall case has to be first or the logo/text get covered up. None of the other settings like changing inner/outer wall order worked. 

Have a per-plate filament order that is always followed FOR ALL LAYERS would greatly help, and going by all the countless forum and reddit threads that I see on the topic - this is an often requested feature. Thank you!

SaltWei, 2024-01-20:
Yes, please wait 1.9version and we will consider and add it.

BonkersBill, 2024-01-28:
I have a part with white and black in the first layer, black is the least (but not at all tiny) area.  When the black is finished, there is often a very fine string left dangling.  When white comes along next, the result is a gray streak.   In the art and printing world, the order is light to dark for this reason.  Please let us choose - there are good reasons for both orderings.

SaltWei, 2024-01-30:
> I have a part with white and black in the first layer, black is the least (but not at all tiny) area. When the black is finished, there is often a very fine string left dangling. When white comes [...]

The filament order of first layer has already been in current version. Can a per-plate filament order option "for all layers other than first" solve your issue?

BonkersBill, 2024-01-30:
Yes, a per-plate filament order option for all-layers-other-than-first would be a big help.

Thank you.

Bill Landolina – 678-596-3625 – ***@***.******@***.***>

zhimin-zeng-bambulab, 2024-02-02:
@ffafe5 @parrothd  @glebonator  @BonkersBill 
If you want to define the filament printing order of each layer, what form do you expect to appear in the interaction?
If an object has 100 layers, and the order of each layer can be defined, how big of a table is needed to accommodate this information?
Could you give us some suggestions to better deal with this problem? Thanks!

BonkersBill, 2024-02-02:
** My need is putting logo images on the bottom and top surfaces where my problem is dark filaments may leave traces that bleed into light areas if dark runs first.  I can use the same filament order throughout.

** The only feature I need is a single order for all layers.  If there is already a table for dealing with a custom order for the first layer, simply adding a checkbox "all layers" would take care of my need. **

If there really is a need to specify filament order for more complex situations, I suggest adding a layer modifers that can be added at any layer with a specification for number of layers above - typically there might be one applied at the bottom, and another applied near the top that is in effect for a few layers.  If somebody needs to put a modifier like that on every layer it would be possible - not easy, but possible.

Thanks!

[10 more replies]

closed 2024-03-26 by SaltWei