bambulab/BambuStudio #1818: Adding option of number of slow layers.

werewolfenn, 2023-05-18:
On some filaments, I am keeping my prints running at 50% for a number of layers before switching back to 100% speed.  I know I can edit the first layer speed and first infill to whatever speeds I want, but I'd also like to choose how many additional layers to go that speed.

bambu123, 2023-05-19:
You can use modifiers to fulfill this requirement.

https://help.prusa3d.com/article/modifiers_1767#height-range-modifier

werewolfenn, 2023-05-22:
> You can use modifiers to fulfill this requirement.
> 
> https://help.prusa3d.com/article/modifiers_1767#height-range-modifier

Perfect!   Not where I was thinking, but will certainly do the trick.  Thank you!

bambu123, 2023-05-23:
Glad to hear that!

closed 2023-05-23 by bambu123

werewolfenn, 2023-05-24:
The given option is really helpful, but thinking it over, it's not 100% ideal.  If I wish to do this with everything I print, I'd have to do this process every time.  A box I can check off, or something I could save as a preference in a saved profile would be best (for me).

man55, 2023-08-16:
One setting is missing a lot: the number of slow layers. I like to put 2 layers in most cases, like for example in Cura Slicer. Using modifiers every time is very inconvenient.
The fan by default turns on only from 4 layers. As a result, the 2nd and 3rd layers break, the speed is already too high, and there is no cooling yet. And it's too early to turn on the fan already on the second layer.
Damaged second layer in the photo. PETG.
[image]