bambulab/BambuStudio #1820: Camera stream does not work on Ubuntu

RobRobM, 2023-05-18:
**Describe the bug**
When starting the camera stream on Ubuntu I get an error message that the codecs are missing (see screenshot) even though all necessary codecs are installed.

**To Reproduce**
Press the play button in the device tab to start the camera feed and the error message appears.

**Expected behavior**
The camera stream should appear.

**Screenshots**
[image]

**Printer model**
Bambu Labs X1C

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 23.04

tburschka, 2023-05-28:
I don't get the message, but i also can't see any video feed. It's just saying "playing" but the screen keeps black.

Ubuntu 22.04.1
Kernel: 5.19.0-42-generic

Package: gstreamer1.0-libav
Architecture: amd64
Version: 1.20.1-1

Package: gstreamer1.0-plugins-bad
Architecture: amd64
Version: 1.20.1-1ubuntu2

bambu123, 2023-05-29:
@tburschka Could you please send us the logs in '~/.config/BambuStudio'?

tburschka, 2023-05-29:
@bambu123 via mail, to which address? I think this is the relevant log:

```
2023-05-29 08:25:27.370822[Thread 0x00007f3c67089980]:update_slice_print_status m_slice_select 1: can_slice= 1, can_print 1, enable_slice 1, enable_print 0 
2023-05-29 08:25:27.370835[Thread 0x00007f3c67089980]:update_background_process, Line 4532: exit, return_state=0
2023-05-29 08:25:27.370844[Thread 0x00007f3c67089980]:restart_background_process, Line 4570: not started
2023-05-29 08:25:27.511766[Thread 0x00007f27693f5640]:process_task{ type:Backup, id:0, path:, object:0, extra:3}
2023-05-29 08:25:27.524267[Thread 0x00007f3c67089980]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:3}
2023-05-29 08:25:37.511792[Thread 0x00007f27693f5640]:process_task{ type:Backup, id:0, path:, object:0, extra:4}
2023-05-29 08:25:37.512010[Thread 0x00007f3c67089980]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:4}
2023-05-29 08:25:47.511779[Thread 0x00007f27693f5640]:process_task{ type:Backup, id:0, path:, object:0, extra:5}
2023-05-29 08:25:47.512001[Thread 0x00007f3c67089980]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:5}
2023-05-29 08:25:57.511786[Thread 0x00007f27693f5640]:process_task{ type:Backup, id:0, path:, object:0, extra:6}
2023-05-29 08:25:57.512011[Thread 0x00007f3c67089980]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:6}
2023-05-29 08:26:05.178391[Thread 0x00007f3c67089980]:set_selected_machine=00M09A322500316
2023-05-29 08:26:05.178419[Thread 0x00007f3c67089980]:load_ams_list: ams 0 tray 0 id  color 
2023-05-29 08:26:05.178431[Thread 0x00007f3c67089980]:load_ams_list: ams 0 tray 1 id GFA05 color F72323FF
2023-05-29 08:26:05.178435[Thread 0x00007f3c67089980]:load_ams_list: ams 0 tray 2 id GFL99 color DCDCDCFF
2023-05-29 08:26:05.178438[Thread 0x00007f3c67089980]:load_ams_list: ams 0 tray 3 id  color 
2023-05-29 08:26:05.178441[Thread 0x00007f3c67089980]:load_ams_list: 4 items
2023-05-29 08:26:05.183344[Thread 0x00007f3c67089980]:command_get_version
2023-05-29 08:26:05.183506[Thread 0x00007f3c67089980]:MediaPlayCtrl switch machine: 00M09A322500316
2023-05-29 08:26:05.183524[Thread 0x00007f3c67089980]:MediaPlayCtrl::SetStatus: 
2023-05-29 08:26:05.188306[Thread 0x00007f3c67089980]:monitor: show_status = 2
2023-05-29 08:26:05.195180[Thread 0x00007f3c67089980]:monitor: change to sub task id = 14985852
2023-05-29 08:26:07.511771[Thread 0x00007f27693f5640]:process_task{ type:Backup, id:0, path:, object:0, extra:7}
2023-05-29 08:26:07.511866[Thread 0x00007f3c67089980]:process_ui_task{ type:Backup, id:0, path:, object:0, extra:7}
2023-05-29 08:26:07.866389[Thread 0x00007f3c67089980]:MediaPlayCtrl::SetStatus: Initializing...
2023-05-29 08:26:08.397046[Thread 0x00007f25f25fd640]:MediaPlayCtrl camera_url: bambu:///KTT2KE7N9HM6VHED111A?authkey=XXXXX&passwd=YYYYY&region=us&device=00M09A322500316&version=01.06.02.01, machine: 00M09A322500316
2023-05-29 08:26:08.397199[Thread 0x00007f3c67089980]:MediaPlayCtrl::SetStatus: Loading...
2023-05-29 08:26:08.397588[Thread 0x00007f27a9ffb640]:MediaPlayCtrl: start load
2023-05-29 08:26:10.070223[Thread 0x00007f27a9ffb640]:MediaPlayCtrl: end load
2023-05-29 08:26:10.070226[Thread 0x00007f3c67089980]:MediaPlayCtrl::onStateChanged: 0, last_state: 5
2023-05-29 08:26:10.070243[Thread 0x00007f3c67089980]:MediaPlayCtrl::onStateChanged: size: 1280x720
2023-05-29 08:26:10.070266[Thread 0x00007f3c67089980]:MediaPlayCtrl::SetStatus: Playing...
2023-05-29 08:26:10.070627[Thread 0x00007f3c67089980]:MediaPlayCtrl::onStateChanged: 0, last_state: 0
2023-05-29 08:26:10.070632[Thread 0x00007f3c67089980]:MediaPlayCtrl::onStateChanged: skip when task not finished
2023-05-29 08:26:10.071376[Thread 0x00007f3c67089980]:MediaPlayCtrl::onStateChanged: 0, last_state: 0
2023-05-29 08:26:10.338758[Thread 0x00007f3c67089980]:MediaPlayCtrl::onStateChanged: 2, last_state: 0
2023-05-29 08:26:17.511771[Thread 0x00007f27693f5640]:process_task{ type:Backup, id:0, path:, object:0, extra:8}cat: d: Datei oder Verzeichnis nicht gefunden
```

tburschka, 2023-05-30:
Mobile phone and on windows it works well. I've tested OBS Streaming and that works well: 

[image]

bambu123, 2023-06-07:
@RobRobM what's the output of 'gst-inspect-1.0 -a' in your envrioment?

taylor009, 2023-06-14:
I'm also seeing this error in Ubuntu 23.04

bambu123, 2023-06-14:
> what's the output of 'gst-inspect-1.0 -a' in your envrioment?

what's the output of 'gst-inspect-1.0 -a' in your envrioment?

bambu123, 2023-06-14:
> I'm also seeing this error in Ubuntu 23.04

what's the output of 'gst-inspect-1.0 -a' in your envrioment?

taylor009, 2023-06-14:
```
pipeline:   name                : The name of the object
pipeline:                         flags: readable, writable
pipeline:                         String. Default: "pipeline0"
pipeline:   parent              : The parent of the object
pipeline:                         flags: readable, writable
pipeline:                         Object of type "GstObject"
pipeline:   async-handling      : The bin will handle Asynchronous state changes
pipeline:                         flags: readable, writable
pipeline:                         Boolean. Default: false
pipeline:   message-forward     : Forwards all children messages
pipeline:                         flags: readable, writable
pipeline:                         Boolean. Default: false
pipeline:   delay               : Expected delay needed for elements to spin up to PLAYING in nanoseconds
pipeline:                         flags: readable, writable
pipeline:                         Unsigned Integer64. Range: 0 - 18446744073709551615 Default: 0 
pipeline:   auto-flush-bus      : Whether to automatically flush the pipeline's bus when going from READY into NULL state
pipeline:                         flags: readable, writable
pipeline:                         Boolean. Default: true
pipeline:   latency             : Latency to configure on the pipeline
pipeline:                         flags: readable, writable
pipeline:                         Unsigned Integer64. Range: 0 - 18446744073709551615 Default: 18446744073709551615 

Total count: 25 plugins, 253 features

```

bambu123, 2023-06-14:
> ```
> pipeline:   name                : The name of the object
> pipeline:                         flags: readable, writable
> pipeline:                         String. Default: "pipeline0"
> pipeline: [...]

It seems not completed. Please send all output of 'gst-inspect-1.0 -a' as an attachment.

bambu123, 2023-06-14:
> [output.txt](https://github.com/bambulab/BambuStudio/files/11743339/output.txt)

Can't find any 'h264' in the output.
Install gstreamer1.0-plugins-bad and retry.

taylor009, 2023-06-14:
@bambu123 that solved it for me.

tburschka, 2023-06-14:
any ideas what could be wrong on my system?

[out.txt](https://github.com/bambulab/BambuStudio/files/11746106/out.txt)

bambu123, 2023-06-15:
> any ideas what could be wrong on my system?
> 
> [out.txt](https://github.com/bambulab/BambuStudio/files/11746106/out.txt)

Seems all ok. could you provider full log files? Don't worry the passwd in the url is temperary.

RobRobM, 2023-06-15:
The camera is working again. In don't really know why or what changed, but all is ok now.

closed 2023-06-15 by RobRobM

bambu123, 2023-06-19:
> any ideas what could be wrong on my system?
> 
> [out.txt](https://github.com/bambulab/BambuStudio/files/11746106/out.txt)

Please set envriment 'GST_DEBUG=4' and run Studio from console.
Then more log messages relate to GST Video Player will be dumpped on console.
Send the logs to us.

tburschka, 2023-06-28:
[output-bambu-studio.txt](https://github.com/bambulab/BambuStudio/files/11890523/output-bambu-studio.txt)

tburschka, 2023-06-28:
Thanks to the outout i found the reason:

https://bugs.launchpad.net/ubuntu/+source/gstreamer-vaapi/+bug/1971463
the gstreamer1.0-vaapi seems to be broken on jammy (22.04). I removed it and the package and it worked, but it proccessed by the CPU...

I've followed this tutorial as much  as possible, but i'm not able to build a current/later version of gstreamer-vaapi on my own since they moved the repository (structure) - i beliieve during the takeover from microsoft of github).

So i've currently two options: see the video in the application but disable global usage of the gpu accelerated video or install the package and don't see it in bambu-studio----

bambu123, 2023-06-28:
Maybe we can disable GPU video sink from Bambu Studio.

tburschka, 2023-06-28:
im courious whats the difference of OBS, since it also seems to use the gpu if i use it stream the camera, at least `intel-gpu-top` shows usage....

bambu123, 2023-06-28:
Maybe we are not using a hardware window for video. The window is configured by gtk3.

tburschka, 2023-07-13:
Is there something i can do?

darkcruix, 2023-09-22:
I have a related issue when compiling Bambu Studio on Debian 12. Everything works fine with the exception of the video screen on the P1P.  I get a Loading failed [2].
The output of my gst-inspect-1.0 -a is attached as gst-inspect.txt
[gst-inspect.tar.gz](https://github.com/bambulab/BambuStudio/files/12704346/gst-inspect.tar.gz)

yozh1que, 2024-10-13:
Had the same issue on ubuntu 22.04. `sudo apt install gstreamer1.0-plugins-bad-apps` worked for me

RobRobM, 2024-10-17:
The camera works when I use nvidia prime-select on-demand, but it doesn't work when I select prime-select nvidia. It seems like the camera functions correctly when the system is in on-demand mode, but not when using the dedicated NVIDIA GPU exclusively.