bambulab/BambuStudio #1821: Plate 12 Overlap Issues

FurysSIN, 2023-05-18:
**Describe the bug**
When slicing a plate with several "flexible" files and a brim (any type), some sections will have the brim and the bottom layers overlap, causing the object to print incorrectly. I have checked that the file is completely flat by cutting off 0.2mm, reloading the file, and/or separating the object out to print individually. The only thing that has corrected the problem is not using that plate as I have noticed that it's strictly on Plate 12

**3mf File for This Bug**
Paid-for files have been cut to respect designers' IP. Cut does not affect issue. 
[Google Drive 3mf](https://drive.google.com/file/d/1jbGdM0djwhUvb03nInvEDHx8jnCQQTN-/view?usp=sharing)

**To Reproduce**
Load multi-part objects (as single objects) on Plate 12. Seems to need more than 5 parts to display the issue. Moving the object can cause the issue to not fully be apparent.

**Expected behavior**
The brim should not be under bottom layers...

**Screenshots**
Images will show one file, however I have had this issue with several different files. The only commonality is Plate 12.

Image showing the sliced preview. Back and front have the brim overlapped.
[image]

Image showing that the bottom layers are slicing as bottom layers and not large overhangs
[image]

Single Object on Plate 12
[image]

Object cloned and sliced correctly as a single item on plate 13
[image]

Bad slice as printed
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Home
 - Version: 10.0.19044

FurysSIN, 2023-05-20:
Plate 16 is also seeming to repeat the behavior of overlapping the brim and bottom layers. After more checking, if you have enough plates to for Bambu Studios to put 4 or 5 plates in a row, the 4th plate in the row is exhibiting the overlap behavior.

tangketan, 2023-05-29:
Thanks for reporting the issue. I've confirmed this is a bug with brim shifting. We'll fix it in the next version.

SaltWei, 2023-06-16:
Fixed in internal version already. Will be included in next user release! Thnaks

tangketan, 2023-08-02:
The latest Studio 1.7 has fixed this bug. Please take a look.

closed 2023-08-02 by tangketan