bambulab/BambuStudio #1822: Add ability to give names to Plates for organization

DMFirmy, 2023-05-18:
It would be very useful when working on larger projects to be able to rename each of the plates in your project to give them a name more useful than "Plate #". I also notice that when Bambu Studio sends a file to the printer, the file gets saved to the SD card of the printer as a file with the file name "plate#.gcode". It would be even better if this file name also reflected the name we assign for a given plate.

julie777, 2023-05-21:
@DMFirmy 
Duplicate of #1708 which has a lot of detail. Please add your comments there. Another opinion would be welcome.

Haidiye00, 2023-05-22:
[image]
[image]

Is this feature what you need? Please take a screenshot

DMFirmy, 2023-05-22:
Not exactly. I did find that feature , but it only applies for your prints, and the name isn't preserved as part of the project. As an example, I have a project with over a dozen plates, some of which have different versions of a same part for the left or right side of the assembly. It would be nice to be able to name the plates, for example left or right side, to easily differentiate between plates of similar models.

[img_v2_085c3c0b-003e-4f4b-9e6b-cd0327e92b4g]<https://user-images.githubusercontent.com/132528822/239785046-278f36e7-776d-49eb-83c6-a6a04292de75.jpg>
[image]<https://user-images.githubusercontent.com/132528822/239785349-cf5436e9-72b6-4a8f-827b-878990de61b8.png>

Haidiye00, 2023-05-22:
Can you draw a schematic diagram to show your requirements? Thank you

DMFirmy, 2023-05-22:
What I mean is that I would like to see the labels in the Object browser. I notice in my installation of Bambu Studio, the labels are not shown in place of the plate numbers on the build area as you have highlighted in your screenshot.  I did discover the “Show Labels” feature after reading your previous email, but see my second screenshot showing the plate numbers do not update.

Haidiye00, 2023-05-22:
My side is a newly developed feature that has not been opened to users as of now. May I ask if there is any need to supplement the development of the screenshot effect

DMFirmy, 2023-05-22:
Adding the labels in the object browser also would be wonderful, but the new feature you showed me is a very good idea. I appreciate your prompt feedback and response. I myself am a software developer, and always appreciate prompt and open communication with other development teams. Not too many are as responsive.

DMFirmy, 2023-05-22:
Yes

Can I close this issue now? Thank you~