bambulab/BambuStudio #1824: Auto Tree Supports no longer tree supports

Forty206, 2023-05-19:
**Describe the bug**
A model that was previously printed last week using auto tree supports is now generating a significantly larger structure that is simply regular supports but wrapped in a shell. Due to the size of this print, this new support structure has over doubled the filament usage for the entire print vs proper tree supports, from 300 and change grams to nearly 700 grams, and an extra 7 hours of print time

**3mf File for This Bug**
I am uncomfortable attaching the file as it is a product I intend to sell and I don't know how public this forum is

**To Reproduce**
Steps to reproduce the behavior:
Slice a model with a significant flat overhang. 
Type : Tree(auto)
Style: default
angle: 30
neither checked

**Expected behavior**
Tree supports. not regular supports with a wall around them

**Screenshots**
See File Section for reason there is no screenshot

**Desktop (please complete the following information):**
 - OS: windows
 - Version 1.6.2.4

TechnicalMacaroon, 2023-05-25:
Ran into this myself today, doubled filament usage and added 2hrs of print time. Didn't even think to check the supports initially as it's a hollow item and they are on the inside. For me it's a AA to D battery adapter so small flat area and using Tree(auto) with either Default or Hybrid style.
Slim and Strong styles appear to work normally.

tangketan, 2023-05-29:
Yes, we have changed tree(auto) to tree(hybrid). Previously tree(auto)=tree(slim), but we received multiple reports that tree(slim) breaks when getting tall. As a successful print is our top priority, we changed the default style. I'm sorry for the trouble caused. We'll try to improve on that, to make tree(auto) strong enough and also cost less filaments.

TechnicalMacaroon, 2023-05-29:
Perhaps adding a tooltip pop up message when you change feature like that would be helpful. Similar to how you show the message regarding the interface height when you try to use tree slim or tree heavy. Without any sort of notification or reading release notes, which lets be honest the majority of people don’t do, your average user won’t know when something like this changes. Sent from my iPhone

tangketan, 2023-05-30:
Yes, you're right. We can use that tooltip in the future if we have similar changes.

closed 2023-08-02 by tangketan