bambulab/BambuStudio #1827: Top Z distance and Bottom Z distance inaccurate with tree supports

LordBoos, 2023-05-19:
**Describe the bug**
Top Z distance and Bottom Z distance option for supports doesn't work correctly for tree supports. For normal supports Top Z distance and Bottom Z distance works as expected.

**3mf File for This Bug**
Not relevant, happens with all models.

**To Reproduce**
Steps to reproduce the behavior:
1. Open any model applicable for support use.
2. Enable tree supports.
3. Set "Top Z distance" option to some number lesser than layer height. Eg. 0.1mm with 0.2mm layer height.
4. Slice the model.
5. See that the Top Z distance on sliced Preview is actually 0.2mm and not 0.1mm as configured.

**Expected behavior**
I expect that whatever I set Top Z distance to it will actually be used.

**Screenshots**
Tree supports, you can clearly see, that the Top Z distance is the same height as a layer, so 0.2mm, but Top Z distance is set to 0.1mm:
[image]
Normal supports, you can see, that the Top Z distance is half the layer height, which is correct and expected, because Top Z distance is set to 0.1mm:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 16.2.4

tangketan, 2023-05-22:
This is an old problem with tree support. The top z distance is quantized as a multiple of layer height, and it's not quite easy to set an arbitrary value for it. I'll try to fix it, but not very soon.

toddanglin, 2023-08-08:
I want to add some extra attention to this bug since it is **not** a behavior exhibited by OrcaSlicer (as of 1.6.4).

Within Bambu Studio, if Support Type is set to "tree(auto)", it appears the "Top Z distance" modifier is essentially ignored. Changing this distance has no impact on the produced support distance UNLESS you set it to "0" (then the output is changed). This is true even when "normal" supports are produced in "tree(auto)" mode.

Meanwhile, in OrcaSlicer, with the same "tree(auto)" support mode, changing the "Top Z distance" does correctly change the interface height for supports.

This issue caused problems for me as I was slicing the same object in OrcaSlicer (for my X1Cs) and Bambu Studio (for my P1S) and getting very different support interface behavior. After a lot of debugging, I realized Bambu Studio was not correctly adjusting Top Z distance for support.

The workaround for now (which is acceptable in my case) is to switch support type to "normal(auto)". Then, Bambu Studio correctly calculates the Top Z distance. Hope this helps someone getting confused/frustrated by Top Z distance problems in Studio.

toddanglin, 2023-08-20:
@tangketan Given what you shared about how top z distance is calculated for tree supports, is there a trick to reduce the z distance if it's producing poor print results? Or if you need finer control over z distance to get good results, is the only option to switch to "normal" supports?

titaniumschmapple, 2023-09-12:
@tangketan DEFINITELY need a fix for this asap please!

I kept checking that it wasn't me when I was printing with Bambu Studio and despite changing the Top Z Distance dramatically (0.2mm to 0.02mm) there was no change after slicing.

I am trying to print thin walls and NEED to be able to print at a lower Top Z Distance because atm the Bambu layers are getting dragged everywhere because the surface isn't close enough to the support!

@toddanglin any updates on this using tree supports? Does PrusaSlicer do the same thing?

toddanglin, 2023-09-12:
@tangketan Unfortunately, the situation "got worse" because OrcaSlicer now exhibits the same behavior (as of 1.6.4). It makes sense if Orca merged the code from Bambu Studio, but disappointing. Hopefully, someone with more knowledge of the internals of these slicers can restore the Top Z Distance behavior for tree supports in Orca 1.6.3.

Meanwhile, the only sustainable workaround I've got is to use "normal" support mode when I need more precise control over Top Z Distance. Far from ideal, but it mostly works.

I've not spent time deeply comparing in PrusaSlicer, but I wouldn't be surprised if this behavior is in the core.

toddanglin, 2023-09-16:
A follow-up to this issue...

I finally spent time combing over this issue because I really wanted "tree(auto)" supports to work better in newer versions of OrcaSlicer. I documented a lot of my findings in this OrcaSlicer issue: https://github.com/SoftFever/OrcaSlicer/issues/2139

In short, I can get Top Z Distance behavior to work the same in the latest versions OrcaSlicer by unchecking the "Independent support layer height" checkbox under Support settings. Once unchecked, it produces the same support offsets as 1.6.3. (Which, to be clear, are still less controllable than Top Z distance for "normal" supports...I assume for the reasons @tangketan suggested.)

I also checked the latest version of Bambu Slicer and the same applies. Hope that helps.

lesi123, 2024-05-23:
I'm experiencing the same issue. Top Z Distance is ignored on any file I slice with tree(auto) supports. It doesn't matter if it's set to 0.2 or 2.0, it never changes past 1 layer worth of gap.