bambulab/BambuStudio #1828: Blank pages and impossible to Login on Linux

GuillaumeLeclerc, 2023-05-20:
I'm on Linux latest release and it is impossible to login/see the home page.

[image]

lanewei120, 2023-05-22:
@GuillaumeLeclerc 

hello, thanks for reporting this issue
which Linux Os and appimage are you using?
is it a 100% reproduciable issue or a random issue/

GuillaumeLeclerc, 2023-05-22:
The latest Ubuntu release. Yeah it's always happening I can't print anything

bambu123, 2023-05-23:
@GuillaumeLeclerc 
Please send me the app logs which can be located at ’~/.config/BambuStudio/log‘.

replicant0wnz, 2023-05-23:
I'm getting the same errors under Pop_OS 22.04 (Ubuntu derivative). I've attached the above log file you requested. 
[debug_Mon_May_22_23_26_53_2598948.log.zip](https://github.com/bambulab/BambuStudio/files/11539584/debug_Mon_May_22_23_26_53_2598948.log.zip)

replicant0wnz, 2023-05-23:
It's a permissions issue. When it mounts the filesystem under /tmp.mount_Bambu_XXXX the files are owned by root. Executing as root allows it to run.

GuillaumeLeclerc, 2023-05-23:
It's not a solution:

1. You should not run a program as root that has no reason to be root
(especially if you can't audit it completely like the network plugin)
2. Firefox (and righfully so will refuse to run as root) as a result one
can't login with Google since it will open a new window for you to login in
your default browser). Moreover, there is no way on Bambulab website to
convert a Google account to a email+password one

bambu123, 2023-05-23:
It is always mounted as a root. I don't think it relates to a root account.
@GuillaumeLeclerc 
Do you have any logs? Does it run as root in your system?

GuillaumeLeclerc, 2023-05-23:
Yes it runs as root on my system too but I can't login because browsers
refuse to run as root

replicant0wnz, 2023-05-23:
> 

I didn't say it was a solution now did I? I said it runs as root and it appears to be a permissions issue.

replicant0wnz, 2023-05-23:
The following fixed it for me:

```bash
~ λ ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage --appimage-portable-home 
~ λ mkdir Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage.home/.confg
~ λ mv ~/.config/BambuStudio Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage.home/.confg
```

replicant0wnz, 2023-05-23:
I also had to copy the `plugins` directory from `/root/.config/BambuStudio` in order to get the network plugin to work. It also segfaulted the 1st couple of times. It's running now though and was able to send a print and view it via the camera.

GuillaumeLeclerc, 2023-05-24:
That helped thanks @replicant0wnz . I do get segfaults during the visualization though. At least I can start jobs!

bambu123, 2023-05-24:
> The following fixed it for me:
> 
> ```shell
> ~ λ ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage --appimage-portable-home 
> ~ λ mkdir Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage.home/.confg
> ~ λ mv ~/.config/BambuStudio Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage.home/.confg
> ```

Great!

GuillaumeLeclerc, 2023-05-24:
Any way to get the camera to not segfault. There is nothing in the logs other than `segfault` :/

lanewei120, 2023-05-24:
> Any way to get the camera to not segfault. There is nothing in the logs other than `segfault` :/

could you check your libjpeg version used by your gstreamer?
just refer to this https://github.com/bambulab/BambuStudio/issues/1814

GuillaumeLeclerc, 2023-05-24:
Isn't the purpose of AppImages to include all the dependencies so that this doesn't happen ?

lanewei120, 2023-05-24:
currently gstreamer is a big one and not built into the appimage
so the appimage can not work in this case

GuillaumeLeclerc, 2023-05-26:
What about an appImage that actually contains everything even if it's large ? I'm sure most people don't care about size and just want something that works.

replicant0wnz, 2023-05-26:
I'm gonna take a stab at creating a .deb file as I prefer them to appimages next week if I find the time. Just been running it on my Windows laptop for now unfortunately ..

GuillaumeLeclerc, 2023-05-26:
Unfortunately won't work on Archlinux :/ Hopefully support from BambuLab will improve in the future

bambu123, 2023-06-05:
@GuillaumeLeclerc 
What is the version of the libwebkit2gtk in your system?

harley-codes, 2023-10-13:
> I'm gonna take a stab at creating a .deb file as I prefer them to appimages next week if I find the time. Just been running it on my Windows laptop for now unfortunately ..

Good man! I wanted to try to build one too, but as a flatpak.
Got stuck here -> https://github.com/bambulab/BambuStudio/issues/2606 
Just giving you the heads-up in case you hit a similar issue. 
Don't want to modify the files to fix the issue, just get a pipeline going where I can make builds from latest release.

AskAlice, 2023-10-18:
```WEBKIT_DISABLE_COMPOSITING_MODE=1 ./Bambu_Studio_Linux_Fedora_v01.07.07.89-20231013104019.AppImage```
fixed it for me

harley-codes, 2023-10-21:
Thank you, @AskAlice. For anyone who was unable to login via email, due to nothing happening after clicking login. It appears the session is established, but some redirect it not working. To work around this. Just attempt login, click forgot password, then click back. You should see the correct window after that.

lanewei120, 2024-08-08:
already included WEBKIT_DISABLE_COMPOSITING_MODE=1 by default

closed 2024-08-08 by lanewei120