bambulab/BambuStudio #1832: Add currently selected printer name to Printer Sidebar and last synch

BryanHaven, 2023-05-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lanewei120, 2023-07-18:
@MadMax389 
hello, how many printers are you using at the meantime?
will you use these printers to print the same task at a time?

MadMax389, 2023-07-18:
I have 2 printers: X1C w/ 2 AMS and a P1P w/ 1AMS. I do run them at the same time, but not typically on the same plate. Sometimes different plates from the same project. I used to run two instances of Studio, one for each printer, but that is problematic having the same project open in both instances. I just switch back and forth in one instance of Studio. When I switch printer profiles in the slicer, I also have to switch them on the device page to sync the AMS's. A "select printer" option from the sync button in the slicer would be ideal!

BryanHaven, 2023-07-18:
I’m running two (X1C w/ 2 AMS) frequently at the same time with plates from the same project. Though it does cause issues so I run separate instances of BambuStudio, one for each printer. Sent from my iPhone