bambulab/BambuStudio #1833: DLLs and system profiles are updated with no notification

julie777, 2023-05-21:
**Describe the bug**

I just happened to be looking at the debug log to diagnose a problem and I noticed entries about updating system profiles. It seems that when system profiles are updated that they also get copied to the AppData directory replacing those already there.

Knowing when something is being updated is important. It helps to determine the cause for changes in operation, or bugs that appear.

It is even more important if the update is related to a firmware update and it changes the user customizable G code sections. It is possible that this kind of change could cause damage to the printer, if the user had no notice and the G code interpretation changed.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows]
 - Version 1.6.2.4

lanewei120, 2023-05-22:
hello

the system profiles currently is updated with the BambuStudio together
if the version of Bambu Studio changes from v1.5 to v1.6
then the system profiles will also be updated to v1.6, and will be copied to app data

and the system profiles is consistent with bambustudio in the same version
we will make sure the system profile correct when published

and we can find the system profile difference between these two version on github's commit history

julie777, 2023-05-25:
Regarding the system profiles, I think what made me think they had been updated was a combination of log entries that were about updating profiles, and sycing, and the timestamps on the system profile files in the AppData/Roaming/BambuStudio/system directory. My last update of Bambu Studio was 2023-04-26, but I have many system profiles that were copied to the AppData directory dated 2023-05-04.

I know that I saw this copying in the log file, so maybe that is part of my misunderstanding.

The DLL updates are a different issue and they have been updated without any notification. During install there is a notification, but in normal use when the DLLs are updated I do not recall seeing any notification.

lanewei120, 2023-05-26:
> ue and they have been updated without any notification. During install there is a notification, but in normal use when the DLLs are updated I do not recall seeing any notification.

the dll updating also has logs in the log file,
to improve the user experience, we don't want to popup a notification to let the user do more confirm

lanewei120, 2023-06-09:
there is a popup window when updating the network plugins at the bootup
[image]

lanewei120, 2023-09-06:
close this issue as above comment

closed 2023-09-06 by lanewei120