bambulab/BambuStudio #1834: Wrong decimal seprarator in German Language (Swiss-Settings)

TomtheRom, 2023-05-21:
**Describe the bug**
Working with Keyboard and Windows Settings "German (Switzerland)" I am forced to use tha comma as a decimal separator and not the system setting (which is a decimal point like in US-settings).

**To Reproduce**
Steps to reproduce the behavior:
1. Start Bambu Studio on a System with Swiss-German System Settings
2. Try to pur any decimal number in an input field by using the number field on the keyboard and use the decimal separetor in the same aerea ot the keyboard
3. The decimal separator gets ignored and any decimal number is enterd as an integer

**Expected behavior**
Use the correct decimal separator which is set in the System

**Desktop (please complete the following information):**
 -  iOS, Windows 11
 - bAMBU sTUDIEO 1.6.2.4

bambu123, 2023-05-29:
Sorry, we don't support language sub region now. So it is a feature request, not a bug.

TomtheRom, 2023-05-29:
Sorry, but that's a lame answer.

If this is the case,  then at least mention itjin the Wiki or the
manual....

The only choice we have as usef is Language and Region.
If you are not able or don't want to integrate proper region settings which
are common in DACH,  then at least you have to mention it.....

And sorry for beeing direct.... but I am in customer service as well and I
am vefy aware of the issues witg regional settings.

I you truly concern about customers needs and questions please care about
what youmget to know.

If you need some support in testing your adjustments I would volounteer to
do Beta tests.....

bambu123, 2023-05-30:
Ok, let me try.

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.