bambulab/BambuStudio #1838: Add better support for managing settings applied by scope

julie777, 2023-05-21:
## Is your feature request related to a problem?

(I am still struggling with the change from "print settings" to "process" coming from Prusa Slicer. Actually those settings are a mix of print, and slice settings based on when they are applied.)

There are multiple scopes in the slicer, and the addition of multiple plates extended that list from what many of us were used to. It also adds a complication which I am hoping can be simplified.

These are the possible scopes that can be used to apply a setting.
  - Project Scope - actions and settings apply to the entire project, which includes every plate and every object and part.
  - Workspace scope - actions apply to all plates and objects shown in the workspace. This is really only applicable to workspace manipulations and selecting what to print. 
  - Selected Plates Scope - actions apply to all plates selected, including everything on them. (Currently this is mentioned in tool-tips, but give there is no way to select multiple plates, it is not a usable scope.)
  - Plate Scope - actions apply to the plate and all objects on the plate
  - Selected Objects Scope - multiple objects are selected and actions apply to all selected objects
  - plate subset objects scope
  - Object Scope - one object is selected and actions apply to that object
  - part scope
  - partial object scope
  - (I am not sure how assembly fits in)

The purpose of this request is to help
  1. clarify what scopes exist and are useful
  2. which scopes need work because they are missing part of all of the implementation
  3. which scopes should be eliminated
  4. How to apply and view settings by scope

### Project Scope

Project scope is the most obvious, simplest to use and most commonly used in a project. Most projects probably only use project scope. Most projects, at least currently, probably only have one plate.

For a project choosing settings should consist of selecting
  - a printer model
  - a printer nozzle
  - a (default) plate type
  - a (default) filament profile
  - a process profile

You will notice that I separated the printer model from the nozzle profile. Currently they are smashed together, but I believe that is causing a problem of choices. If one assumes that a user might have ten printers, 5 of one model and 5 of another model, and some of each with different nozzle sizes installed, then the printer profile list could potentially have dozens of entries. By selecting printer model separately the list of printer configurations need only contain those specific to that model. 
In fact, determining which printer can handle a given print job consists of matching model, nozzle, plate type.

### Selected Plates Scope

Both Plates and Plate scope are not implemented but I believe they are very useful and can replace some of the ways that have been used in the past to group objects. In any reasonably sized project with multiple plates there are likely to be parts that should have different filament types, different nozzle sizes, different settings for quality, etc. Plates is the magic that allows an entire project with dozens or hundreds of parts to be organized in ways that make managing and printing so much simpler.

Plates scope is not yet implemented, but it is a very useful scope. Some use cases include:
  - Selecting 3 of 10 plates in the project and slicing them and printing them.
  - selecting two of the plates in the project and setting a different filament profile for those two plates

### Plate Scope

This is even more useful and needed immediately than the above.

Being able to select a plate in the browser/object list and act on the plate as a unit instead of requiring selecting all the objects on the plate has a huge benefit and there is an extremely strong motivation for doing it by plate.

Currently Project scope is implicit in the sidebar. One must switch to object view to see the list of objects. Then it is possible to change settings for the objects. However, one must change settings individually. It is not possible to change the printer or nozzle profile for example. This is where both implementing plate scope, and changing the layout of the **browser can make a huge different and improve the state of the art in 3D printing**.

Separating object onto individual plates to allow more object than fit on one plate to be added to a project is just the tip of the iceberg. It is not ground breaking. It is only marginally useful. It does allow a project with common parts and multiple sets of optional parts to be well organized. But, it does not support the most important aspect of a large project - some parts need a completely different printing process. That might even require a different kind of printer, perhaps not even FDM.

I would restructure the browser to include the project at the top of the hierarchy. Then I would move what are project only choices currently to be set for the project, and inherited as they are now, but changed at the plate level.

Scope Hierarchy
  - Project
         - printer
         - nozzle
         - bed
         - filament type
         - color
         - process
     - plate 1
             - printer
             - nozzle
             - bed
             - filament type
             - color
             - process
         - object 1
         - object 2

Process Settings
   as current

### Selected Object Scope

Currently this allows selecting object from multiple plates and changing settings. This even allows select part of the objects on a plate. This should be removed. It is very confusing organizationally and there is no need to support it. Selecting entire plates, or a subset of objects on a single plate is manageable and sufficient.

### Selected Objects on a Single Plate

This can be useful and is supported by most if not all slicers. It can only be used to change a subset of the settings. For example the printer, nozzle, and plate type must all must be the same for a plate.

For the purpose of settings, selecting a single object is still the selected objects scope.

It could be possible to select different filaments per object, and even process profiles. Being able to actually choose using profiles instead of requiring modifying individual settings would be a benefit. This would add the filament type, color, and process profile drop down under each object.

It can also be argued that instead of having the complication of selecting presets per object selection that if completely different presets are to be used then the objects should be moved to separate plates.

### Summary

Providing better identification of scopes that can be used, and making the plate a scope with the same ability to select all types of preset profiles as the project, will greatly improve usability and efficiency. In addition it will **allow what was previously impossible** - having one project that uses completely different printers for some of the objects to be printed.

Only the following scopes need to be supported for settings
  - project
  - plate
  - selection group (subset of a plate)

Project and plate allow selecting by preset for printer, nozzle, filament, color, process.
Selection Groups only allows modifications of individual settings, but not changing any profiles.

The browser sidebar can be modified to make project scope part of the hierarchy.

zarthcode, 2023-10-04:
Just chiming in to give an anecdote.  I have a project that spans 9+ plates, and growing.  Different size parts, materials, and profiles.  I would like to use "fine" for some parts, and draft for others.  I avoid using the per-object settings unless required.  But it's not currently possible to apply settings to the entire plate, or choose a default profile for a plate.  Instead, I have to maintain multiple project files.  That makes the assembly-view useless (it can demonstrate glaring mistakes).

Additionally, a bulk "reload all from disk" would be helpful.

powershellwhizz, 2024-10-16:
I too would like this feature per plate. It is such a pain havin to set different print profiles per object. What is the status on this? I noticed this has been linked to another issue which is now "closed" but the linked issue seems to be talking about updating print profies in general and not having them associated per plate. I'd have thought that such an imporant feature req would have been added by now.

Any progress?