bambulab/BambuStudio #1840: Lose all custom presets when there is a Bamby Cloud outage

AdrianGarside, 2023-05-21:
1. Get one of the frequent cloud outages.
2. Switch to lan mode to avoid being blocked.
BUG:
All custom presets have vanished from Bambu Studio. This is especially annoying with a 0.6mm nozzle since there is only one system present.
EXPECT:
Custom settings I created locally should not vanish just because your cloud went down again. Cloud sync is fine but cloud-only for locally created settings isn't.

sanford, 2023-05-22:
Hey Bambu Team, how about a local cache of presets and any other cloud info?

psiberfunk, 2023-05-22:
There IS a local cache  in your appdata folder.  Back up your computers just in case .  When the “Bamby” cloud goes down, it burns the whole preset forest with it .

bambu123, 2023-05-22:
You should turn on this switch to see your custom presets.
[image]

They may be in the directory 'user/default' or 'user/<userid>' under the configuration directory; which one is used depends on if you have logged in.
So please keep login or logoff, not mix use them.

psiberfunk, 2023-05-22:
> You should turn on this switch to see your custom presets. [image]
> 
> They may be in the directory 'user/default' or 'user/' under the configuration directory; which one is used depends on if you have logged in. So please keep login or logoff, not mix use them.

Or maybe don’t have your cloud sync code blow away the local sync folder when the cloud is malfunctioning?

bani6809, 2023-05-22:
> You should turn on this switch to see your custom presets. [image]
> 
> They may be in the directory 'user/default' or 'user/' under the configuration directory; which one is used depends on if you have logged in. So please keep login or logoff, not mix use them.

why would we ever want to use login if we are going to lose presets when bambu cloud is down? you are making a very strong case for using LAN only mode and never using bambu cloud, ever.

psiberfunk, 2023-05-22:
> why would we ever want to use login if we are going to lose presets when bambu cloud is down? you are making a very strong case for using LAN only mode and never using bambu cloud, ever.

They've made LAN mode suck hard in a few ways.  It's a real pain to switch back and forth and you inexplicably lose some features.  It almost feels like that's on purpose.. but that's a different discussion.

bambu123, 2023-05-22:
> 

The preset files in the local cache folder should not be deleted unless deleted by the user.
The most possibility is not being loaded by Studio because of not opening the sync switch.

We'll modify the effect of the sync switch in the next version. Let it load the login user presets even if the sync switch is not open.

bambu123, 2023-05-22:
> > > You should turn on this switch to see your custom presets. [image]
> > > They may be in the directory 'user/default' or 'user/' under the configuration directory; which one is used depends on if you have logged in. So please keep login or logoff, not mix use them.
> > 
> > 
> > why would we ever want to use login if we are going to lose presets when bambu cloud is down? you are making a very strong case for using LAN only mode and never using bambu cloud, ever.
> 
> They've made LAN mode suck hard in a few ways. It's a real pain to switch back and forth and you inexplicably lose some features. It almost feels like that's on purpose.. but that's a different discussion... like how they refuse to let people do offline firmware updates via SD card..

By design, the presets will be loaded even if the network is completely down.
Please help check if it is deleted from the cache folder. The folder can be reached from the menu 'Help > Open Configuration Folder'.

mathewsamueldesign, 2023-05-23:
> > > > > > You should turn on this switch to see your custom presets. [image]
> > > > > > They may be in the directory 'user/default' or 'user/' under the configuration directory; which one is used depends on if you have logged in. So please keep login or logoff, not mix use them.
> > > > > 
> > > > > 
> > > > > why would we ever want to use login if we are going to lose presets when bambu cloud is down? you are making a very strong case for using LAN only mode and never using bambu cloud, ever.
> > > > 
> > > > 
> > > > They've made LAN mode suck hard in a few ways. It's a real pain to switch back and forth and you inexplicably lose some features. It almost feels like that's on purpose.. but that's a different discussion... like how they refuse to let people do offline firmware updates via SD card..
> > > 
> > > 
> > > By design, the presets will be loaded even if the network is completely down. Please help check if it is deleted from the cache folder. The folder can be reached from the menu 'Help > Open Configuration Folder'.
> > 
> > 
> > Hey, I checked the config folder and the presets are still there. They're not in the default folder though. It's in a sub folder with a long number on it. I switch to LAN mode when the cloud service is down, but my filament presents don't show up.
> > [image]
> 
> Then I think you haven't turned on the sync switch above. Please turn on it and see what happens. Since your custom presets are filament presets, you should choose a compatible printer or they may be filter-out by compatibility.

Hey, I do have sync turned on already.
[image]

bambu123, 2023-05-23:
@mathewsamueldesign 
Then it must be a preset compatible problem.
What printer preset are you using?
Please send the six files in the user-id folder if you like. I can check the compatibility then.

mathewsamueldesign, 2023-05-23:
> @mathewsamueldesign Then it must be a preset compatible problem. What printer preset are you using? Please send the six files in the user-id folder if you like. I can check the compatibility then.

Here ya go:
[filament.zip](https://github.com/bambulab/BambuStudio/files/11546769/filament.zip)

bambu123, 2023-05-24:
> > @mathewsamueldesign Then it must be a preset compatible problem. What printer preset are you using? Please send the six files in the user-id folder if you like. I can check the compatibility then.
> 
> Here ya go: [filament.zip](https://github.com/bambulab/BambuStudio/files/11546769/filament.zip)

Your filament presets are only compatible with printer X1C, not P1P.
Please select printer inherits from X1C.
It is not related to the LAN mode of the printer or the Bambu Cloud. Once login, the presets are loaded.

mattkerrison, 2023-05-25:
No I can confirm I have the same problem. Whenever the Bambu cloud is inaccessible, all my presets disappear.

I have that sync setting ticked, but with presets disappearing, I can't even export gcode to a sd card and walk it over to the machine in an outage. 

Please update your code to ensure that it doesn't disable / wipe out the presets in a cloud outage, it renders the printer useless.

mathewsamueldesign, 2023-05-25:
> Your filament presets are only compatible with printer X1C, not P1P. Please select printer inherits from X1C. It is not related to the LAN mode of the printer or the Bambu Cloud. Once login, the [...]

I don't have a P1P. I have an X1C. I have the X1C selected also in my printer settings. Everything syncs fine with Bambulab Cloud is working.

mattkerrison, 2023-05-25:
> I don't have a P1P. I have an X1C. I have the X1C selected also in my printer settings. Everything syncs fine with Bambulab Cloud is working. Why does it lose all the presets only when Bambulab Cloud is down?

+1

bambu123, 2023-05-26:
I see. The user presets can not be used without login. This is a mechanism for user identification.
You can make the presets default (not related to the user) by importing them when you have not login.

bambu123, 2023-05-29:
Found a bug relate to this. I will fix it in the next version.

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.