bambulab/BambuStudio #1841: "Can't Connect to Internet" misleads user when Bambu Cloud is down

nleskiw, 2023-05-21:
**Describe the bug**
During a cloud services outage, the X1C incorrectly reports that it cannot connect to the internet. 

**To Reproduce**
Steps to reproduce the behavior:
1. Block access to Bambu Cloud services via a firewall.
2. Wait for the "Can't connect to internet" error to appear on the printer. 

**Expected behavior**
1. The printer should detect that internet connectivity is available, but Bambu Cloud is down. 
2. The printer should suggest to switch to LAN Only mode if a user cannot wait for Bambu Cloud to return.

**Suggestions:**
Internet connectivity should use both Bambu Cloud IPs and multiple unaffiliated IPs that are unlikely to be down or change which respond to ICMP Echo requests (aka "ping"), such as:

Cloudflare: 1.1.1.1 and 1.0.0.1
google-public-dns-a.google.com. 	8.8.8.8
google-public-dns-b.google.com 	8.8.4.4

 - OS: Windows 10 Pro
 - Version: 21H2
 - Bambu Studio Version: 1.6.2.4

walterwongbbl, 2023-05-29:
Hi.
We will modify the error message when unable to connect to the Bambu server.

Thank you very much for your suggestion. I will provide feedback to the cloud department on this suggestion

walterwongbbl, 2023-05-31:
@sodachen22  
Please confirm this feature

vic1707, 2024-11-30:
Can we have updates on this ?
My printer shows "Failed to connect to the internet" but the internet test in the network tab succeeds 100% of the time and binding the printer to the account also works (my username shows after scanning the qr code)....
I believe this is an instance of this misleading case. 
Please fix this I spent way too much time trying to understand.