bambulab/BambuStudio #1842: "Detect narrow internal solid infill" results in lots of gaps and weaker parts

EvaFPV, 2023-05-22:
Discovered this with OrcaSlicer but as far as I can tell it's present here too. 

**Describe the bug**
The concentric infill from "Detect narrow internal solid infill" being on leaves lots of gaps in both Arachne and Classic modes resulting in a much weaker part. Somewhat visible in sliced preview, very visible in printed part.

**3mf File for This Bug**
[DetectNarrowInternalSolidInfill-bug.zip](https://github.com/bambulab/BambuStudio/files/11525972/DetectNarrowInternalSolidInfill-bug.zip)

**To Reproduce**
Slice a model with small areas to be filled with internal solid infill. Compare with setting on and off.

**Expected behavior**
On or off, solid infill should fill the area leaving no gaps (pinholes maybe but no large gaps)

**Screenshots**
[image]
[image]

**Printer model**
Voron V0.1 with a 0.5mm CHT nozzle

**Desktop (please complete the following information):**
 - OS: Windows 10
 - OrcaSlicer Version 1.6.2

SaltWei, 2023-05-23:
@QingZhangBambu Please use bambu printer to check

QingZhangBambu, 2023-05-29:
good point. We will investigate it

martian099, 2023-11-11:
Yeah I have the same issue on Bambulab P1S with BambuStudio. I think the issue is for some internal infills in top surfaces right above the infills, this setting will cause those layers to print in concentric pattern instead of the original rectilinear pattern, leading to a lot of bridges. But the bridges still prints at high speed set for internal infills at 250mm/s, it causes the bridges to fail and curl up, leading to the holes. Please find some smart ways to fix this!

martian099, 2023-11-11:
I guess one way is to detect if it's causing bridging to happen after changing the solid infill pattern from rectilinear to concentric, and if there is then slow down the printing speed to bridging speed. That should solve the problem.