bambulab/BambuStudio #1843: OpenSSL in Network-plugin is rapidly approaching EOL (2023-09-11)

deftdawg, 2023-05-22:
Current version of OpenSSL used by the network-plugin will no longer be updated after 2023-09-11.

```
strings ~/.config/BambuStudio/plugins/libbambu_networking.so | grep -i openssl | grep 20
OpenSSL 1.1.1k  25 Mar 2021
```

Source: https://www.openssl.org/blog/blog/2023/03/28/1.1.1-EOL/

> OpenSSL 1.1.1 was released on 11th September 2018, and so it will be considered EOL on 11th September 2023. It will no longer be receiving publicly available security fixes after that date.

related: https://github.com/NixOS/nixpkgs/pull/206495

lanewei120, 2023-05-22:
Hello

thanks for the information,

we will update it in the following versions

norpol, 2023-06-16:
New blog posting by OpenSSL: https://www.openssl.org/blog/blog/2023/06/15/1.1.1-EOL-Reminder/ including a link to their migration guide https://www.openssl.org/docs/man3.1/man7/migration_guide.html

deftdawg, 2023-08-11:
FYI this issue of deprecated OpenSSL is now blocking BambuStudio from being merged and distributed as a Nix Package (Nix & NixOS).

 see https://github.com/NixOS/nixpkgs/pull/206495#issuecomment-1674367932

lanewei120, 2023-08-15:
@deftdawg

thanks for reminder,
it is scheduled in v1.8, we will submit the codes as soon as we finished the testing

deftdawg, 2023-09-23:
The release notes for https://github.com/bambulab/BambuStudio/releases/tag/v01.07.06.92 suggest OpenSSL has been updated to 3.1, if that's the case this issue is resolved.

Is there a new version of the network plugin available on Linux that has this?  I don't (yet) have a Linux build of 1.07.06.92 to confirm it.

lanewei120, 2023-09-23:
> The release notes for https://github.com/bambulab/BambuStudio/releases/tag/v01.07.06.92 suggest OpenSSL has been updated to 3.1, if that's the case this issue is resolved.
>
> Is there a new version of [...]

yes, it should be already included in the network plugin

thanks

closed 2023-09-23 by deftdawg

lanewei120, 2023-12-18:
Hi @deftdawg 

after upgrade openssl to 3.1, we found the following function can not success on OpenSUSE
SSL_CTX_new(TLS_client_method());

it can success by using openssl 1.1.1
and it can success on windows/macos/ubuntu/fedora

any ideas?