bambulab/BambuStudio #1844: Incorrect color order when slicing

bogdi1988, 2023-05-22:
Describe the bug
When a model is printed with multiple colors, the slicer chooses a random color for the calibration lines as well as the first filament to be printed.
In the attached screenshot, notice that the model is sliced with blue as the primary color, yet the slicer sets the white accent color as the material that will be used for calibration, as well as the very first section of the first layer that is printed.
The issue is present in BambuStudio 1.6.2. Previous builds of the app would assign the flow calibration to the correct primary object color instead of a random color.

3mf File for This Bug
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

To Reproduce
Paint a model with multiple colors. Slice model. Notice the calibration color is not the same as the main model color.

Expected behavior
When slicing, the first color to be used is the main, primary color of the model

Screenshots
[image]

Printer model
X1C

Desktop (please complete the following information):

OS: Windows
Version 11

SaltWei, 2023-05-23:
> The issue is present in BambuStudio 1.6.2. Previous builds of the app would assign the flow calibration to the correct primary object color instead of a random color.

Please share a 3mf file for debugging. Thanks!

bogdi1988, 2023-05-23:
[Chrome Badge.zip](https://github.com/bambulab/BambuStudio/files/11541450/Chrome.Badge.zip)
Attached is a 3mf file. You can load it in 1.6.2 and older and slice it and see how the color order changes.

SaltWei, 2023-05-24:
The flow calibration uses the first normal filament (non-support filament) of model.
So this issue is caused by wrong printing order. The first layer should prints largest area first.
@zhimin-zeng-bambulab Please follow this issue of color order.

zhimin-zeng-bambulab, 2023-05-30:
#1816

bogdi1988, 2023-05-30:
Any ETA on this? I am stuck with older versions of the slicer due to this...

zhimin-zeng-bambulab, 2023-08-18:
Included in latest release https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52.

You can customize the material printing order of the first layer.

closed 2023-08-18 by zhimin-zeng-bambulab