bambulab/BambuStudio #1847: New project has invalid build plate selected

julie777, 2023-05-23:
### Description of the bug

I opened Bambu Studio, which created a new project.
when I switched to prepare it had the default profile for my only printer selected. (Bambu Lab P1P .4 nozzle).
However, it has the engineering plate selected. The engineering plate does not even ship with the P1P. 

There was one time in the past where I did not notice this and as you would expect the print was a failure.

### Project file & How to reproduce

1. open bambu studio with the only defined printer being a P1P with a .4mm nozzle.
2. look at the plate setting.

### Version

1.6.2.4

### System Information

OS: Windows 10

### Printer model

P1P

### Where is the application from?

Bambu Lab Website

lanewei120, 2023-05-24:
Hi

thanks for reporting this issue

currently the plate type is a project setting which will be saved in the 3mf
and it will be kept the same as your last setting when you new a project.
even you select the printer to another one, it won't be changed

for your case, have you selected engineering plate before you switch to Bambu Lab P1P .4 nozzle?

julie777, 2023-05-25:
I have never selected the engineering plate. I don't have one. I only have the textured PEI that came with the P1P

lanewei120, 2023-05-25:
[image]
could you give a short gif to show how to reproduce?

we can not reproduce here

julie777, 2023-06-09:
I cannot reproduce it. I have had it happen a couple times, but have no idea what caused it.

lanewei120, 2023-09-09:
still can not reproduce it, close it and re-open it when we find a way to reproduce it

closed 2023-09-09 by lanewei120