bambulab/BambuStudio #1849: Start gcode for P1P has texture bed compensation in the wrong position

julie777, 2023-05-23:
### Description of the bug

I am not sure this is a bug, but this was the only reasonable choice for submitting this question/bug.

In the start gcode, there is code to modify the nozzle offset to compensate for the textured PEI plate. It adjusts for the fact that homing and bed calibration touch the nozzle to the top of the texturing, but actual printing prints to the bottom of the the texture so the nozzle offset is adjusted to compensate. This is the section I am referring to:

```
;===== for Textured PEI Plate , lower the nozzle as the nozzle was touching topmost of the texture when homing ==
;curr_bed_type={curr_bed_type}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.04} ; for Textured PEI Plate
{endif}
```

The nozzle load line is printed just before this nozzle height adjustment is made. 
I would think that the compensation would come first to make sure the nozzle load line was printed normally and squished enough to adhere to the plate.

If this is a bug I would like to know.
If it is not a bug then I would like to understand the reason for printing the nozzle load line with a greater offset.

### Project file & How to reproduce

not needed

### Version

1.6.2.4

### System Information

Windows 10

### Printer model

P1P

### Where is the application from?

BBL site

lanewei120, 2023-05-24:
Hello

thanks for reporting this issue
we have got feadback from our printer team:
"this is a minor issue , as the first load line is about 50% over extruded so it is ok for now . we will fix this later."

FlowForm, 2024-01-31:
Will changing this height setting only apply for the first layer or also the following ones?