bambulab/BambuStudio #1851: Support floating in air

kfh13, 2023-05-23:
### Description of the bug

Support generated for model is just floating in air. Incorrect support is generated for "On build plate only".
[image]
[image]

### Project file & How to reproduce

Support generated for model cut by bambu studio with support setting given in screenshot.
[TOP_split.zip](https://github.com/bambulab/BambuStudio/files/11548837/TOP_split.zip)

### Version

Bambu Studio 1.6.2.4

### System Information

OS: Windows 11
CPU: Ryzen 5 3700x
GPU: Radeon 5700xt

### Printer model

Bambu lab P1P

### Where is the application from?

Bambu lab github release

manch1n, 2023-06-14:
This is an internal design fault of normal support.Maybe you could try to set Noraml support expansion to -1 mm.

UndeadKernel, 2023-08-05:
I would like to keep this bug open. I still see floating supports when using, for example, tree support options.

tangketan, 2023-08-31:
> I would like to keep this bug open. I still see floating supports when using, for example, tree support options.

You are welcome to raise another issue related to tree support. But for this issue, there is little we can do without making normal support too complicated, which violates the principle of normal support.