bambulab/BambuStudio #1855: 256X256X256 full size print layers not slicing the print correctly

vzb007, 2023-05-24:
### Bambu Studio Version

1.6.24

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

i7 32gb ram 1tb Hp dragonfly

### Printer

Bambu labs X1C

### How to reproduce

put a 256X256X256 cube to print select center by right clicking 

once postioned select .24 or .28 layer draft and extra draft it will give a error "a g-code path goes beyond plate boundaries"
by try the same print with .20 layer layer it works. all other layer heights done work

[image]
[image]

two images attached for reference 

### Actual results

there is a error when selecting .20 layer and .24 layer

### Expected results

there should be no error

### Project file & Debug log uploads

the siles are supplied to the support team too

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2023-05-29:
[image]
The model is 256mm, and sliced with fixed layer height 0.24.
Finally, the layer number is round up to 1067. And height is 1066*0.24 + initial layer 0.2 = 256.04mm, which is mostly close to the real height of model.

SaltWei, 2023-05-29:
> a g-code path goes beyond plate boundaries

In this case, the message should be "the gcode path goes beyond printable height"
This should be modified.

vzb007, 2023-05-30:
Thanks for this. So the slicer will not make a adjustment to fit the last layer within the size. Could there be a better explanation and option to assist people. You have very helpful thank you.

Have a another issue with the walls on the same print not sticking together properly. I am attaching a image for reference 
[image]
[image]
[image]

SaltWei, 2023-07-06:
Could you share gcode file so we can check and reprint as well ?

SaltWei, 2023-08-14:
No response for a long time. Close.
Reopen if has new update

closed 2023-08-14 by SaltWei