bambulab/BambuStudio #1858: Auto Arrange In Multiplate Project Causes Some Items To Indicate Unprintable

MadMax389, 2023-05-25:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Win 11

### Additional system information

Core i5 10th gen 32GB RAM
Nvidia 3060

### Printer

X1 Carbon Combo

### How to reproduce

1. Open uploaded project. Project contains some objects set to unprintable on plate 5
2. Click auto arrange on plate 4

### Actual results

The project contains some objects set to unprintable on plate 5
[image]

After selecting auto arrange on plate 4, several items are now showing unprintable on plates 2 and 3 and several items that were previously unprintable are now indicating printable on plate 5.
[image]

However, the item still indicates printable in the context menu and it does slice correctly.
[image]

Conversely, the items that were previously unprintable (that now indicate printable) are shown correctly in the context menu as unprintable. Again, it slices correctly.
[image]

Selecting auto arrange on other plates causes different objects to indicate incorrectly.
[image]

### Expected results

Expect no change in the printable status on any plate after using auto arrange.

### Project file & Debug log uploads

Project file was too big to upload here. Please follow [link](https://drive.google.com/file/d/1qxPXtZYY5Tkie9ANVsCq-SKV4iR0Gfn2/view?usp=sharing) 

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-05-26:
Thanks for reporting the bug. We are investigating it.

lanewei120, 2023-08-05:
already fixed in previous versions

closed 2023-08-05 by lanewei120