bambulab/BambuStudio #1862: Send button moves

TennisRunner, 2023-05-26:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Whenever I need to send a print over lan. 
It takes 10-20 seconds and when it finishes "synchronizing device information".
The label disappears making the entire GUI shrink which makes the send button move up the entire height of the send button.
So that I have to move my mouse up in order to click it.

My recommended solution would be to have it not be slow with the sync, or have the send button available for clicking immediately and have it gray out once you've clicked it instead of making me sit there waiting to click and miss every time.

Here is a picture of the before and after:
[image]

### Printer

p1p

### How to reproduce

Click print plate with lan set up.

### Actual results

Button is unclickable for long time and when it is clickable, it moves out the way.

### Expected results

Button should be instantly clickable 

### Project file & Debug log uploads

[foo.3mf.zip](https://github.com/bambulab/BambuStudio/files/11578531/foo.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-06-05:
Hi.
Thank you for your suggestion
we will consider using a new UI layout.

PippoAe, 2023-08-18:
Why is the sync in lan mode so low compared to cloud mode?
I get lots of "Synchronizing device information" and have to try multiple times to connect to the LAN-Mode Bambu.

TennisRunner, 2023-08-18:
> Why is the sync in lan mode so low compared to cloud mode? I get lots of "Synchronizing device information" and have to try multiple times to connect to the LAN-Mode Bambu.

For real. Lan mode is buggy on purpose.

walterwongbbl, 2023-10-13:
@TennisRunner 
Please download and use the latest version of Studio

closed 2023-12-21 by walterwongbbl