bambulab/BambuStudio #1863: First print after starting bambu studio has option to enable AMS on device that doesn't support AMS

TennisRunner, 2023-05-26:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1p

### How to reproduce

Boot up bambu studio.
Load any file.
Click print plate.

### Actual results

A checkbox with Enable AMS appears and is always checked on.
[image]

### Expected results

Enable AMS shouldn't even appear since p1p doesn't support AMS.
It disappears after the finishes syncing in about 20 seconds.

### Project file & Debug log uploads

[foo.3mf.zip](https://github.com/bambulab/BambuStudio/files/11578682/foo.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-06-06:
Hi.
You do not have an AMS device. Is that right？

TennisRunner, 2023-06-06:
> Hi. You do not have an AMS device. Is that right？

Correct

walterwongbbl, 2023-06-06:
Thank you. We will fix this issue

walterwongbbl, 2023-06-08:
Hi.
Can you upload the studio log file

walterwongbbl, 2023-06-08:
Hi.
We have fixed this issue and will include it in the next version

closed 2023-10-13 by walterwongbbl