bambulab/BambuStudio #1864: Problem parsing on first attempt

TennisRunner, 2023-05-26:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

p1p

### How to reproduce

Click on print plate

### Actual results

This happens on the first attempt at printing a step file.
It happens intermittently.
When I click to print and it says there was a problem parsing gcode.3mf.
Then I tell it to print the plate again (sometimes, I move the object just a mm on the plate to make it re-slice) and it works.

[image]

### Expected results

Should just print 

### Project file & Debug log uploads

[foo.3mf.zip](https://github.com/bambulab/BambuStudio/files/11578699/foo.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sodachen22, 2023-05-29:
@TennisRunner Was it a random issue or did it happen on some specific sliced files? Normally, the possible reason is the file name is not supported because of some specific character or there is a problem with sliced files.

TennisRunner, 2023-05-29:
I import .step files and it happens at random, but usually only on the first slice and print attempt

TennisRunner, 2023-06-10:
The popup happened again:
[image]

So while the popup was up, I ejected my sd card from the printer and copied over the latest file which is:
[Untitled_plate_1.gcode.zip](https://github.com/bambulab/BambuStudio/files/11711283/Untitled_plate_1.gcode.zip)

I am able to open the file in bambu studio and view it in the prepare tab.

sodachen22, 2023-06-12:
The gcode.3mf file is OK. It looks like there is a problem when downloading or processing the file. Would you please create a ticket and share the LOG? Here is a guide https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

fangly, 2023-07-04:
I have the same issue using the same version of Bambu Lab Studio on the same (brand new) printer. I had to resend the print job quite a few times... and now finally, it seems like it is starting.

Additional information:
1. I am running Studio on Ubuntu Linux 22.04
2. The SDcard I provided is a 64 GB Samsung card, which I just formatted using ext32. It seems to be accepted by Studio, as evidenced by the small green check mark under Devices.
3. The model I wanted to print was the popular 3D Benchy, and I used pretty much all default options in Studio. 

@sodachen22 Could you identify a reason for the issue?

adminy, 2024-02-26:
I have a similar issue @fangly, I noticed if I set the temperature to 40/40/40 on all 3 it finishes the print. Its really annoying because it stops past first 3 layers on all my prints ... then I set that temperature magic, then it finishes.

dylan-conlin, 2024-03-07:
I experienced this issue today. Restarting the printer fixed it for now.