bambulab/BambuStudio #1865: Aligned Rectilinear - Layer offset

MrBenGriffin, 2023-05-27:
**Describe the solution you'd like**
When printing transparent materials, **aligned rectilinear** is definitely the sparse infill of choice (along withmatched top/bottom pattern).

However, it would be really advantageous (when printing transparency) useful to add an offset aligned rectilinear, such that (and depending upon line thickness), the intermediate layer fill the 'gaps' of the prior layer by offsetting the position. 

Currently the aligned rectilinear prints on top of itself each time, which creates ridges - and leaves ridged effects. (see the picture on the left).  By using a suitable offset as depicted on the right, (and adjusting layer height accordingly), a far better transparency could be achieved.

[image]

This might be better depicted as a distinct fill pattern, rather than adding an offset metric to the current fill.

QingZhangBambu, 2023-08-10:
I get you. However, the cutaway of the path is not a pure circle. The nozzle squash it during the printing, so the gap is filled.
[image]

QingZhangBambu, 2023-08-10:
so if you are concerned about ridges and gaps, it would be fine.

MrBenGriffin, 2023-08-10:
I have seen the ridges and gaps - it is not theoretical.

While they aren’t circular, the squash itself does not fully fill  - and while the gap should be filled it often is not. 

This is easy to see from the defects arising when printing aligned rectilinear with transparent filament - especially under microscope.  

When printing normally, the defects are very small and do not matter. But when printing glassy windows the defects are very noticeable. Ironing does not help either. 

You may have more experience with fine-tuning - maybe I should be using a higher temperature in order to improve adhesion on adjacent lines. 

Regardless - if you actually investigate optimal  transparency printing - I believe you will come to a similar conclusion to the one here.

QingZhangBambu, 2023-08-10:
if you don't mind, could you provide 3mf file, I'd love to test it. it could help me to know how to adjust the pattern.

MrBenGriffin, 2023-08-10:
Of course.  You can also see the [forum article I wrote about printing transparency](https://forum.bambulab.com/t/transparently-curious-about-ironing/8930/8)

QingZhangBambu, 2023-08-10:
@MrBenGriffin THANKS

MrBenGriffin, 2023-08-10:
Further comments - I printed out the model above and hand-polished the surfaces - both sides - starting from P1200 and going to P10k, in order to remove surface defects.  

Needless to say, without any surface defects (both sides are now reflective) there is a dramatic improvement to the transparency - but (and this was my point) the internal aligned rectilinear gaps are visible.  The thing is that they are being accentuated by lining exactly on top of each other.  It might be true that if they are offset on each layer, the problem is not improved: Maybe I need to  increase the line width on sparse infill (which is 100% aligned rectilinear), speed up the  extrusion on the fill part of the model, heat the filament a bit more, reduce the speed - or increase it in order to improve melt...  
I am also running a fine calibration print - which may help also.

[image]

QingZhangBambu, 2023-08-11:
@MrBenGriffin I adjust the pattern, and there is the comparison. They are almost the same.
[image]

[image]
 The surface curvature of the extrusion path and the air content in between can affect being transparent.  small layer height and big with could be more helpful.  And There is our wiki for transparent filament. https://wiki.bambulab.com/en/knowledge-sharing/transparent-petg

QingZhangBambu, 2023-08-11:
To get a good result, the wall of the printed part must be uniform: the composition is uniform and dense, without defects such as bubbles and voids, and the direction of the lines is consistent, so as to avoid different refractive indices and different refraction directions of the printed matter. Every part has light.

MrBenGriffin, 2023-08-11:
@QingZhangBambu thats very interesting - note that my work has been with PLA, not PETG.  I did further tests yesterday and found that high temperatures and fast local speed can help to unify the thread, but at the cost of many more surface defects.

One thing I also noticed was that, while line width improves translucency, a larger width also adds to the ridges on the top. I might try a different width for the top layer in order to reduce diffraction - I believe it will increase transmission at the cost of increased opacity.

Your samples seem to be quite milky in comparison to my tests. 

Thankyou so very much for having a look at this. I believe I can improve this further. I would very much like to know how to send GCode to the printer so that I can explore this further without having to ask you to look for me!

QingZhangBambu, 2023-08-11:
@MrBenGriffin My transparent filament is PETG， and high temperature helps the filament fill gaps, and exclude the air which improves the translucency. 
and I am not sure what the meaning of "how to send the GCode to the printer"
is that does work?
[image]

QingZhangBambu, 2023-08-11:
and how about ironing for each layer?

MrBenGriffin, 2023-08-11:
I hear what you say but I have had much better results with transparent PLA - I do not know why!
when I tried ironing it added some smaller defects.

Oh… thankyou - I needed to turn on developer mode!! 🙏🏻

QingZhangBambu, 2023-08-11:
you can turn on developer mode here. 
[image]
[image]

MrBenGriffin, 2023-08-11:
[image]

But no upload anyway..
[image]

QingZhangBambu, 2023-08-11:
are you want to send a .gcode(not .gcode.3mf) file to the printer? it could not be sent by Studio. You need to copy that to an SD card and choose to print it on the printer

MrBenGriffin, 2023-08-11:
Hi! yes, that worked really well! I was [looking at the full control python system](https://fullcontrol.xyz/#/models/4d0e78) - and it worked! No retraction!
I am hoping I can use this library for my own programmable slicing...
[image]

Thanks for your help!

MrBenGriffin, 2023-08-12:
While there appear to be many other issues that need to be resolved to improve transparency, I believe that an offset would still be useful for increasing laminar strength - even if it is just a regular 50% of line thickness.

QingZhangBambu, 2023-08-15:
Good