bambulab/BambuStudio #1866: MC Fan Checkbox/ enable feature, GUI feature.

bl4ckm0onk3y, 2023-05-27:
**Is your feature request related to a problem? Please describe.**
Bought MC Fan for X1 (compatible with P1P, fits perfectly, it's also plugged in to MC Fan port) Need the MC Fan because the P1P is enclosed in ARC Case. There is a GCode for the MC fan to turn on but it has to be added every time i want to print. And i don't know if the fan works, or it's the correct GCode. 

**Describe the solution you'd like**
Add a checkbox or add control of the MC Fan to Bambu Studio and Bambu Handy.

HobinBambu, 2023-05-29:
Hi @bl4ckm0onk3y 
    The next firmware version of P1P will support the setting and **saving** feature of MC fan,so you only need to execute Gcode once.you can use "M710 A1 S255" to turn on the automatic control of the MC fan,the MC fan will automatically turn on when the stepper motor is turned on.

bl4ckm0onk3y, 2023-05-29:
Thank you, glad to see that this will be added.

closed 2023-05-29 by bl4ckm0onk3y