bambulab/BambuStudio #1868: failed to open on Ubuntu 22.04.2

ryancpu, 2023-05-28:
### Bambu Studio Version

Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage 

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

Bambu X1 carbon

### How to reproduce

try to run appimage

### Actual results

[image]

### Expected results

seg fault

### Project file & Debug log uploads

not able to open so so project file

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-05-29:
Please also provide the core dump file.

ryancpu, 2023-05-29:
file:///var/lib/systemd/coredump/core.bambustu_main.1000.4ff935a8e40547a3a00bc25d7b1f04a1.88606.1685399079000000.zst

bambu123, 2023-05-30:
> file:///var/lib/systemd/coredump/core.bambustu_main.1000.4ff935a8e40547a3a00bc25d7b1f04a1.88606.1685399079000000.zst

Maybe this file, please send it to us zipped.

ryancpu, 2023-05-30:
[core.bambustu_main.1000.4ff935a8e40547a3a00bc25d7b1f04a1.88606.1685399079000000.zip](https://github.com/bambulab/BambuStudio/files/11594728/core.bambustu_main.1000.4ff935a8e40547a3a00bc25d7b1f04a1.88606.1685399079000000.zip)

bambu123, 2023-05-30:
> [core.bambustu_main.1000.4ff935a8e40547a3a00bc25d7b1f04a1.88606.1685399079000000.zip](https://github.com/bambulab/BambuStudio/files/11594728/core.bambustu_main.1000.4ff935a8e40547a3a00bc25d7b1f04a1.88606.1685399079000000.zip)

Sorry, I can't restore the crash stack from this file.

Could you please help debuging the appimage? To do:

# gdb Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage
run

then send the console output back here.

ryancpu, 2023-05-30:
when I execute "$ gdb ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage"

it's probably not much help unless I'm doing it wrong

GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage...
(No debugging symbols found in ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage)
(gdb)

bambu123, 2023-05-30:
> Find the GDB manual and other documentation resources online at:> http://www.gnu.org/software/gdb/documentation/.For help, type "help".

please input the 'run' command in gdb.nd then when it stops on a crash, input the 'bt' command.

ryancpu, 2023-05-30:
(gdb) run
Starting program: /home/11111111/3d printing/bambu_official/Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 118946]
process 118943 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 118951]
process 118943 is executing new program: /tmp/.mount_Bambu_o9eBoV/bin/bambu-studio
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2023-05-30 17:24:08.215778] [0x00007fffed499980] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7fffec8b1640 (LWP 118961)]
[Thread 0x7fffec8b1640 (LWP 118961) exited]
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
[New Thread 0x7fffec8b1640 (LWP 118962)]
[New Thread 0x7fffe7fff640 (LWP 118963)]
[Detaching after vfork from child process 118964]
[Detaching after vfork from child process 118966]
[New Thread 0x7fffe77fe640 (LWP 118968)]
[New Thread 0x7fffe6ffd640 (LWP 118969)]
[Thread 0x7fffe77fe640 (LWP 118968) exited]
[Thread 0x7fffe6ffd640 (LWP 118969) exited]
[New Thread 0x7fffe6ffd640 (LWP 118970)]
[New Thread 0x7fffe77fe640 (LWP 118971)]
[Thread 0x7fffe6ffd640 (LWP 118970) exited]
[Thread 0x7fffe77fe640 (LWP 118971) exited]
[New Thread 0x7fffe77fe640 (LWP 118972)]
[New Thread 0x7fffe6ffd640 (LWP 118973)]
[Thread 0x7fffe77fe640 (LWP 118972) exited]
[Thread 0x7fffe6ffd640 (LWP 118973) exited]
[New Thread 0x7fffe6ffd640 (LWP 118974)]
[New Thread 0x7fffe77fe640 (LWP 118975)]
[Thread 0x7fffe6ffd640 (LWP 118974) exited]
[Thread 0x7fffe77fe640 (LWP 118975) exited]
[New Thread 0x7fffe77fe640 (LWP 118976)]
[New Thread 0x7fffe6ffd640 (LWP 118977)]
[Thread 0x7fffe77fe640 (LWP 118976) exited]
[Thread 0x7fffe6ffd640 (LWP 118977) exited]
[New Thread 0x7fffe6ffd640 (LWP 118978)]
[New Thread 0x7fffe77fe640 (LWP 118979)]
[Thread 0x7fffe6ffd640 (LWP 118978) exited]
[Thread 0x7fffe77fe640 (LWP 118979) exited]
[New Thread 0x7fffe77fe640 (LWP 118980)]
[New Thread 0x7fffe6ffd640 (LWP 118981)]
[New Thread 0x7fffe4a2b640 (LWP 118982)]
[Thread 0x7fffe77fe640 (LWP 118980) exited]
[Thread 0x7fffe6ffd640 (LWP 118981) exited]
[New Thread 0x7fffe6ffd640 (LWP 119026)]
[New Thread 0x7fffe77fe640 (LWP 119027)]
[Thread 0x7fffe6ffd640 (LWP 119026) exited]
[Thread 0x7fffe77fe640 (LWP 119027) exited]
[New Thread 0x7fffe77fe640 (LWP 119028)]
[New Thread 0x7fffe6ffd640 (LWP 119029)]
[Thread 0x7fffe77fe640 (LWP 119028) exited]
[Thread 0x7fffe6ffd640 (LWP 119029) exited]
[New Thread 0x7fffe6ffd640 (LWP 119030)]
[New Thread 0x7fffe77fe640 (LWP 119031)]
[Thread 0x7fffe6ffd640 (LWP 119030) exited]
[Thread 0x7fffe77fe640 (LWP 119031) exited]
loading network module at /home/11111111/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/1111111/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:118943): Gtk-CRITICAL **: 17:24:12.969: gtk_window_resize: assertion 'height > 0' failed
[New Thread 0x7fffe77fe640 (LWP 119032)]
[New Thread 0x7fffe6ffd640 (LWP 119033)]
[Thread 0x7fffe77fe640 (LWP 119032) exited]
[... 125 lines cut ...]
#5  0x00007ffff117bf4a in __malloc_assert
    (assertion=assertion@entry=0x7ffff12b99a8 "(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)", file=file@entry=0x7ffff12b46ac "malloc.c", line=line@entry=2617, function=function@entry=0x7ffff12ba218 <__PRETTY_FUNCTION__.8> "sysmalloc")
    at ./malloc/malloc.c:307
#6  0x00007ffff117e9b7 in sysmalloc (nb=nb@entry=3088, av=av@entry=0x7ffff12f4c80 <main_arena>) at ./malloc/malloc.c:2617
#7  0x00007ffff117f95d in _int_malloc (av=av@entry=0x7ffff12f4c80 <main_arena>, bytes=bytes@entry=3072) at ./malloc/malloc.c:4407
#8  0x00007ffff11801b9 in __GI___libc_malloc (bytes=3072) at ./malloc/malloc.c:3329
#9  0x00005555588b40ac in wxImage::Create(int, int, bool) ()
#10 0x00005555564e3738 in Slic3r::GUI::BitmapCache::insert_raw_rgba(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, unsigned char const*, bool) ()
#11 0x00005555564e4ccb in Slic3r::GUI::BitmapCache::load_svg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float) ()
#12 0x00005555568de9c1 in create_scaled_bitmap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, wxWindow*, int, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) ()
#13 0x00005555568df554 in ScalableBitmap::ScalableBitmap(wxWindow*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, bool, bool) ()
#14 0x0000555556828613 in Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
#15 0x0000555556829428 in Slic3r::GUI::TabPrint::build() ()
[... cut ...]
#18 0x00005555566c53f8 in Slic3r::GUI::MainFrame::create_preset_tabs() ()
#19 0x00005555566d24bd in Slic3r::GUI::MainFrame::init_tabpanel() ()
#20 0x00005555566d37f8 in Slic3r::GUI::MainFrame::MainFrame() ()
#21 0x00005555566934a8 in Slic3r::GUI::GUI_App::on_init_inner() ()
#22 0x000055555669469a in Slic3r::GUI::GUI_App::OnInit() ()
#23 0x0000555558b99617 in wxEntry(int&, wchar_t**) ()
#24 0x000055555665c14b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#25 0x0000555555e162b7 in Slic3r::CLI::run(int, char**) ()
#26 0x0000555555d71b99 in main ()
(gdb)

bambu123, 2023-05-31:
@ryancpu 
Thank you very much! The stack is very clear.
But it shows that the fault is the in the system library or maybe a random fault.
To make it clear, I need you to repeat the above steps.
A second case: remove the ~/.config/BambuStudio directory and try it again.
And compare with the old versions of BambuStudio.

ryancpu, 2023-05-31:
I deleted the .config file and it still crashed.  I will delete the config file again and provide a new gdb output

GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage...
(No debugging symbols found in ./Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage)
(gdb) run
Starting program: /home/111111/3d printing/bambu_official/Bambu_Studio_linux_ubuntu_v01.06.02.04-20230427094209.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 131540]
process 131537 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 131545]
process 131537 is executing new program: /tmp/.mount_Bambu_ip9fTO/bin/bambu-studio
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2023-05-30 21:26:34.582172] [0x00007fffed499980] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7fffec8b1640 (LWP 131549)]
[Thread 0x7fffec8b1640 (LWP 131549) exited]
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
[New Thread 0x7fffec8b1640 (LWP 131550)]
[New Thread 0x7fffe7fff640 (LWP 131551)]
[Detaching after vfork from child process 131552]
[Detaching after vfork from child process 131554]
[New Thread 0x7fffe77fe640 (LWP 131556)]
[New Thread 0x7fffe6ffd640 (LWP 131557)]
[Thread 0x7fffe77fe640 (LWP 131556) exited]
[Thread 0x7fffe6ffd640 (LWP 131557) exited]
[New Thread 0x7fffe6ffd640 (LWP 131558)]
[New Thread 0x7fffe77fe640 (LWP 131559)]
[Thread 0x7fffe6ffd640 (LWP 131558) exited]
[Thread 0x7fffe77fe640 (LWP 131559) exited]
[New Thread 0x7fffe77fe640 (LWP 131560)]
[New Thread 0x7fffe6ffd640 (LWP 131561)]
[Thread 0x7fffe77fe640 (LWP 131560) exited]
[Thread 0x7fffe6ffd640 (LWP 131561) exited]
[New Thread 0x7fffe6ffd640 (LWP 131562)]
[New Thread 0x7fffe77fe640 (LWP 131563)]
[Thread 0x7fffe6ffd640 (LWP 131562) exited]
[Thread 0x7fffe77fe640 (LWP 131563) exited]
[New Thread 0x7fffe77fe640 (LWP 131564)]
[New Thread 0x7fffe6ffd640 (LWP 131565)]
[Thread 0x7fffe77fe640 (LWP 131564) exited]
[Thread 0x7fffe6ffd640 (LWP 131565) exited]
[New Thread 0x7fffe6ffd640 (LWP 131566)]
[New Thread 0x7fffe77fe640 (LWP 131567)]
[Thread 0x7fffe6ffd640 (LWP 131566) exited]
[Thread 0x7fffe77fe640 (LWP 131567) exited]
[New Thread 0x7fffe77fe640 (LWP 131568)]
[New Thread 0x7fffe6ffd640 (LWP 131569)]
[New Thread 0x7fffe4a2b640 (LWP 131570)]
[Thread 0x7fffe4a2b640 (LWP 131570) exited]
[Thread 0x7fffe6ffd640 (LWP 131569) exited]
[New Thread 0x7fffe6ffd640 (LWP 131572)]
[... 139 lines cut ...]
#1  __pthread_kill_internal (signo=6, threadid=140737174411648) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737174411648, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff111d476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff11037f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff11646f6 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff12b6b8c "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007ffff117bd7c in malloc_printerr (str=str@entry=0x7ffff12b4764 "free(): invalid pointer") at ./malloc/malloc.c:5664
#7  0x00007ffff117dac4 in _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at ./malloc/malloc.c:4439
#8  0x00007ffff11804d3 in __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3391
#9  0x00005555564e4ca3 in Slic3r::GUI::BitmapCache::load_svg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float) ()
#10 0x00005555568de9c1 in create_scaled_bitmap(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, wxWindow*, int, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) ()
#11 0x00005555568df554 in ScalableBitmap::ScalableBitmap(wxWindow*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, bool, bool) ()
#12 0x0000555556828613 in Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
#13 0x0000555556829428 in Slic3r::GUI::TabPrint::build() ()
#14 0x0000555556823228 in Slic3r::GUI::Tab::create_preset_tab() ()
[... cut ...]
#16 0x00005555566c53f8 in Slic3r::GUI::MainFrame::create_preset_tabs() ()
#17 0x00005555566d24bd in Slic3r::GUI::MainFrame::init_tabpanel() ()
#18 0x00005555566d37f8 in Slic3r::GUI::MainFrame::MainFrame() ()
#19 0x00005555566934a8 in Slic3r::GUI::GUI_App::on_init_inner() ()
#20 0x000055555669469a in Slic3r::GUI::GUI_App::OnInit() ()
#21 0x0000555558b99617 in wxEntry(int&, wchar_t**) ()
#22 0x000055555665c14b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#23 0x0000555555e162b7 in Slic3r::CLI::run(int, char**) ()
#24 0x0000555555d71b99 in main ()
(gdb)

bambu123, 2023-05-31:
@ryancpu Thank you again. It seams to be a random fault.

ghedo, 2023-05-31:
FWIW, I have this same problem too (program crashes on start inside the `load_svg()` function).

From the libc error logs (`sysmalloc: Assertion '...' failed` and `free(): invalid pointer`) it seems like there could be some kind of memory corruption in that function?

Unfortunately I haven't been able to build BambuStudio with debug information yet to pinpoint the problem more accurately.

alexivkin, 2023-06-02:
I had the same problem on 22.04 which turned out to be a library version mismatch. In my case Bambu Studio wanted libwebkit2gtk-4.0-37 but the system had v4.1
Solved with `apt install libwebkit2gtk-4.0-37`

@bambu123 It would be great if you can build AppImage the way it's meant to be, i.e. by including all the right dependencies, not expecting them to come from the system.

ryancpu, 2023-06-02:
@alexivkin I have v4.1 also.  Did you remove 4.1 and install 4.0-37?  I tried to apt install libwebkit2gtk-4.0-37 but it still leaves me with v4.1.

I really hope this will solve the issue.  Thank you

bambu123, 2023-06-05:
> I had the same problem on 22.04 which turned out to be a library version mismatch. In my case Bambu Studio wanted libwebkit2gtk-4.0-37 but the system had v4.1 Solved with `apt install [...]

Thank you. What is the behavior of the Studio without libwebkit2gtk-4.0?

bambu123, 2023-06-06:
> I had the same problem on 22.04 which turned out to be a library version mismatch. In my case Bambu Studio wanted libwebkit2gtk-4.0-37 but the system had v4.1 Solved with `apt install [...]

Gtk is a system library and we can't provide a copy to run parallel with the system one.

bambu123, 2023-06-06:
@ryancpu
As @alexivkin said. A miss-match library may cause memory corruption. It is better to build the app image by yourself.
Follow the steps here: https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide

ryancpu, 2023-06-06:
Thank you, I've tried both the script and manual compile listed on the page and I get this error:

[ 85%] Completed 'dep_wxWidgets'
[ 85%] Built target dep_wxWidgets
make: *** [Makefile:91: all] Error 2

bambu123, 2023-06-06:
> Thank you, I've tried both the script and manual compile listed on the page and I get this error:
> 
> [ 85%] Completed 'dep_wxWidgets' [ 85%] Built target dep_wxWidgets make: *** [Makefile:91: all] Error 2

Input 'make' command, try again and sea what's the error.

ryancpu, 2023-06-06:
If I input make -j8 I get the same error.  I will copy farther up the screen.   If I just type 'make' there is a different error I will add below.

make -j8
CMake Error: cmake version 3.22.1
Usage: /usr/bin/cmake -E <command> [arguments...]
Available commands: 
  capabilities              - Report capabilities built into cmake in JSON format
  cat <files>...            - concat the files and print them to the standard output
  chdir dir cmd [args...]   - run command in a given directory
  compare_files [--ignore-eol] file1 file2
                              - check if file1 is same as file2
  copy <file>... destination  - copy files to destination (either file or directory)
  copy_directory <dir>... destination   - copy content of <dir>... directories to 'destination' directory
  copy_if_different <file>... destination  - copy files if it has changed
  echo [<string>...]        - displays arguments as text
  echo_append [<string>...] - displays arguments as text but no new line
  env [--unset=NAME]... [NAME=VALUE]... COMMAND [ARG]...
                            - run command in a modified environment
  environment               - display the current environment
  make_directory <dir>...   - create parent and <dir> directories
  md5sum <file>...          - create MD5 checksum of files
  sha1sum <file>...         - create SHA1 checksum of files
  sha224sum <file>...       - create SHA224 checksum of files
  sha256sum <file>...       - create SHA256 checksum of files
  sha384sum <file>...       - create SHA384 checksum of files
  sha512sum <file>...       - create SHA512 checksum of files
  remove [-f] <file>...     - remove the file(s), use -f to force it (deprecated: use rm instead)
  remove_directory <dir>... - remove directories and their contents (deprecated: use rm instead)
  rename oldname newname    - rename a file or directory (on one volume)
  rm [-rRf] <file/dir>...    - remove files or directories, use -f to force it, r or R to remove directories and their contents recursively
  sleep <number>...         - sleep for given number of seconds
  tar [cxt][vf][zjJ] file.tar [file/dir1 file/dir2 ...]
                            - create or extract a tar or zip archive
  time command [args...]    - run command and display elapsed time
  touch <file>...           - touch a <file>.
  touch_nocreate <file>...  - touch a <file> but do not create it.
  create_symlink old new    - create a symbolic link new -> old
  create_hardlink old new   - create a hard link new -> old
  true                      - do nothing with an exit code of 0
  false                     - do nothing with an exit code of 1

CMake Error: cmake version 3.22.1
Usage: /usr/bin/cmake -E <command> [arguments...]
Available commands: 
  capabilities              - Report capabilities built into cmake in JSON format
  cat <files>...            - concat the files and print them to the standard output
  chdir dir cmd [args...]   - run command in a given directory
  compare_files [--ignore-eol] file1 file2
                              - check if file1 is same as file2
  copy <file>... destination  - copy files to destination (either file or directory)
  copy_directory <dir>... destination   - copy content of <dir>... directories to 'destination' directory
  copy_if_different <file>... destination  - copy files if it has changed
  echo [<string>...]        - displays arguments as text
  echo_append [<string>...] - displays arguments as text but no new line
  env [--unset=NAME]... [NAME=VALUE]... COMMAND [ARG]...
                            - run command in a modified environment
  environment               - display the current environment
  make_directory <dir>...   - create parent and <dir> directories
  md5sum <file>...          - create MD5 checksum of files
  sha1sum <file>...         - create SHA1 checksum of files
  sha224sum <file>...       - create SHA224 checksum of files
  sha256sum <file>...       - create SHA256 checksum of files
  sha384sum <file>...       - create SHA384 checksum of files
  sha512sum <file>...       - create SHA512 checksum of files
  remove [-f] <file>...     - remove the file(s), use -f to force it (deprecated: use rm instead)
  remove_directory <dir>... - remove directories and their contents (deprecated: use rm instead)
  rename oldname newname    - rename a file or directory (on one volume)
  rm [-rRf] <file/dir>...    - remove files or directories, use -f to force it, r or R to remove directories and their contents recursively
  sleep <number>...         - sleep for given number of seconds
  tar [cxt][vf][zjJ] file.tar [file/dir1 file/dir2 ...]
[... 29 lines cut ...]
Making install in mpq
Making install in mpf
Making install in printf
Making install in scanf
Making install in rand
Making install in cxx
Making install in demos
Making install in calc
Making install in expr
Making install in tune
Making install in doc
 /usr/bin/mkdir -p '/home/111111/3d printing/bambu_official/code/BambuStudio_deps/usr/local/share/info'
 /usr/bin/install -c -m 644 ./gmp.info ./gmp.info-1 ./gmp.info-2 '/home/111111/3d printing/bambu_official/code/BambuStudio_deps/usr/local/share/info'
 install-info --info-dir='/home/111111/3d printing/bambu_official/code/BambuStudio_deps/usr/local/share/info' '/home/1111111/3d printing/bambu_official/code/BambuStudio_deps/usr/local/share/info/gmp.info'
 /usr/bin/mkdir -p '/home/1111111/3d printing/bambu_official/code/BambuStudio_deps/usr/local/lib'
[... cut ...]
/usr/bin/install: target 'printing/bambu_official/code/BambuStudio_deps/usr/local/lib/libgmp.la' is not a directory
make[6]: *** [Makefile:848: install-libLTLIBRARIES] Error 1
make[5]: *** [Makefile:1303: install-am] Error 2
make[4]: *** [Makefile:997: install-recursive] Error 1
make[3]: *** [Makefile:1297: install] Error 2
make[2]: *** [CMakeFiles/dep_GMP.dir/build.make:105: dep_GMP-prefix/src/dep_GMP-stamp/dep_GMP-install] Error 2
make[1]: *** [CMakeFiles/Makefile2:452: CMakeFiles/dep_GMP.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

bambu123, 2023-06-06:
Try to use the 4th method.
[image]

ryancpu, 2023-06-06:
that method gives me the same error as make -j8

[ 85%] Completed 'dep_wxWidgets'
[ 85%] Built target dep_wxWidgets
make: *** [Makefile:91: all] Error 2

bambu123, 2023-06-07:
> /usr/bin/install: target 'printing/bambu_official/code/BambuStudio_deps/usr/local/lib/libgmp.la' is not a directory
> make[6]: *** [Makefile:848: install-libLTLIBRARIES] Error 1
> make[5]: *** [...]

Please remove the old build cache to make a refresh build.

bambu123, 2023-06-07:
@lanewei120 Please help the build issue on linux.

/usr/bin/install: target 'printing/bambu_official/code/BambuStudio_deps/usr/local/lib/libgmp.la' is not a directory
make[6]: *** [Makefile:848: install-libLTLIBRARIES] Error 1
make[5]: *** [Makefile:1303: install-am] Error 2
make[4]: *** [Makefile:997: install-recursive] Error 1
make[3]: *** [Makefile:1297: install] Error 2
make[2]: *** [CMakeFiles/dep_GMP.dir/build.make:105: dep_GMP-prefix/src/dep_GMP-stamp/dep_GMP-install] Error 2
make[1]: *** [CMakeFiles/Makefile2:452: CMakeFiles/dep_GMP.dir/all] Error 2

lanewei120, 2023-06-07:
> ES] Error 1
> make[5]: *** [Makefile:1303: install-am] Error 2
> make[4]: *** [Makefile:997: install-recursive] Error 1
> make[3]: *** [Makefile:1297: install] Error 2

it doesn't include the error reason

[ryancpu](https://github.com/ryancpu)
could you give the full building commands here?
please use 'make' instead of 'make -j8'
and then paste all the logs into a txt file and upload it here

thanks very much

ryancpu, 2023-06-07:
I created a new directory and did a complete new git clone into that directory.  for the first time ever ./BuildLinux.sh -dsi completed without errors

however the new appimage does not work  I will post the error I see when executing the appimage and below I will post the gdb and bt output

The error I get when executing the new appimage:

[2023-06-06 22:48:28.828545] [0x00007f64f96ae980] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/1111/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/1111/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.923: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:144405): GLib-GObject-WARNING **: 22:48:31.944: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.944: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:144405): GLib-GObject-WARNING **: 22:48:31.945: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.945: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:144405): GLib-GObject-WARNING **: 22:48:31.946: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.946: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:144405): GLib-GObject-WARNING **: 22:48:31.946: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.946: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:144405): GLib-GObject-WARNING **: 22:48:31.950: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.950: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:144405): GLib-GObject-WARNING **: 22:48:31.951: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.951: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.984: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.986: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.986: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.986: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.986: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.987: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.996: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.997: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:31.997: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:32.096: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:32.102: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:32.102: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:144405): Gtk-CRITICAL **: 22:48:32.102: gtk_widget_set_size_request: assertion 'width >= -1' failed
[... 219 lines cut ...]
#5  0x00007ffff117bf4a in __malloc_assert
    (assertion=assertion@entry=0x7ffff12b99a8 "(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)", file=file@entry=0x7ffff12b46ac "malloc.c", line=line@entry=2617, function=function@entry=0x7ffff12ba218 <__PRETTY_FUNCTION__.8> "sysmalloc")
    at ./malloc/malloc.c:307
#6  0x00007ffff117e9b7 in sysmalloc (nb=nb@entry=3088, av=av@entry=0x7ffff12f4c80 <main_arena>) at ./malloc/malloc.c:2617
#7  0x00007ffff117f95d in _int_malloc (av=av@entry=0x7ffff12f4c80 <main_arena>, bytes=bytes@entry=3072) at ./malloc/malloc.c:4407
#8  0x00007ffff11801b9 in __GI___libc_malloc (bytes=3072) at ./malloc/malloc.c:3329
#9  0x00005555588af76c in wxImage::Create(int, int, bool) ()
#10 0x00005555564e0008 in Slic3r::GUI::BitmapCache::insert_raw_rgba(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, unsigned char const*, bool) ()
#11 0x00005555564e159b in Slic3r::GUI::BitmapCache::load_svg(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > c [... cut ...]
#18 0x00005555566c1b58 in Slic3r::GUI::MainFrame::create_preset_tabs() ()
#19 0x00005555566cec7d in Slic3r::GUI::MainFrame::init_tabpanel() ()
#20 0x00005555566cffb8 in Slic3r::GUI::MainFrame::MainFrame() ()
#21 0x000055555668fc08 in Slic3r::GUI::GUI_App::on_init_inner() ()
#22 0x0000555556690dfa in Slic3r::GUI::GUI_App::OnInit() ()
#23 0x0000555558b94cd7 in wxEntry(int&, wchar_t**) ()
#24 0x00005555566588ab in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#25 0x0000555555e155e7 in Slic3r::CLI::run(int, char**) ()
#26 0x0000555555d70f09 in main ()
(gdb)

bambu123, 2023-06-07:
Same error as the public app image. The memory is corrupt.

bambu123, 2023-06-07:
@ryancpu There may be library misuse.  use the command 'info sharedlibrary' in gdb to show all loaded libraries.

ryancpu, 2023-06-08:
(gdb) info sharedlibrary
From                To                  Syms Read   Shared Object Library
0x00007ffff7fc5090  0x00007ffff7fee335  Yes         /lib64/ld-linux-x86-64.so.2
0x00007ffff7e0d440  0x00007ffff7f14b02  Yes (*)     /lib/x86_64-linux-gnu/libstdc++.so.6
0x00007ffff7d28fa0  0x00007ffff7d50b1e  Yes (*)     /lib/x86_64-linux-gnu/libfontconfig.so.1
0x00007ffff7d07280  0x00007ffff7d17c14  Yes (*)     /lib/x86_64-linux-gnu/libz.so.1
0x00007ffff7cc11c0  0x00007ffff7cc3f11  Yes (*)     /lib/x86_64-linux-gnu/libGL.so.1
0x00007ffff7ba53a0  0x00007ffff7c208e8  Yes         /lib/x86_64-linux-gnu/libm.so.6
0x00007ffff7b857a0  0x00007ffff7b8e3a5  Yes (*)     /lib/x86_64-linux-gnu/libEGL.so.1
0x00007ffff73eec70  0x00007ffff775f74e  Yes (*)     /lib/x86_64-linux-gnu/libgtk-3.so.0
0x00007ffff72832f0  0x00007ffff72ffac6  Yes (*)     /lib/x86_64-linux-gnu/libgdk-3.so.0
0x00007ffff72466a0  0x00007ffff724c8cc  Yes (*)     /lib/x86_64-linux-gnu/libpangocairo-1.0.so.0
0x00007ffff71ed090  0x00007ffff721f854  Yes (*)     /lib/x86_64-linux-gnu/libpango-1.0.so.0
0x00007ffff70c5cc0  0x00007ffff7195327  Yes (*)     /lib/x86_64-linux-gnu/libcairo.so.2
0x00007ffff70890b0  0x00007ffff70a1401  Yes (*)     /lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0
0x00007ffff6ee53e0  0x00007ffff6fee0cd  Yes (*)     /lib/x86_64-linux-gnu/libgio-2.0.so.0
0x00007ffff6e55fd0  0x00007ffff6e87c5e  Yes (*)     /lib/x86_64-linux-gnu/libgobject-2.0.so.0
0x00007ffff6d2ab20  0x00007ffff6db71d2  Yes (*)     /lib/x86_64-linux-gnu/libglib-2.0.so.0
0x00007ffff6be60c0  0x00007ffff6c7048e  Yes (*)     /lib/x86_64-linux-gnu/libX11.so.6
0x00007ffff6bc8080  0x00007ffff6bc8219  Yes (*)     /lib/x86_64-linux-gnu/libwayland-egl.so.1
0x00007ffff6bb9540  0x00007ffff6bbe39d  Yes (*)     /lib/x86_64-linux-gnu/libwayland-client.so.0
0x00007ffff6a96720  0x00007ffff6b4b298  Yes (*)     /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
0x00007ffff69bba20  0x00007ffff6a269af  Yes (*)     /lib/x86_64-linux-gnu/libgstvideo-1.0.so.0
0x00007ffff3259df0  0x00007ffff58c1ac2  Yes (*)     /lib/x86_64-linux-gnu/libwebkit2gtk-4.0.so.37
0x00007ffff14f0240  0x00007ffff27a6312  Yes (*)     /lib/x86_64-linux-gnu/libjavascriptcoregtk-4.0.so.18
0x00007ffff1393240  0x00007ffff139d53a  Yes (*)     /lib/x86_64-linux-gnu/libpangoft2-1.0.so.0
0x00007ffff13141e0  0x00007ffff13648cf  Yes (*)     /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
0x00007ffff1103700  0x00007ffff1295abd  Yes         /lib/x86_64-linux-gnu/libc.so.6
0x00007ffff10be660  0x00007ffff10d4be5  Yes (*)     /lib/x86_64-linux-gnu/libgcc_s.so.1
0x00007ffff10007b0  0x00007ffff108859c  Yes (*)     /lib/x86_64-linux-gnu/libfreetype.so.6
0x00007ffff0fc6290  0x00007ffff0fe45a7  Yes (*)     /lib/x86_64-linux-gnu/libexpat.so.1
0x00007ffff0fbb550  0x00007ffff0fbed61  Yes (*)     /lib/x86_64-linux-gnu/libuuid.so.1
0x00007ffff0f3f240  0x00007ffff0f41187  Yes (*)     /lib/x86_64-linux-gnu/libGLdispatch.so.0
0x00007ffff0ece6c0  0x00007ffff0ee8080  Yes (*)     /lib/x86_64-linux-gnu/libGLX.so.0
0x00007ffff0ec64a0  0x00007ffff0ec7433  Yes (*)     /lib/x86_64-linux-gnu/libgmodule-2.0.so.0
0x00007ffff0eb3460  0x00007ffff0ebe67b  Yes (*)     /lib/x86_64-linux-gnu/libXi.so.6
0x00007ffff0eaa310  0x00007ffff0eac961  Yes (*)     /lib/x86_64-linux-gnu/libXfixes.so.3
0x00007ffff0ea0100  0x00007ffff0ea126d  Yes (*)     /lib/x86_64-linux-gnu/libcairo-gobject.so.2
0x00007ffff0e7bd40  0x00007ffff0e89be4  Yes (*)     /lib/x86_64-linux-gnu/libatk-1.0.so.0
0x00007ffff0e47010  0x00007ffff0e5ce52  Yes (*)     /lib/x86_64-linux-gnu/libatk-bridge-2.0.so.0
0x00007ffff0d63240  0x00007ffff0dcbfc1  Yes (*)     /lib/x86_64-linux-gnu/libepoxy.so.0
0x00007ffff0ce9100  0x00007ffff0cebfb3  Yes (*)     /lib/x86_64-linux-gnu/libfribidi.so.0
0x00007ffff0c21c10  0x00007ffff0cb19b2  Yes (*)     /lib/x86_64-linux-gnu/libharfbuzz.so.0
0x00007ffff0c121b0  0x00007ffff0c12bd1  Yes (*)     /lib/x86_64-linux-gnu/libXinerama.so.1
0x00007ffff0c06360  0x00007ffff0c0cec9  Yes (*)     /lib/x86_64-linux-gnu/libXrandr.so.2
0x00007ffff0bfa660  0x00007ffff0bff1fc  Yes (*)     /lib/x86_64-linux-gnu/libXcursor.so.1
0x00007ffff0bf4140  0x00007ffff0bf4b75  Yes (*)     /lib/x86_64-linux-gnu/libXcomposite.so.1
0x00007ffff0bef1a0  0x00007ffff0befb2b  Yes (*)     /lib/x86_64-linux-gnu/libXdamage.so.1
0x00007ffff0bab550  0x00007ffff0bc9541  Yes (*)     /lib/x86_64-linux-gnu/libxkbcommon.so.0
0x00007ffff0b9f460  0x00007ffff0ba0e4b  Yes (*)     /lib/x86_64-linux-gnu/libwayland-cursor.so.0
0x00007ffff0b8a600  0x00007ffff0b9494e  Yes (*)     /lib/x86_64-linux-gnu/libXext.so.6
0x00007ffff0b7d320  0x00007ffff0b80ba7  Yes (*)     /lib/x86_64-linux-gnu/libthai.so.0
0x00007ffff0ada300  0x00007ffff0b5f4bf  Yes (*)     /lib/x86_64-linux-gnu/libpixman-1.so.0
0x00007ffff0a9a530  0x00007ffff0ac2479  Yes (*)     /lib/x86_64-linux-gnu/libpng16.so.16
0x00007ffff0a910e0  0x00007ffff0a91c37  Yes (*)     /lib/x86_64-linux-gnu/libxcb-shm.so.0
0x00007ffff0a6f640  0x00007ffff0a82660  Yes (*)     /lib/x86_64-linux-gnu/libxcb.so.1
0x00007ffff0a5a120  0x00007ffff0a5eedf  Yes (*)     /lib/x86_64-linux-gnu/libxcb-render.so.0
--Type <RET> for more, q to quit, c to continue without paging--c
0x00007ffff0a4a440  0x00007ffff0a507e2  Yes (*)     /lib/x86_64-linux-gnu/libXrender.so.1
0x00007ffff09cb540  0x00007ffff0a13861  Yes (*)     /lib/x86_64-linux-gnu/libjpeg.so.8
0x00007ffff098c290  0x00007ffff09b6dc1  Yes (*)     /lib/x86_64-linux-gnu/libmount.so.1
[... cut ...]
0x00007fffd7a532a0  0x00007fffd7a59215  Yes (*)     /lib/x86_64-linux-gnu/libdrm_radeon.so.1
0x00007fffd7a48760  0x00007fffd7a4cf72  Yes (*)     /lib/x86_64-linux-gnu/libdrm_amdgpu.so.1
0x00007fffd7a3c400  0x00007fffd7a4014a  Yes (*)     /lib/x86_64-linux-gnu/libdrm_nouveau.so.2
0x00007fffd7a08ff0  0x00007fffd7a256d0  Yes (*)     /lib/x86_64-linux-gnu/libedit.so.2
0x00007fffd79dc780  0x00007fffd79ec19b  Yes (*)     /lib/x86_64-linux-gnu/libtinfo.so.6
(*): Shared library is missing debugging information.

ryancpu, 2023-06-08:
[shared library output.txt](https://github.com/bambulab/BambuStudio/files/11683498/shared.library.output.txt)

bambu123, 2023-06-08:
> [shared library output.txt](https://github.com/bambulab/BambuStudio/files/11683498/shared.library.output.txt)
> 
> easier to read format

Thank you.

bambu123, 2023-06-09:
> > [shared library output.txt](https://github.com/bambulab/BambuStudio/files/11683498/shared.library.output.txt)
> > easier to read format
> 
> Thank you.

Please help debugging with valgrind.
- Install valgrind with apt-get in ubuntu. 
   $ sudo apt-get install valgrind
- execute BambuStudio in valgrind 
   $ valgrind  BambuStudio_ubu64.AppImage

ryancpu, 2023-06-10:
==171745== Memcheck, a memory error detector
==171745== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==171745== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==171745== Command: ./BambuStudio_ubu64.AppImage
==171745== 
==171746== 
==171746== HEAP SUMMARY:
==171746==     in use at exit: 3,943 bytes in 23 blocks
==171746==   total heap usage: 64 allocs, 41 frees, 44,840 bytes allocated
==171746== 
==171746== LEAK SUMMARY:
==171746==    definitely lost: 0 bytes in 0 blocks
==171746==    indirectly lost: 0 bytes in 0 blocks
==171746==      possibly lost: 0 bytes in 0 blocks
==171746==    still reachable: 3,943 bytes in 23 blocks
==171746==         suppressed: 0 bytes in 0 blocks
==171746== Rerun with --leak-check=full to see details of leaked memory
==171746== 
==171746== For lists of detected and suppressed errors, rerun with: -s
==171746== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
[2023-06-09 21:52:38.192639] [0x00007f29349d3980] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/111111/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/11111/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.231: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:171745): GLib-GObject-WARNING **: 21:52:44.274: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.274: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:171745): GLib-GObject-WARNING **: 21:52:44.276: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.276: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:171745): GLib-GObject-WARNING **: 21:52:44.279: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.279: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:171745): GLib-GObject-WARNING **: 21:52:44.281: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.281: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:171745): GLib-GObject-WARNING **: 21:52:44.291: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.291: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:171745): GLib-GObject-WARNING **: 21:52:44.293: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.293: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.396: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.405: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.405: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.406: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.406: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.407: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.407: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.407: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.407: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.429: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.431: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.431: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.647: gtk_window_resize: assertion 'height > 0' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.672: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.672: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.672: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:171745): Gtk-CRITICAL **: 21:52:44.672: gtk_widget_set_size_request: assertion 'width >= -1' failed
free(): invalid pointer
Aborted (core dumped)
me:~/bambu/BambuStudio/build$ ==171749== 
==171749== HEAP SUMMARY:
==171749==     in use at exit: 337 bytes in 2 blocks
==171749==   total heap usage: 6,170 allocs, 6,168 frees, 163,706,692 bytes allocated
==171749== 
==171749== LEAK SUMMARY:
==171749==    definitely lost: 65 bytes in 1 blocks
==171749==    indirectly lost: 0 bytes in 0 blocks
==171749==      possibly lost: 272 bytes in 1 blocks
==171749==    still reachable: 0 bytes in 0 blocks
==171749==         suppressed: 0 bytes in 0 blocks
==171749== Rerun with --leak-check=full to see details of leaked memory
==171749== 
==171749== For lists of detected and suppressed errors, rerun with: -s
==171749== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

bambu123, 2023-06-21:
> Any update on this?

Not yet.

bambu123, 2023-07-03:
@ryancpu We found it related to the X11 display scale. Set scale to 200% will cause this issue. Is that correct?

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.