bambulab/BambuStudio #1869: archimedean chords/octagram sprial changed since 1.6.0 / give us more options for them

moswald14, 2023-05-28:
It used to generate the path based on the center of the model itself, now it bases the center on the separate parts of the model. Give us the option to use either the new way or the old way in the slicer. bonus if it can have a feature to make the archimedean chords printed in a single tool path after printing all the walls of the first layer to experiment to see if it makes a better gradient with multicolor co-extrusion filament

To achieve this, your proposed workflow involves printing the walls of the first layer of the object using the conventional method. Then, after completing the walls, the printer would transition to printing the archimedean chords in a single continuous tool path, allowing the multicolor co-extrusion filament to generate the desired gradient effect.

The purpose of this feature is to experiment and evaluate whether this technique improves the overall quality and aesthetic appeal of 3D printed objects with gradient effects. By focusing on optimizing the tool path for archimedean chords, you aim to explore new possibilities for achieving visually stunning gradients in your multicolor co-extrusion prints.

moswald14, 2023-05-29:
actually, it seems like it doesn't have to print all the walls and then the bottom surface in one path, it seems I get the desired effect when it happens to prints the lines in the same direction. this was just my first idea when confronting the problem.

to add, it's important for co-extruded filament to be extruded in a consistent pattern, either convergent or divergent. it may also be helpful to set the direction of rotation.

I imagine the settings could be like so:

When Archimedean Chords is selected, additional options appear:

- Surface Convergence
1. Default - prints surfaces which ever way is faster.
2. Convergent - prints surface lines from the outside to the inside.
3. Divergent - prints surface lines from inside to outside.

- Monotonic surface (this could
 be inferred by selecting convergent or divergent)
1. Enabled - Prints the surface pattern in the same direction, increasing the number of retractions and travel moves it has to do
2. Disabled - prints the surface pattern in a way that is the faster.

- Pattern Centricity
1. Per model/assembly - 1.5.0 behavior - bases the centricity on the center of the model or assembly
2. Per object - 1.6.0+ behavior - bases the centricity on the center of each object or enclosed wall structure
3. Per plate - bases the centricity on the center of the build plate

[image]
[image]

QingZhangBambu, 2023-05-29:
good point. i will investigate it

QingZhangBambu, 2023-06-13:
#1463

iamkim17, 2025-07-24:
No option for this in the next update?