bambulab/BambuStudio #1871: Docking BambuStudios no longer works

Benderrodrigu3z, 2023-05-29:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Catalina 10.15.7

### Additional system information

2012 MacBook Pro

### Printer

X1C

### How to reproduce

open any stl or 3mf file

### Actual results

a second Bambu Studios opens

### Expected results

overtake the current windows especially when there is nothing in the current window
This is a Giant Ram Hog doing this process, once more not everyone always has there MacOS Dock on the screen in view to know that yall are now doing this, give us an option button to turn this "feature off" for me its a bug because it consumes extra RAM
[image]

### Project file & Debug log uploads

[Screen Shot 2023-05-29 at 8.39.13 AM.png.zip](https://github.com/bambulab/BambuStudio/files/11592187/Screen.Shot.2023-05-29.at.8.39.13.AM.png.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-05-31:
We always use new window to open a 3mf file, but not stl file.
Does the STL model goes to the opened window？

Benderrodrigu3z, 2023-05-31:
The stl does current, this never happened on the older version, I want to dock the icon and only use that one not have 5 of the program open, why are yall making these changes but not making an option in Preferences saying if we want it off or on? My laptop crawls with more then 1 openSincerely,Shane FSent from my iPhone13