bambulab/BambuStudio #1873: Need the Slicing Tolerance option from Curas experimental settings.

tjc2005, 2023-05-29:
Part tolerance is not good, I have multiple objects that won't fit together, however if I use cura on any of my printers and use exclusive mode in cura experimental settings, things work much better.

Explanation here https://support.makerbot.com/s/article/1667418054188

The only way to get it to work for me currently is to make a part smaller or bigger but then that presents it's own set of issues with tolerances.

Can we get these added to bambu studio?

michaeltavel, 2023-11-13:
I agree, this is different from X-Y hole compensation.  Slicing Tolerance takes into consideration the bulging sides of each layer line and adjusts the perimeter of a slice to set the tolerance for each bulging layer perimeter to the correct point for that particular design.  X-Y compensation jusy applies a sizing factor to anything that looks like a hole, which can have the effect of desctroying text or other circular forms that are not actully important for tolerance between parts.

chuck-bz, 2025-03-03:
1.5 years later, anything happen with this? I've used the "Exclusive" slicing setting in Cura on threads in the past and it worked great.

There is also this issue posted on Orca slicer: https://github.com/SoftFever/OrcaSlicer/issues/1186

A lot of information on there. A lot of people not understanding that x-y hole and contour compensation is not the same.

Exclusive slicing is a great one-click setting for threads, inclusive is a great one-click setting for parts that will be sanded smooth.

Seems like this should be right up Bambu Lab's alley.