bambulab/BambuStudio #1874: Default starting gcode causes bed to crash into bottom hard limit if started straight after a large print

Lex-FX, 2023-05-30:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

WIN10

### Printer

X1C

### How to reproduce

Finish a print that is more than 150mm in code height. This will cause the end gcode to leave the bed 2mm off the bottom gcode max travel:

M17 Z0.4 ; lower z motor current to reduce impact if there is something in the bottom
{if (max_layer_z + 100.0) < 250}
    G1 Z{max_layer_z + 100.0} F600
    G1 Z{max_layer_z +98.0}
{else}
    G1 Z250 F600
    G1 Z248

Or, just move the bed to the bottom of the Z travel.

Remove the print, then start another print. 

### Actual results

As the title says, the start gcode has the bed move 12mm down as it's first move which is beyond the hard limit so it crashes and stalls the motors. Yes the current has been reduced, but it's still a physical crash. The 12mm move is a relative move, so regardless of whether it's been homed or not, it's going to crash the bed if the bed is at the bottom of the chamber. 

G91
M17 Z0.4 ; lower the z-motor current
G0 Z12 F300 ; lower the hotbed , to prevent the nozzle is below the hotbed
G0 Z-6;

So what has to be remembered before or after every print is to home the printer, or at least move it off the bottom. 

### Expected results

So the question is, can a simple G28 command be used for the X1C? I can't find anywhere through BambuLab what the actual home gcode is for the X1C. It's not just a simple drive to axis home to zero plus maybe a 2nd soft bounce home. There's a Z bed vibration as well. 

If the G28 command can't be used directly, can someone please post the gcode to replicate the homing routine when you press the home button on the printer or BBStudio? That way I can insert this as the first start move. 

As an aside, I'm not really sure of the logic behind having a relative move Z down as a start gcode for a printer with the bed on the Z. The comments in the code say to make sure the nozzle is above the bed, but in what situation is the nozzle going to be near the bed during a start? For my Troodon/Voron with the Z on the gantry I get it coz the gantry droops when the motors turn off so my first move I want to always be up. But for the X1C, I don't get it. 

### Project file & Debug log uploads

[Pillow Indenter.zip](https://github.com/bambulab/BambuStudio/files/11594714/Pillow.Indenter.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/11594715/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

moebis, 2023-06-05:
I have complained about this issue for over half a year and nothing has been done to fix it. I can't believe they still haven't fixed the z axis from bottoming out issue. If you ever have a large print, and the bed ends up on the bottom, if you don't move it up a little, it will grind, rattle and shake as it tries to go lower then the bottom. This also happens (more often) if you shut the printer down, and bring it back up for a new print with the bed near the bottom, because the start g-code always has it test Z by dropping it 1-2cm. You would think the stepper motor would have feedback or sensorless homing. I swear my z belt sounded like it was going to break this weekend.

Lex-FX, 2023-06-05:
I thought I'd be clever and just reverse the direction of the moves in the start gcode which is accessible in the nozzle/machine settings. Worked great until I started a print after having homed the bed. It seems the bed is left within 12mm of the head after it finishes homing coz the next thing I see is the printer driving the bed up into the head, flexing the rods, luckily not breaking anything. 

So switched it back and just have to remember to home it every time. 

I haven't gone through all of the gcode line by line to see if I can identify the homing commands. Couldn't see G28 when I had a brief look, but by the looks of it, it there might be explicit movement commands used instead. 

It's also odd the there aren't soft limits. Even after it's homed, if you press the bed down button too quickly too many times, you can drive the bed into the hard stop.

Lex-FX, 2023-06-05:
I went over the start code properly and there is actually G28 mentioned although for specific axis at different times. 
Anyway, I tried just putting plain G28 as the first command in the start gcode then sent a print. (For those that don't know, it's in the printer nozzle settings section).
All good, no crashes. Doesn't matter where the bed is, it homes first which is a move towards the nozzle. Easy fix