bambulab/BambuStudio #1876: Bug with prime tower when print sequence is set to “by object”

nkacena, 2023-05-30:
### Bambu Studio Version

1.5.0.61

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

x1c

### How to reproduce

add multiple objects to plate (each object has 2 or more colors, AMS enabled)
select print by object
ensure prime tower is enabled
slice

prime tower is missing. 

### Actual results

prime tower is missing after executing above steps

### Expected results

prime tower (or multiple prime towers) should be created when printing multiple multi-color objects and print sequence is set to "by object"

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-05-31:
@nkacena 
If  is print "by object", a prime tower will only be generated when there is only one object. If there are multiple objects, no prime will be generated.

nkacena, 2023-05-31:
@zhimin-zeng-bambulab could you add the option to add a prime tower when printing multiple multi-color objects? I routinely print multiple multi-color prints but without a prime tower they're usually not as crisp on the transitions so I'm then forced to print one by one, taking much more time.

zhimin-zeng-bambulab, 2023-05-31:
Due to the code framework, printing multiple objects with multiple colors "by object" is not supported currently.  You could print multiple objects with multiple colors  "by layer".

closed 2023-06-05 by zhimin-zeng-bambulab

x-caliper, 2025-03-21:
I would find this extremely useful as well.

vanowm, 2025-06-29:
Currently a work around is to create a small `cube` for each color and merge them with the part that uses multiple colors. Make sure these cubes are on top of the assembly list. This way after each color change it will start printing these cubes first.
The downside of this method is it requires manually adjusting the size of the cubes and will increase the required boundaries per part.