bambulab/BambuStudio #1877: BambuSource.dll crashes Microsoft Teams

ruffy91, 2023-05-30:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 22H2

### Printer

Bambu Lab P1P

### How to reproduce

1. Install Bambu Studio
2. Install New Teams
3. Open New Teams
4. New Teams Crashes after 15min with the following application error:
'''
Faulting application name: ms-teams.exe, version: 23140.303.2111.9674, time stamp: 0x646c4864
Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x642f6637
Exception code: 0xc0000005
Fault offset: 0x0000000000087f01
Faulting process id: 0x0x1DF4
Faulting application start time: 0x0x1D992C42F9CE791
Faulting application path: C:\Program Files\WindowsApps\MSTeams_23140.303.2111.9674_x64__8wekyb3d8bbwe\ms-teams.exe
Faulting module path: BambuSource.dll
Report Id: 27667cf2-1b00-4feb-a047-54f3bd2708e0
Faulting package full name: MSTeams_23140.303.2111.9674_x64__8wekyb3d8bbwe
Faulting package-relative application ID: MSTeams
'''

### Actual results

New Teams loads BambuSource.dll
New Teams crashes after exactly 15 Minutes

### Expected results

New Teams doesn't load BambuSource.dll
New Teams doesn't crash

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/11597171/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-06-01:
Would you please provider the link to the installer of New Terms?

ruffy91, 2023-06-01:
The New Teams client can be downloaded from here: https://statics.teams.cdn.office.net/production-windows-x64/23140.303.2111.9674/MSTeams-x64.msix

bambu123, 2023-06-05:
> The New Teams client can be downloaded from here: https://statics.teams.cdn.office.net/production-windows-x64/23140.303.2111.9674/MSTeams-x64.msix

Could you please help debug the software?
1. Download the dump.txt attached, and rename it to dump.reg
2. Double-click the downloaded file to import some registries
3. Run Team and crash
4. Open the folder '%APPDATA%\BambuStdio\log' where the minidump is created
5. Send back the DMP files in it

[dump.txt](https://github.com/bambulab/BambuStudio/files/11648753/dump.txt)

VeryKross, 2023-07-22:
I am also seeing this affecting the Camtasia Snagit2023 product. Snagit will only run for a short while before it crashes and when I investigated, this is what I found:

Faulting application name: SnagitCapture.exe, version: 23.2.0.30713, time stamp: 0x64ad9c74
Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x64a7650e
Exception code: 0xc0000005
Fault offset: 0x00000000000876d6
Faulting process id: 0x0xB5C0
Faulting application start time: 0x0x1D9BCBBA2201359
Faulting application path: C:\Program Files\TechSmith\Snagit 2023\SnagitCapture.exe
Faulting module path: BambuSource.dll
Report Id: 13de1f4f-5c1b-423c-a0b3-1b1ba07e103f

bambu123, 2023-07-24:
> I am also seeing this affecting the Camtasia Snagit2023 product. Snagit will only run for a short while before it crashes and when I investigated, this is what I found:
>
> Faulting application name: [...]

These informations are not enough.
Please help debug with the steps above.

Replace 'ms-teams.exe' with 'SnagitCapture.exe' in the dump.txt registery.

KD7LRJ, 2023-07-24:
Here's a dump from a crash of SnagitCapture.

[SnagitCapture.exe.9696.zip](https://github.com/bambulab/BambuStudio/files/12143412/SnagitCapture.exe.9696.zip)

bambu123, 2023-07-24:
> Here's a dump from a crash of SnagitCapture.
> 
> [SnagitCapture.exe.9696.zip](https://github.com/bambulab/BambuStudio/files/12143412/SnagitCapture.exe.9696.zip)

To anylize the .dmp file, the ntdll.dll file in your system is needed.
Please send me this file.
C:\Windows\System32\ntdll.dll

bambu123, 2023-07-24:
@KD7LRJ 

It may be sovled by removing the 'Bambu Camera Source' from registry.
The whole path is:
HKEY_CLASSES_ROOT\CLSID\{860BB310-5D01-11d0-BD3B-00A0C911CE86}\Instance\{D47E3D12-D0FA-4D0E-B6F6-FD879AB452C1}

KD7LRJ, 2023-07-24:
[ntdll.zip](https://github.com/bambulab/BambuStudio/files/12147929/ntdll.zip)

KD7LRJ, 2023-07-24:
@bambu123

Renaming the entries in the registry path listed ^ seems to have stopped the problem with SnagitCapture.exe exiting after a few minutes.
[image]

VeryKross, 2023-07-25:
I'm seeing the same result - renamed the entries in registry and Snagit is now stable.

chrdavis, 2023-08-13:
According to issue #1748 this is fixed in 1.7.2

closed 2023-10-24 by bambu123

magic-happenz, 2024-01-04:
I am on v1.8.2 and it is still causing the new Teams client to crash. The registry entry is fixing the issue but I don't get how its causing the app to crash?

bambu123, 2024-01-05:
> I am on v1.8.2 and it is still causing the new Teams client to crash. The registry entry is fixing the issue but I don't get how its causing the app to crash?

maybe you need a reboot

mydevpeeps, 2024-03-03:
I am still getting this error as of build 1.8.4.51 with MS Teams.

`Faulting application name: ms-teams.exe, version: 24004.1309.2689.2246, time stamp: 0x65c6ca17
Faulting module name: BambuSource.dll_unloaded, version: 0.0.0.0, time stamp: 0x64faff69
Exception code: 0xc0000005
Fault offset: 0x000000000008ab16
Faulting process id: 0x1154
Faulting application start time: 0x01da6cfe30c398d1
Faulting application path: C:\Program Files\WindowsApps\MSTeams_24004.1309.2689.2246_x64__8wekyb3d8bbwe\ms-teams.exe
Faulting module path: BambuSource.dll
Report Id: cb19f5c7-9f7a-4b95-9d1e-3b4326795bb3
Faulting package full name: MSTeams_24004.1309.2689.2246_x64__8wekyb3d8bbwe
Faulting package-relative application ID: MSTeams`

mydevpeeps, 2024-03-03:
[image]

mydevpeeps, 2024-03-03:
changing the registry didnt seem to matter for me.

[image]

I just disabled all cameras since I don't have one anyway on my desktop.

[image]