bambulab/BambuStudio #1878: Wrong filament is used for printing (race condition at AMS startup)

mzanetti, 2023-05-30:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu Linux

### Additional system information

Shouldn't matter. I don't think this is related to the platform the application is running at all, actually this is likely even a bug in the printers firmware. 

### Printer

X1C with AMS

### How to reproduce

* On the AMS, select the option to re-read filaments upon power on.
* Sync filament settings to Bambu studio so app and printer/AMS are in sync.
* Turn off the printer
* Reopen Bambu Studio, slice something, select a filament which is in AMS slot 1 (may also work with others, but since it's a race condition, it's easier to reproduce with slot 1 - even though I also had this with slot 3. Just don't select the one in slot 4)
* Hit the print button to open the "Send to printer" dialog. It should show that the printer is offline at this point.
* Now turn on the printer and quickly return to Bambu Studio
* Hit the refresh button repeatedly until the printer shows up. It should now auto-match the AMS slot with your filament selection
* Send the print as quickly as possible

### Actual results

The printer will start printing (bed leveling etc). The display of the printer will show the correct filament color which you have selected. However, because the AMS will still switch through all the slots in order to re-read the filaments, it will actually print with filament in the 4th AMS slot.

### Expected results

Well, obviously, the expected result would be that the printer uses the correct filament slot.

### Project file & Debug log uploads

[debug_Tue_May_30_14_11_16_148564.log.zip](https://github.com/bambulab/BambuStudio/files/11599921/debug_Tue_May_30_14_11_16_148564.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-06-01:
Maybe slot 4 is read first.

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl