bambulab/BambuStudio #1879: Crashing when opening large 3mf file

drpeppershaker, 2023-05-30:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Monterey 12.5.1

### Additional system information

MacBook Pro (16-inch, 2021)
Apple M1 Max
64 GB

### Printer

Bambu X1C

### How to reproduce

Attempt to open 3mf file. 
Wait for Bambu Studio.
Program does not open

### Actual results

Loading screen appears, loads partially then freezes.
Must force quit to close.

### Expected results

Program should open

### Project file & Debug log uploads

File too large to upload. Zipped 3mf file is approx 140 MB. Here is a google drive link
https://drive.google.com/file/d/1wsM99CWJHx8BACgS29BlthaIcibQIqZ3/view

[log.zip](https://github.com/bambulab/BambuStudio/files/11605441/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-06-05:
@drpeppershaker Are there any Crash logs releate to BambuStduio in the Console App?

bambu123, 2023-06-05:
It is really very slow, but finally will be loaded.
We will optimize the process on macOS.

bambu123, 2023-07-11:
Close as fixed in the 1.7 beta version.

closed 2023-07-11 by bambu123