bambulab/BambuStudio #1881: 3mf files exported on my PC and shared to printables are unusable by multiple other people

GekoPrime, 2023-05-31:
### Bambu Studio Version

1.6.2

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

i5-5300U
8 GB RAM

### Printer

X1CC

### How to reproduce

I'm unable to reproduce directly

### Actual results

Multiple users on Printables have reported that 3mf files I've exported from Bambu Studio are warped are show errors when they open then in their instances of Bambu Studio

Here's the one of the comments that includes a screenshot:
https://www.printables.com/model/466883-pastamatic-filament-spool-winder-for-bambu-lab-x1c/comments/892811

### Expected results

I expect that when I export files from Bambu Studio other people will be able to use them

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/11608576/log.zip)
[Project.zip](https://github.com/bambulab/BambuStudio/files/11608579/Project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-05-31:
@GekoPrime 
hello, thanks for reporting this issue

do you mean this error after slicing?
[image]

GekoPrime, 2023-05-31:
Yes, somehow I don't see that error on my own installation.

I'm also concerned about this screenshot, where the model appears to be skewed or distorted?

https://media.printables.com/media/comment_images/92/2f82c4-57bf-445e-9eb6-019424cd5351/thumbs/outside/640x480/png/image.webp

lanewei120, 2023-06-01:
> 

the error is just because we use the convex hull to check the gcode path conflicts
as there is a exclude area on the left bottom,
the convex hull intersects with it

we will try to improve this logic

for the screenshot, I don't think it has any skew or distortion
it is just because the camera view

GekoPrime, 2023-06-02:
Thanks, that makes sense!