bambulab/BambuStudio #1884: Cut->Add Connector: Issue with Greyed Out Dowel Shape Option and Confusing Dropdown

leandrolima-nyc, 2023-06-01:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13.2.1

### Additional system information

Mac Silicon chip

### Printer

Bambu Lab X1C

### How to reproduce

When performing object cutting and adding a connector, there is an issue where the dowel shape option becomes greyed out. While the selection remains functional, the dropdown itself has a transparent background, leading to confusion and difficulty in visualizing the available options.

### Actual results

[image]

### Expected results

Firstly, the shape dropdown should not be greyed out, ensuring that all available options are easily accessible. Secondly, the dropdown menu itself should have a clearly visible appearance, eliminating any confusion and enabling users to select their desired shape with ease.

### Project file & Debug log uploads

Not required.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-06-14:
Thank you for reporting this issue, ​​we will fix it in the next version.

zhimin-zeng-bambulab, 2023-08-18:
You can have a try in the latest version

closed 2023-08-18 by zhimin-zeng-bambulab