bambulab/BambuStudio #1885: Issue syncing PLA Marble with AMS

markdesouza, 2023-06-01:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 13.3.1

### Printer

Bambu X1 Carbo Combo

### How to reproduce

1. Set the filament type to "Bambu PLA Marble" either directly on the printer itself or in the devices tab of bambu labs
2. Under the preview tab, click on the sync AMS function, and a dialog will appear saying unrecognised filaments and set the filament type as "Generic PLA"

### Actual results

After sync completes, bambu labs records the filament type as  "Generic PLA"

### Expected results

It should correctly retrieve the saved filament type "Bambu PLA Marble"

### Project file & Debug log uploads

No needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-06-05:
It is really not compatible with X1C 0.2 nozzle.
Are you using this printer preset?

bambu123, 2023-06-14:
@markdesouza Please check the printer preset.

bambu123, 2023-08-23:
@markdesouza The new version of BambuStudio was released. Can you help confirm that the new version still has this problem?

SaltWei, 2024-01-18:
> @markdesouza The new version of BambuStudio was released. Can you help confirm that the new version still has this problem?

@bambu123 Yes, I can reproduce.
This is caused by "Bambu PLA Marble" not compatible with 0.2 nozzle.
0.2 nozzle can not be used to print PLA marble which is high possibility to clog.
So the world in popup windows should be changed to handle this situation.