bambulab/BambuStudio #1887: More detailed filament remaining data

McfarlaneRob, 2023-06-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
N/A

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Would love to see either a remaining weight or percentage shown on the spool indicator for the AMS to be able to better gauge amount left on a roll.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
The current alternative is just to approximate based on the fill bar indicator currently used.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

I feel that this could help a lot with managing the allocation of prints, as well as just being a nicer way to see how much is left on a roll, especially when using white filaments.

Jdplays, 2023-08-09:
It would also be nice if when you sliced a plate, it would tell you if you will have enough filament for the print.

frasier2k, 2026-01-18:
Please implement a feature where the slicer tells me whether the remaining filament is sufficient!