bambulab/BambuStudio #1889: Dark mode resets after RDP session to computer with Bambu running

tangcla, 2023-06-03:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Start Bambu Lab on computer (in dark mode)
Connect to computer via remote desktop (over network)

### Actual results

Dark mode checkbox is unticked.
[image]

### Expected results

Dark mode should remain as this is not configured by system settings.

### Project file & Debug log uploads

This bug is not project specific.
[log.zip](https://github.com/bambulab/BambuStudio/files/11640009/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-06-05:
Hi.
Is this problem reproducible every time?

tangcla, 2023-06-05:
> Hi. Is this problem reproducible every time?

Yes, each time after I remote desktop into my PC, it reverts back to the light theme.

TackleBerryCrash, 2023-06-06:
I can also confirm this happening every time. I was wondering why studio color scheme was behaving wildly but it was not random after all. Thanks for solving this mystery...
Closing the app and opening solves the UI problem but it reverts to the light mode.

DMFirmy, 2023-06-09:
I too confirm this issue. You can go directly into the settings and re-apply dark mode without having to first close and re-open the application, but as noted, light mode gets set after connecting via RDP, and until reset or closed the application goes all wonky with its coloring.

walterwongbbl, 2023-06-13:
Hi.
We have optimized the dark mode in the next version.

tangcla, 2023-06-14:
Note that this also happens when another user signs into the computer (and then returning back to original user).

walterwongbbl, 2023-08-02:
Hi.
Please use the latest version of Bambus Studio

closed 2023-08-02 by walterwongbbl