bambulab/BambuStudio #1890: The UI isn't compatible with accessibility standards and does not work with a screen reader

AmosMiller, 2023-06-04:
### Bambu Studio Version

all

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10/11

### Printer

X1 Combo

### How to reproduce

I'm a blind maker and the Bambu Studio application UI isn't accessible with a screen reader. I'm not experienced with the accessibility API in Windows but hope that someone can look at it and at least see if the main features can be labled and accessible so that I can connect to the printer, slice files I get from my designers, and print. Happy to help / show / explain the problem

To repro you can start narrator on Windows and touch the UI, you will notice that nothing is voiced other than the main tabs of the UI.

### Actual results

only the tabs are voiced

### Expected results

the screen reader (e.g. Narrator) should voice all the menues, UI objects and user interactions.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

aBlindReefer, 2023-08-22:
I would like to also add that the same issue exists using Voiceover on Mac. Some of the initial setup options work with the screen reader but once you get to the actual slicer the only verbal announcements I get are “button” and “image”. There is no description of what the buttons function is, and any drop-down selection menu is only called an image with no description of what the menu is or what the contents of the drop-down are. Cura seems to work with voiceover on Mac and I understand this to be built off of Cura? So, it should be an easy-ish fix.

ThomasAtBBTF, 2023-08-26:
For me, it seems that the problems with accessibility of Bambu Studio using ScreenReaders starts with complete lack of keyboard accessibility. 
There is no way to use the keyboard to change focus from one control to another inside the "dialogs" / panes.
Using accessibility test tools like inspect64 from Windows shows that the controls themselves are exposing quite good information via UI Automation or MSAA. (Fro sure, this can be improved. But it is much better than what is available in the AnkerMake slicer)
The problem is easily testable by any developer of yours....
Just try to use the application without a mouse and you will see the problems immediately!

funkonaut, 2023-12-30:
Hi I work with a lot of visually impaired students teaching them programming through 3D printing and OpenSCAD with Prusa mini printers. If @bambu123  took this issue more seriously then Prusa it would definitely be grounds for ditching the minis and picking up some p1p's or p1s's (love the core xy at the price point...)  We worked with over 1,200 BVI students from kinder to 18+ last year who would appreciate having the independence and autonomy of screen reader access without all the background knowledge required to use a CLI (the only fully accessible solution currently is using prusa-slicer cli). Bambu would then be the first fully accessible GUI slicer (something that has been wished for for so long) ... I am happy to showcase some students work with you all and I am sure they would love to talk to someone on your team to share their work and tell you how much using 3D printing to create independently has meant to them!

h2359, 2023-12-30:
I agree completely with this bamboo studio is extremely inaccessible. I can get around somewhat using object navigation with nvda and the OCR functions, but we should not have to do this to be able to use the program. We should be able to navigate with arrows in the tab key but yes, it is extremely inaccessible. Also, there are a lot of elements that are unlabeled, that should have labels on them

aBlindReefer, 2024-01-31:
Linking an interesting story below. The sound of effort this person (who is blind) put into this seemingly simple project is just nuts. Sure would be nice not to have to go through all these hoops!

https://3dprintingindustry.com/news/blind-man-develops-new-ai-based-design-workflow-for-vision-impaired-3d-printing-227986/

h2359, 2024-01-31:
It is very sad but I guess at least someone is trying to do something to fix the issue. It seems that the developers are not willing to fix anything.Sent from my iPhone

https://3dprintingindustry.com/news/blind-man-develops-new-ai-based-design-workflow-for-vision-impaired-3d-printing-227986/

ClovernookQuentin, 2024-12-12:
This is still an issue, could we get a fix, or could Bambu Lab work with Simplify3D to get Bambu printers properly accessible on their slicer? It is an issue that, if fixed, would help to further equitable education. 3D printing is starting to be used for tactile representations of complex systems due to it's versatility and price point. I would suggest starting with the ability to check auto support, use auto orient and slice without using a mouse and monitor.
[https://repository.rit.edu/jsesd/vol23/iss1/6/#:~:text=3D%20prints%20such%20as%20DNA,the%20printing%20and%20distribution%20service.](https://repository.rit.edu/jsesd/vol23/iss1/6/#:~:text=3D%20prints%20such%20as%20DNA,the%20printing%20and%20distribution%20service.)
This links to an abstract about a nonprofit organization that crowdsources 3D printing for the blind and low vision community, they are still actively running and have over 100 people volunteering their time and 3D printers to further accessible interpretations. Along with that, a small makerspace in Nairobi, Kenya has ran through more than 80 spools in less than a year with 2 3D printers to provide local schools for the blind with more than 2000 models for reading enrichment and STEM lessons. With all of this in mind, I believe that this is a more than worthwhile endeavor to give people the ability to properly make their own accessible artifacts.

RDMurray, 2026-03-26:
Still no accessibility, not even for simple buttons like loading filament. All I can do is enable developer mode and hope that enough of the protocol has been reverse engineered. Can't even rebuild Bambu Studio with accessibility fixes because the network plug-in won't load. Bambu makes great hardware, why do you have to put roadblocks in the software?

h2359, 2026-09-23:
As of today, the same issues still exist I won't go into further explanations of what needs to be done because this has been specified here in this issue already what I will say though is that fixing Accessibility issues in your software and printers is only going to help you business wise not hurt you because then more people will want to buy your products because they will be usable by everyone and not just the sided population also it would be nice if bamboo lab could let us know if we are being heard because we have not had anyone comment on this issue yet from bamboo labs Tech Support team