bambulab/BambuStudio #1892: Directly Import colored 3D files

xuhao1, 2023-06-04:
**Is your feature request related to a problem? Please describe.**
When I trying to import STEP, obj or 3MF with color/texture. It will loss color in Bambu Studio

**Describe the solution you'd like**
When import these 3D file, the color/texture will be keep, so we can directly edit the color/texture in 3D software like Fusion360, and use bambu X1CC to print it. That will benefit a lot to use Bambu create impressive 3D arts.

**Describe alternatives you've considered**
No

**Additional context**
No

lanewei120, 2023-07-10:
thanks very much for this suggestion

we will evaluate this solution and give the feedback later

xuhao1, 2023-07-23:
@mypillowsateyou Amazing work you have done!

lanewei120, 2023-07-26:
@mypillowsateyou 
the most important thing is to tranfrom the the vertex color to face's color,
so currently it seems you set the color of the face to either the most frequently used color among all its vertices or to the color of a specific vertex?

mypillowsateyou, 2023-07-26:
@lanewei120
yes exactly, the script uses the most frequent color of the face's vertices, otherwise it will just use the color of the first vertex if all three are different colors, for me this was good enough

tinogithub, 2023-10-20:
@mypillowsateyou 
Sounds like an approach I would like to try however I run in an error running your script.
In line 46: IndexError: list index out of range

Before I begin changing your code I would like to know if I started correctly.
1. I used Blender 3.4 to export the standard monkey "Suzanne" model to an obj.
2. Imported into Bambu Lab 1.7.7.89
3. Saved and later extracted the .3mf-file
4. In your script I referenced the model-file to the extracted subdirectory /3D/Objects
5. Loaded and started your script in blender

Looking at the error I wondered that your script says in line 46 
r, g, b = row[4], row[5], row[6]

however i seem to cound just 4 rows i.e.:
v 0.021875 0.008203 0.038281

Maybe I am misundestanding something?

Thank you for help

creechoftheeast, 2023-12-27:
This would be a great addition. Currently, I'm using nozzleboss addon for blender to export gcode for multicolor prints.

VickyTGAW, 2024-01-13:
> 
> Looking at the error I wondered that your script says in line 46 r, g, b = row[4], row[5], row[6]
> 
> however i seem to cound just 4 rows i.e.: v 0.021875 0.008203 0.038281
> 
> Maybe I am misundestanding something?
> 
> Thank you for help

I found when going to the **Export->Wavefront (.obj)** option in Blender, I had to check the **Color** under the **Geometry Options** section.  Then the V lines in the resulting .obj file included the extra data:

v -4.616998 3.848527 15.021108 0.7608 0.7412 0.4314

ScottFerg56, 2024-01-26:
+1 for support for recognizing material/color information already present in imported 3D files, especially in my case, OBJ/MTL and STEP. Transferring the actual colors is not so important to me as simply recognizing the different materials in the geometry and allowing them to be assigned from the colors available in the AMS. Thanks!

THE-SIMPLE-MARK, 2024-01-29:
> +1 for support for recognizing material/color information already present in imported 3D files, especially in my case, OBJ/MTL and STEP. Transferring the actual colors is not so important to me as [...]

Exactly, I'm facing the same problem as well.

I want to add multi color support (painting included) to my online model viewer but I don't want to write my own logic to interface with Bambu's own propriatery spin-off of the 3mf standard. This would be sooo easy to do if Bambu would at least check for any colors and at least separate them in the editor once you import it so filaments can be assigned afterwards. I'm currently looking for any easier methods than the python script mentioned above in combination with using OBJs.

I already brought this up with to a couple of people over at the Bambu Lab Official discord community in the #mqtt channel, so you'll find answers there. If we ever get any, that is.

mypillowsateyou, 2024-02-03:
> +1 for support for recognizing material/color information already present in imported 3D files, especially in my case, OBJ/MTL and STEP. Transferring the actual colors is not so important to me as [...]

I saw your comment and felt inspired to make a better workaround than my previous one. I learned a lot about Blender geometry nodes since then and came up with a way to extract a manifold shell of each material of an object. You can see the result of this extraction here:

[image]

The geometry nodes modifier is pretty simple to use, just specify a shell thickness in mm and the material slot you want to extract. There is a calculation which finds the distance internally to the other side of the geometry when raycast from each vertex position, and the thickness is capped at half of that so the shell doesn't protrude out the other side of the geometry. Here is some example geometry generated from a black and white version and a many color version of a random shape:

[image]

.blend and .3mf files included for you to play with, hope others find this useful as well. I think I will also work on some geometry nodes that takes a textured model and generates a copy with new materials assigned to each face based on the color of that face, and limited by some number of materials like 4, 8, 16 depending on how many AMS slots you have. Combine that with the geometry nodes I included here and you would be able to 3D print any manifold textured model with relative ease

Let me know how your experiments go :)

[separate model by materials.zip](https://github.com/bambulab/BambuStudio/files/14150007/separate.model.by.materials.zip)

ScottFerg56, 2024-02-03:
@mypillowsateyou I don't use Blender. I'm actually creating OBJ and STEP files from scratch in my own program. I'll likely wind up creating 3MF as well, before this is done. That Bambu uses 3MF for its project files suggests I could write compatible files with usable color content. If that process were documented somewhere, it would certainly help. Part of my justification for buying the X1C was to make sure the model-creating program I'm developing was useful on the best printer on the market. Presently, that goal is blocked!

THE-SIMPLE-MARK, 2024-02-03:
> @mypillowsateyou I don't use Blender. I'm actually creating OBJ and STEP files from scratch in my own program. I'll likely wind up creating 3MF as well, before this is done. That Bambu uses 3MF for [...]

Even after Bambu's recent efforts to standardize their 3mf, it's still not up to standard. Instead of using color groups as defined by the 3mf consortium, they attach a color property to each triangle.

I have yet to make my logic for converting STEP/OBJ files to 3mf, but this is something I plan to do in the next couple of weeks. Thankfully their method of storing the color seems pretty standard. - They have a project config XML which stores data about the filaments in the project in a very unconventional way (each property has an array as the value, where each array item refers to a filament configured in the project).
- Then, each `.model` file is an XML containing the triangles that make up that model. Each triangle can contain 2 properties (the names of which I don't remember). One is a simple number from 1-16 indicating the filament for that triangle based on the project XML config I talked about above. Then there's another property which contains a string which I presume is for keeping track of the brush strokes of that triangle, I have yet to determine the format of that one.

mypillowsateyou, 2024-02-04:
> > @mypillowsateyou I don't use Blender. I'm actually creating OBJ and STEP files from scratch in my own program. I'll likely wind up creating 3MF as well, before this is done. That Bambu uses 3MF for its project files suggests I could write compatible files with usable color content. If that process were documented somewhere, it would certainly help. Part of my justification for buying the X1C was to make sure the model-creating program I'm developing was useful on the best printer on the market. Presently, that goal is blocked!
> 
> Even after Bambu's recent efforts to standardize their 3mf, it's still not up to standard. Instead of using color groups as defined by the 3mf consortium, they attach a color property to each triangle.
> 
> I have yet to make my logic for converting STEP/OBJ files to 3mf, but this is something I plan to do in the next couple of weeks. Thankfully their method of storing the color seems pretty standard.
> 
> * They have a project config XML which stores data about the filaments in the project in a very unconventional way (each property has an array as the value, where each array item refers to a filament configured in the project).
> * Then, each `.model` file is an XML containing the triangles that make up that model. Each triangle can contain 2 properties (the names of which I don't remember). One is a simple number from 1-16 indicating the filament for that triangle based on the project XML config I talked about above. Then there's another property which contains a string which I presume is for keeping track of the brush strokes of that triangle, I have yet to determine the format of that one.
> 
> These are all the things I found which are not up to the 3mf consortium standard, every other information is available on their site, so converting to a .model and then making custom a custom drawing utility shouldn't be that hard.

Regarding the formatting of that string, it does track the brush strokes by describing a tree for each triangle of the mesh, and at the leaves of that tree lies the filament code to be used for that recursed triangle. When you paint colors on a mesh it is creating this tree on the fly and if you zoom in you can see everything resolves down to triangles (not geometry triangles, just for which filament to use at that area)

Looking at this triangle as an example, the upper right half of this face:

[image]

For this triangle the string is "4C4C44C44C44C4344C34C4C34C434C4344C3"

It was a while back when I wrote my original script but I determined it follows these rules:
a. the string recursion seems to happen reading the string backwards
b. "3" means recurse the current triangle
c. other values assign a filament, according to these codes ['4', '8', '0C', '1C', '2C', '3C', '4C', '5C', '6C', '7C', '8C', '9C', 'AC', 'BC', 'CC', 'DC']. Note that the string is parsed backwards but the code remains forward, i.e. 4C in the string DOES mean filament code 4C, don't flip a C4 into filament code 4C
d. in the recursion (which divides each triangle into 4), the middle triangle is first, then each corner triangle, not sure what the absolute order is, could just be the order of vertices of the original triangle

[image]

If we walk through the string backwards we can see the logic in the image, this string is for the upper right triangle of the face of the cube we are looking at, in this case I had yellow filament assigned to 4C and green to 4, and the order of recursion from this perspective seems to be middle triangle, top left, top right, bottom right:

a. 3, recurse the current triangle
b. 4C, the middle triangle uses filament code 4C
c. 4, looks like the top left triangle uses filament code 4
d. 3, recurse the top right sub triangle
e. 4, middle triangle uses filament code 4
f. 4C, top left triangle uses 4C
g. 3, recurse top right triangle
h. 4, middle is green
i. 4C, top left is yellow
j. 3, recurse top right
k. 4C, middle yellow
l. 4C, top left yellow
m. 3, recurse
n. 4C
o. 4
p. 3
q. 4
r. 4C
s. 4
t. 4C, the first bottom right triangle which finally completes the recursion, the next code is for the triangle layer above
u. 4, bottom right completing this layer
v. 4C, bottom right completing layer
w. 4, 
x. 4C
y. 4C, completing the tree

Probably would have helped if I used 16 different codes to help identify the triangles haha, hopefully you follow the logic and if not I can try to create a better example

So in my script above, I decided not to go with this level of complexity and just mapped each mesh triangle to a single filament code, instead of a long string with all this recursion there is only a single code for one layer and no recursion

THE-SIMPLE-MARK, 2024-02-04:
> Regarding the formatting of that string, it does track the brush strokes by describing a tree for each triangle of the mesh, and at the leaves of that tree lies the filament code to be used for that [...]

Sorry for the amount of questions, but can you explain more about the simpler method you talked about the end of your message? If I understand it correctly, the method used by Bambu uses the string method so the model itself does not need more triangles to be added.

I assume what you do is actually add those triangles which (which the Bambu method would "virtually" iterate through) and just add the property to that triangle which assigns the filament color, correct?

SaltWei, 2024-02-05:
Thanks everyone! This is really really a good suggestion and we had already noticed this feature from internal tester. Manual coloring is really time-consuming.
We will investigate how to keep color of obj file firstly. And then step file. Because step file is imported by OCCT lib. Will also investigate whether occt support import color information.

mypillowsateyou, 2024-02-05:
@THE-SIMPLE-MARK no problem the more we all understand and put out there the more likely we get an integrated solution from BL

Yes exactly, it looks like the string is pretty much a way to store the color information for each mesh triangle in a highly compressed manner compared to either creating new geometry or generating a UV mapped texture

For the second question, the script I wrote doesn't add new mesh triangles, just fills out the recursing filament slot property of each existing triangle in the .model file, based on the vertex colors of each triangle in a source .obj file. In the script only a single filament slot code is assigned to a triangle from the array mentioned before with "4" corresponding to slot 1, "8" to slot 2, "0C" to slot 3, etc instead of doing any recursing

With this method the coloring resolution only goes as deep as there are polygons in the mesh, each mesh triangle has one filament assigned

I just found out something else as well, if you have an object with multiple parts where each part is assigned a different color through the Objects hierarchy, the color information is not stored on a per triangle basis but looks like it is located in Metadata/model_settings.config of the .3mf file. For each <part> tag there is an <extruder> tag which looks like it sets the filament for that entire part

[image]

[image]

THE-SIMPLE-MARK, 2024-02-06:
> Thanks everyone! This is really really a good suggestion and we had already noticed this feature from internal tester. Manual coloring is really time-consuming. We will investigate how to keep color of obj file firstly. And then step file. Because step file is imported by OCCT lib. Will also investigate whether occt support importing color information.

@SaltWei Thanks so much for the response! Can we get an ETA regarding the OBJ support? I don't want to implement my own solution if this feature gets implemented in the coming 1-3 months.

ukdavewood, 2024-02-28:
Started investigating this area myself today too and found this thread via a search for paint_color.  
What I was aiming to do is 
1. Export a coloured 3mf from blender.
2. Load it into Bambu Studio and then save it straight back out uncoloured in BS 3mf format
3. Automatically pick up on any saved 3mf files and then run a script which:
4. like @mypillowsateyou simply goes through the BS 3mf file triangles and creates a paint_color entry for every corresponding triangle in the blender exported 3mf file with a p1 entry.  - Have spent a while trying to figure out what paint_color values mean - but think I will just go with the list specified above in this thread.
5. Write out an updated BS format 3mf file and then automatically re-open it in BS

An alternative completely different approach to this is to write some sort of script in Blender that automatically splits up objects by materials but also includes some geometry behind all of the splits to fill in all of the gaps.

nhnifong, 2024-02-29:
Those compact strings appear to come from FacetsAnnotation::get_triangle_as_string() which is writing a hexadecimal representation of the selector map produced by TriangleSelector.serialize, but hats off to @mypillowsateyou for apparently reverse engineering it without looking at the source, among many other impressive feats.

https://github.com/bambulab/BambuStudio/blob/master/src/libslic3r/Model.cpp#L3651
https://github.com/bambulab/BambuStudio/blob/f6930bb6ab5005184dab3c0fb060394d2fb2f3dc/src/libslic3r/TriangleSelector.cpp#L1625

This code appears to be part of libslic3r and is identical between BambuStudio and PrusaSlicer.
The blender 3mf exporting plugin does not appear to have support for exporting color yet, as the authors said "there aren't very many good color 3d printers anyways" 4 years ago. :P
https://github.com/Ghostkeeper/Blender3mfFormat

But apparently Solidworks can export a colored 3mf file. it would be interesting to see where a simple colored 3mf file exported with solidworks differes from one saved in bambu studio or prusaslicer. does solidworks use the serialized recursive triangle selectors in the FacetsAnnotation data? hopefully if it does, it differs only in some minor detail, such as the name of an attribute.

The 3MF spec does not appear to exactly specify how textured models are supposed to be represented, but this recursive triangle selector trick just strikes me as kind of brittle and not extensible.

I would suggest to Bambu labs to take the initiative to store actual texture images and UV maps in the 3mf file, document the format publicly and clearly, and leave it to blender contributers and others to write export plugins that can match that format.

Within Bambu Studio, specifically libslic3r, add a new method similar to select_patch() in TriangleSelector which samples the texture using the UV map at a configurable resolution, and decides whether a given sample point should be selected or not depending on a user chosen target color and sensitivity value.

nhnifong, 2024-02-29:
or if saving the texture in 3mf is too complicated, continue saving the triangleselectors, but aim only to provide a painting tool that can fill areas by sampling a UV mapped texture from some other 3D file format.

SaltWei, 2024-03-01:
> @SaltWei Thanks so much for the response! Can we get an ETA regarding the OBJ support? I don't want to implement my own solution if this feature gets implemented in the coming 1-3 months.

3 months should be OK for obj support. We are moving at the moment

Haidiye00, 2024-03-19:
[image]
[image]
[image]
Hello, I have learned about three ways to store colors in obj files: vertex storage, lighting information storage in mtl files, and texture coordinate mtl+png storage. May I ask which of the three methods mentioned above is more commonly used and which is less commonly used? Looking forward to your reply

CptJackieSparrow, 2024-03-30:
Waow, this was extremely usefull! Thank you for sharing matey @mypillowsateyou !
Though i wonder is it possible to import Bambu exported 3MF with color data TO blender? Reverse of what you've did?
Couldn't find anything like it online. Tbh, bambu has better coloring tools than blender, for workflow at least!
I coudn't extract color data from 3mf files unfortunately..

Haidiye00, 2024-04-24:
Hello,obj import color function has been released, supporting vertex color and mtl color import. PNG color is currently not supported, please try the new version. thanks

amoose136, 2024-04-26:
Yay! You have no idea the lengths I had to go to last year to procedurally color some parts using Blender. Vertex color and mtl color support makes it so much more simple. Now all I need is a nice dynamic subdivision node setup (sort of like this: https://youtu.be/rP6mujoskmw ) and it should be almost trivial to convert a uv image textured object into something the slicer can use.

amoose136, 2024-04-26:
Worked for me in a quick test. https://www.youtube.com/watch?v=X4TbP9kpcz4

nhnifong, 2024-04-26:
I've tested using the following blend file and found that some imported colors are very different from what they were in blender
Consider the selected cockpit part here. It has a brown base with a material base color of 86,62,43

[cockpit_part.zip](https://github.com/bambulab/BambuStudio/files/15134564/cockpit_part.zip)

[image]

I exported the selected object with the following settings. Triangulate mesh is necessary to prevent an error when importing in bambu studio but that seems reasonable.
[image]

The color assigned to the base once imported in bambu studio is 23,13,6. 

[image]

This is pretty minor, because we can always change the color after importing to match what filament we have, but I thought I would let you know.
No other issues from what I can find.

Haidiye00, 2024-04-28:
[image]
There will indeed be a color difference issue because we did not use Phong lighting to pre render and obtain this color. We only used the diffuse color, so there will be differences

nhnifong, 2024-04-30:
I cannot see multiple colors imported on this test file.

[color experiment.zip](https://github.com/bambulab/BambuStudio/files/15157075/color.experiment.zip)

It should look like this, but after import into bambu studio it's all one color
[image]

Haidiye00, 2024-04-30:
[image]
Hello, you did not provide the mtl file for OBJ.like this 
[image]

nhnifong, 2024-04-30:
I see. I thought everything was contained in the OBJ. thank you for the response.

mypillowsateyou, 2024-05-02:
> ![image](https://private-user-images.githubusercontent.com/132528822/326221562-415d767b-4fd9-4c29-a771-66deaa9e057a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

I think the difference comes from the fact that Bambu Studio is showing the raw (linear) color space, but Blender does not default to raw color space, using instead something like sRGB (non-linear) or however the user has it configured. Look at this for example, shown first with sRGB and then raw colors

[image]

Bambu Studio is using the raw linear color space when displaying the colors and matching when importing the .obj, causing dark colors to appear much darker, and so dark in fact it is automatically mapping my brown and grey colors to black instead of to the grey and brown filaments I have loaded

[image]

So in whatever software we are generating our models from, we can set the color space to raw/linear to see how it will import into Bambu Studio and adjust our colors there before exporting, or it would be great if Bambu could add an option during import to treat colors as linear, sRGB, or other color spaces

By the way, kudos to everyone who worked on implementing the import with color, the implementation is far beyond what I had hoped for!!!

Haidiye00, 2024-05-06:
Thank you for your suggestion. I will take some time to learn about the benefits of SRGB and consider how to improve it. Thank you~

KodeMunkie, 2024-05-07:
@Haidiye00 Please can you filter duplicates in the popup dialog please? This print has just four colors (I know as I created it programmatically from a fixed palette matching my AMS filaments).
[image]

Haidiye00, 2024-05-08:
Hello, can I provide your OBJ file？thanks

KodeMunkie, 2024-05-08:
> Hello, can I provide your OBJ file？thanks

Sure. The original needs repairing due to an edges issue so I've provided the "repaired" version (nb. model is not as designed, it's now missing a few layers), but it this has the same issue of too many identical colours on import.

This model is large so may lock up BambuStudio for a bit on loading (my machine is quite high spec so it's not too bad, a couple of minutes)*.  When it's loaded there should only be 4 colours, white, red, blue and yellow.
[bbl-test.zip](https://github.com/bambulab/BambuStudio/files/15249160/bbl-test.zip)

*On a related note you might want to consider doing the loading on a background thread to the UI since if you click off BambuStudio and back on during the OBJ load Windows tells you it's unresponsive/offers to kill Bambu Studio as it's greyed out.

Haidiye00, 2024-05-09:
Thank you for your files and testing. We have identified the issue and are currently working on resolving it.
[image]
The algorithm results are 6 that should be the best

KodeMunkie, 2024-05-09:
That's great :) Interestingly, the colour component channels programmatically should have been either 0 or 255, which I thought was a related bug in your importer, but could be that the third party ply/obj exporter I'm using is modifying the per vertex colour to do with shading.
I'll double check the RGBs manually this evening in a text editor since it _should_ only have 4 primary colours in that file.

Haidiye00, 2024-05-09:
> 这真是太好了:)有趣的是，以编程方式，颜色分量通道应该是 0 或 255，我认为这是您的导入器进行着色时的一个相关错误，但可能是我正在使用的第三方 ply/obj 导出器正在修改每个顶点颜色。今晚我将在文本编辑器中手动仔细检查 RGB，因为它在该文件中_应该_只有 4 种原色。

[image]
After our algorithm detection, there are five colors, and the duplicates have been removed. There may be a slight color distance deviation between the algorithm and the original color, which should be normal. This has no significant impact. After entering the Bambu software, you can also make modifications.

KodeMunkie, 2024-05-09:
That seems better but there is no magenta (255,0,255) in the original algorithm I'm using. I'm commenting at the moment from a phone so can't check my file itself, but the histogram of the colours should show 0 instances for it. As mentioned this is also likely to be a bug in the exporters in my code :( but will check.

Haidiye00, 2024-05-09:
> 这似乎更好，但是我使用的原始算法中没有洋红色（255,0,255）。我现在正在通过手机发表评论，因此无法检查我的文件本身，但实际颜色使用的直方图应显示 0 个实例。如前所述，这也可能是我代码中导出器的错误:(但会检查。

[image]
These green box positions are magenta

KodeMunkie, 2024-05-09:
Thanks that saved me checking :) Now I have to find how they are getting into my export 😂 I bet it's doing some kind of 50/50 blending between 255.0.0 and 0,0,255 somewhere, and the additional differences are rounding errors, but these are my problems :)

KodeMunkie, 2024-05-12:
Please note I've added the PLY ticket in Orcaslicer rather than here as I require _multiple printer brand_ support for colour PLY files, not sure if it should have been in this project.
Please also note that another variant of my test file (as STL)  is causing Bambu Studio to outright crash https://github.com/bambulab/BambuStudio/issues/4039 .

Apologies for going off topic, this is just added for clarification on whether Orcaslicer was  the best place for the color PLY ticket and awareness of the instability in the BS beta slicer that might affect the way the slicer works in (final) models I may distribute that are created by the same software (models are valid).

Haidiye00, 2024-05-14:
This feature has been released in the latest version. Please download and use it. Thank you~

closed 2024-05-14 by Haidiye00

mew1033, 2024-05-17:
@Haidiye00 Is there a ticket for the PNG texture version of this? It sounds like it's being worked on, I just wanted to follow development.

FEMit160, 2024-07-11:
@haidiye00. Just wanted to say thank you for all your hard work incorporating the reading of color data for obj file import. It is a game changer. 

I do have another request, would you be able to do the same for importing VRML (.wrl) files and maintaining color? VRML file are an older format, but many engineering codes still use it. They encode the color per vertex/face as well, all in one file.

Thanks again!

Haidiye01, 2024-07-12:
Can you provide some VRML (. wrl) files? Let's take a look at this format. Additionally, I suggest that you use Blender software to convert VRML (. wrl) files into obj files.

FEMit160, 2024-07-12:
I can try to get you a file soon (don't have access from my phone).

Here is a good write up on the formatting. [(http://euclid.nmu.edu/~rappleto/Classes/CS295/VRML/vrml-coding.html)](url). Or can try exporting a colored obj file as a VRML from blender, if you already have.

Haidiye00, 2024-07-12:
> I can try to get you a file soon (don't have access from my phone).
> 
> Here is a good write up on the formatting [VRML syntax](http://euclid.nmu.edu/~rappleto/Classes/CS295/VRML/vrml-coding.html%5D) Or can try exporting a colored obj file as a VRML from blender, if you already have.
> 
> Edit: added example file of colored cube with cylinder [Example_VRML.zip](https://github.com/user-attachments/files/16188598/Example_VRML.zip)

[image]
you can use this software to convert wrl to obj.

FEMit160, 2024-07-12:
> ![image](https://private-user-images.githubusercontent.com/132528822/348145408-7920d4ae-f1d7-40a5-9b09-d6a9d9b795d1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Meshlab is actually one of the softwares I use. I just thought it would be good for Bambu Studio to natively import wrl files if it isn't too different to implement than obj import. Thank you

eried, 2024-08-08:
Hi! is something special needed for bambu studio to detect colors? 

I can get simple objects to work, but I was trying with a map with the texture applied on top (looks colores on Meshmixer, blender and 3d builder) but in bambulabs only 1 color is detected. 

[image]

Tried importing/exporting it in multiple ways but I still dont manage :/

[image]

FEMit160, 2024-08-09:
> Hi! is something special needed for bambu studio to detect colors?
>
> I can get simple objects to work, but I was trying with a map with the texture applied on top (looks colores on Meshmixer, blender [...]

How are your colors stored? Bambu Studio doesn't currently support png colors. See release note copied below:

"Imported obj file with color
Added color imported from the vertex in the .obj file, and color imported from the mtl file function.

You can use software such as Blender or Meshlab to color vertices or patches of the model.

The PNG color function is currently under development. Please stay tuned."

Haidiye00, 2024-08-09:
PNG color import is still under development, please wait~

Haidiye00, 2024-08-09:
> 你好！Bambu Studio 是否需要一些特殊的东西来检测颜色？
> 
> 我可以让简单的对象工作，但我正在尝试使用在顶部应用纹理的地图（在 Meshmixer、blender 和 3d builder 上看起来是颜色），但在 bambulabs 中只检测到 1 种颜色。
> 
> ![图像](https://private-user-images.githubusercontent.com/1091420/356238665-64ec5951-792a-4978-b11b-df22d95d0870.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMxNjUzNzYsIm5iZiI6MTcyMzE2NTA3NiwicGF0aCI6Ii8xMDkxNDIwLzM1NjIzODY2NS02NGVjNTk1MS03OTJhLTQ5NzgtYjExYi1kZjIyZDk1ZDA4NzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MDlUMDA1NzU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTQ2ZjM1ZjVmZWMxYzI1M2E4NmRlMWQ2ZTQ0Y2UwOWZiMzk2YzY1MjEzYWQ3MWVjNjdkM2NlZTQ3OTg5Mzg2ZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Vko2XWftqp7eJNzZa2iBgSwMHdS8yCvnSfW16GYOlL0)
> 
> 尝试以多种方式导入/导出它，但我仍然无法管理：/
> 
> ![图像](https://private-user-images.githubusercontent.com/1091420/356238799-c33859e5-6b3d-4b91-87a1-626858159e64.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMxNjUzNzYsIm5iZiI6MTcyMzE2NTA3NiwicGF0aCI6Ii8xMDkxNDIwLzM1NjIzODc5OS1jMzM4NTllNS02YjNkLTRiOTEtODdhMS02MjY4NTgxNTllNjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MDlUMDA1NzU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzMzN2QwNjcyNjc0NDExMTYxNTdhMWYyOTUzZjU4NDQyZTMwM2YxNzJiZjczMjkyZjljZjk4MmQ2YjE0ZWMyMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.X2I4MPb_yd-ElwuB70XuE3N4JyegoerjYvJUBwRXl60)

Hello, can you provide your complete obj file? Thank you

eried, 2024-08-09:
> PNG color import is still under development, please wait~

Thanks! you and @FEMit160 :) with meshlab worked out with the "texture -> vertex color" option.

Anzlc, 2024-08-11:
Hello,
i have been trying to import colored model into Bambu Studio but it detects only one color. I have baked textures to vertex color, but still no color in slicer. Are there any special export settings that should be checked. Thanks.

saywassup, 2025-09-24:
Hi, how can I see the list of colors possible to assign to vertices? 
I noticed these values: ['4', '8', '0C', '1C', '2C', '3C', '4C', '5C', '6C', '7C', '8C', '9C', 'AC', 'BC', 'CC', 'DC'].

But how can I correlate them with the actual colors?

kurtgluck, 2026-03-24:
> Hi, how can I see the list of colors possible to assign to vertices? I noticed these values: ['4', '8', '0C', '1C', '2C', '3C', '4C', '5C', '6C', '7C', '8C', '9C', 'AC', 'BC', 'CC', 'DC'].
> 
> But how can I correlate them with the actual colors? Is there some documentation about it?
> 
> [@Haidiye00](https://github.com/Haidiye00) [@SaltWei](https://github.com/SaltWei) [@lanewei120](https://github.com/lanewei120)

This may help: https://www.printables.com/article/3mf-file-color-specification-what-i-have-found-v4-YNYjaE5. the values are 4 or 8 bits (you are reading them in Hexadecimal).  the high order 6 or 2 bits specify the extruder number (from 0  to 16
00 -> use default color
01 -> extruder 1
10 -> extruder 2
0000 11 -> extruder 3
0001  11 -> extruder 4
0010 11 -> extruder 5
up to extruder 16.

I believe the extra 2 values above 16 are reserved so they can go with more bits in the future. 

The low order 2 bits describe how the triangle is subdivided as in 00 -> no subdivision, 01 -> into 2 triangles, 10 -> into 3, 11 -> into 4.    

This is only true in the model file or the files referenced from the model file.  The config file uses normal numbers for the part as a whole.

amoose136, 2026-03-24:
@kurtgluck 

Unless you're set on making your own script, I think there is already a way swap colors on a painted mesh. [There is a 3mf import/export extension](https://extensions.blender.org/add-ons/threemf-io/) for Blender that was originally developed by Ghostkeeper, then abandoned, and now picked up by Clonephaze, and published for free to the official Blender extensions website. Last year I did a little work on reverse engineering how the segmentation strings work and [when I shared my findings with Clonephaze he was able to quickly do all the actual work](https://github.com/Clonephaze/3MF-Blender-Add-on---Maintained/issues/8) of integrating that into the extension and now we have a free extension that can import and export 3mfs with modifications, including color swapping and painting with UV textures.